coolsnowwolf/lede #10563: 新下载的源码编译失败，机型是N1及HK1底包，附带日志

yishan2019, 2022-12-07:
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/tree-vect-data-refs.c:4218: undefined reference to `highest_pow2_factor(tree_node const*)'
/usr/bin/ld: target-globals.o: in function `save_target_globals()':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/target-globals.h:82: undefined reference to `this_target_expmed'
/usr/bin/ld: target-globals.o:(.data+0x38): undefined reference to `default_target_expmed'
/usr/bin/ld: c/c-lang.o:(.rodata+0x40): undefined reference to `gt_pch_rs_gt_except_h'
/usr/bin/ld: c/c-lang.o:(.rodata+0x180): undefined reference to `gt_ggc_r_gt_explow_h'
/usr/bin/ld: c/c-lang.o:(.rodata+0x190): undefined reference to `gt_ggc_r_gt_except_h'
/usr/bin/ld: lra-spills.o: in function `alter_subregs(rtx_def**, bool)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/lra-spills.c:680: undefined reference to `alter_subreg(rtx_def**, bool)'
/usr/bin/ld: lra-spills.o: in function `remove_pseudos(rtx_def**, rtx_insn*)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/lra-spills.c:431: undefined reference to `alter_subreg(rtx_def**, bool)'
/usr/bin/ld: aarch-common.o: in function `aarch_rev16_shright_mask_imm_p(rtx_def*, machine_mode)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/arm/aarch-common.c:128: undefined reference to `trunc_int_for_mode(long, machine_mode)'
/usr/bin/ld: aarch-common.o: in function `aarch_rev16_shleft_mask_imm_p(rtx_def*, machine_mode)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/arm/aarch-common.c:137: undefined reference to `trunc_int_for_mode(long, machine_mode)'
/usr/bin/ld: aarch-common.o: in function `aarch_rev16_shright_mask_imm_p(rtx_def*, machine_mode)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/arm/aarch-common.c:128: undefined reference to `trunc_int_for_mode(long, machine_mode)'
/usr/bin/ld: aarch-common.o: in function `aarch_rev16_shleft_mask_imm_p(rtx_def*, machine_mode)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/arm/aarch-common.c:137: undefined reference to `trunc_int_for_mode(long, machine_mode)'
/usr/bin/ld: aarch-common.o: in function `arm_md_asm_adjust(vec<rtx_def*, va_heap, vl_ptr>&, vec<rtx_def*, va_heap, vl_ptr>&, vec<machine_mode, va_heap, vl_ptr>&, vec<char const*, va_heap, vl_ptr>&, vec<rtx_def*, va_heap, vl_ptr>&, HARD_REG_SET&)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/arm/aarch-common.c:651: undefined reference to `convert_modes(machine_mode, machine_mode, rtx_def*, int)'
/usr/bin/ld: /home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/arm/aarch-common.c:652: undefined reference to `emit_move_insn(rtx_def*, rtx_def*)'
/usr/bin/ld: init-regs.o: in function `(anonymous namespace)::pass_initialize_regs::execute(function*)':
/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/init-regs.c:111: undefined reference to `emit_move_insn(rtx_def*, rtx_def*)'
collect2: error: ld returned 1 exit status
make[5]: *** [/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/gcc/c/Make-lang.in:87: cc1] Error 1
make[5]: Leaving directory '/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc'
make[4]: *** [Makefile:4428: all-gcc] Error 2
make[4]: Leaving directory '/home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[3]: *** [Makefile:37: /home/lhj/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/lhj/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#14.49#7.44#52.98
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/lhj/lede'
make[1]: *** [toolchain/Makefile:93: /home/lhj/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/lhj/lede'
make: *** [/home/lhj/lede/include/toplevel.mk:231：world] 错误 2

lloyd18, 2022-12-07:
一样
`rm -rf /home/l/lede/tmp/stage-firewall
mkdir -p /home/l/lede/tmp/stage-firewall/host /home/l/lede/staging_dir/target-x86_64_musl/packages
find /home/l/lede/tmp/stage-firewall -name '*.la' | xargs -r rm -f;
if [ -f /home/l/lede/staging_dir/target-x86_64_musl/packages/firewall.list ]; then /home/l/lede/scripts/clean-package.sh "/home/l/lede/staging_dir/target-x86_64_musl/packages/firewall.list" "/home/l/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/l/lede/tmp/stage-firewall ]; then (cd /home/l/lede/tmp/stage-firewall; find ./ > /home/l/lede/tmp/stage-firewall.files);  SHELL= flock /home/l/lede/tmp/.staging-dir.flock -c ' mv /home/l/lede/tmp/stage-firewall.files /home/l/lede/staging_dir/target-x86_64_musl/packages/firewall.list && cp -fpR /home/l/lede/tmp/stage-firewall/* /home/l/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/l/lede/tmp/stage-firewall
touch /home/l/lede/staging_dir/target-x86_64_musl/stamp/.firewall_installed
make[3]: Leaving directory '/home/l/lede/package/network/config/firewall'
time: package/network/config/firewall/compile#6.83#0.97#6.52
make[2]: Leaving directory '/home/l/lede'
make[1]: *** [package/Makefile:110: /home/l/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/l/lede'
make: *** [/home/l/lede/include/toplevel.mk:231: world] Error 2`

lloyd18, 2022-12-07:
看了置顶那个啥OUI，也不知道说的啥，按回复里的方法做了也没用~

ecrasy, 2022-12-08:
默认就不编译OUI的
这错误和OUI就没关联
这是编译工具链出错了
试试升级系统环境看看？

yishan2019, 2022-12-08:
看日志里好像是源码里有几个变量没定义，这用云编译好像可以过但有警告，但是本地编译就过不去了！发自我的 iPhone

lovehackintosh, 2022-12-08:
编译系统是什么？Ubuntu 18.06?

coolsnowwolf, 2022-12-08:
直接删除 staging_dir 目录 和  build_dir 目录后重新编译，你这个明显是工具链生成的时候出错

yishan2019, 2022-12-08:
Debian11.5

liuhejun2020, 2022-12-08:
> 直接删除 staging_dir 目录 和 build_dir 目录后重新编译，你这个明显是工具链生成的时候出错
我试试看，谢谢回复

liuhejun2020, 2022-12-09:
> 版本太新，降级吧
正在重装Debian 11，Ubuntu20.04编译盒子底包（N1等）结果打包后固件无IP，找了很长时间没结果，后来换成Debian后这个问题解决了，看来编译环境很重要

lloyd18, 2022-12-09:
> 直接删除 staging_dir 目录 和 build_dir 目录后重新编译，你这个明显是工具链生成的时候出错

编译成功了，谢谢。

closed 2022-12-18 by aiamadeus