coolsnowwolf/lede #10566: 新下载的源码，进行编译失败。

djjackle, 2022-12-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
新下载的源码，进行编译失败。

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

按照指引逐步操作，前面一路没报错，连续两次编译失败。。。。请大神帮忙看看什么情况

feeds.conf.default   
                       
src-git packages https://github.com/coolsnowwolf/packages
src-git luci https://github.com/coolsnowwolf/luci
src-git routing https://github.com/coolsnowwolf/routing
src-git telephony https://git.openwrt.org/feed/telephony.git
src-git helloworld https://github.com/fw876/helloworld
src-git passwall https://github.com/xiaorouji/openwrt-passwall
src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small
### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A： cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a   -L/home/djjackle/lede/staging_dir/host/lib -L/home/djjackle/lede/staging_dir/host/lib -L/home/djjackle/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz 
g++  -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/. -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/../include -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/../libcpp/include -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/../libcody -I/home/djjackle/lede/staging_dir/host/include -I/home/djjackle/lede/staging_dir/host/include -I/home/djjackle/lede/staging_dir/host/include  -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/../libdecnumber -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/../libbacktrace   -o cc1plus-checksum.o -MT cc1plus-checksum.o -MMD -MP -MF ./.deps/cc1plus-checksum.TPo cc1plus-checksum.c
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1plus \
      cp/cp-lang.o c-family/stub-objc.o cp/call.o cp/class.o cp/constexpr.o cp/constraint.o cp/coroutines.o cp/cp-gimplify.o cp/cp-objcp-common.o cp/cp-ubsan.o cp/cvt.o cp/cxx-pretty-print.o cp/decl.o cp/decl2.o cp/dump.o cp/error.o cp/except.o cp/expr.o cp/friend.o cp/init.o cp/lambda.o cp/lex.o cp/logic.o cp/mangle.o cp/mapper-client.o cp/mapper-resolver.o cp/method.o cp/module.o cp/name-lookup.o cp/optimize.o cp/parser.o cp/pt.o cp/ptree.o cp/rtti.o cp/search.o cp/semantics.o cp/tree.o cp/typeck.o cp/typeck2.o cp/vtable-class-hierarchy.o attribs.o incpath.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o i386-c.o glibc-c.o cc1plus-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a ../libcody/libcody.a  \
	libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a   -L/home/djjackle/lede/staging_dir/host/lib -L/home/djjackle/lede/staging_dir/host/lib -L/home/djjackle/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz 
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
make[6]: *** [/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/lto/Make-lang.in:95: lto1] Error 1
make[6]: *** Waiting for unfinished jobs....
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
make[6]: *** [/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cp/Make-lang.in:136: cc1plus] Error 1
rm gcc.pod
make[6]: Leaving directory '/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/gcc'
make[5]: *** [Makefile:4448: all-gcc] Error 2
make[5]: Leaving directory '/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
make[4]: *** [Makefile:976: all] Error 2
make[4]: Leaving directory '/home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
make[3]: *** [Makefile:91: /home/djjackle/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/.built] Error 2
make[3]: Leaving directory '/home/djjackle/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#1354.21#285.09#547.32
    ERROR: toolchain/gcc/final failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/final/compile] Error 1
make[2]: Leaving directory '/home/djjackle/lede'
make[1]: *** [toolchain/Makefile:93: /home/djjackle/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/djjackle/lede'
make: *** [/home/djjackle/lede/include/toplevel.mk:231：world] 错误 2

djjackle, 2022-12-08:
编译系统为Debian 11.x 64位虚拟机

aiamadeus, 2022-12-08:
看样子是你内存不够。

djjackle, 2022-12-09:
> 看样子是你内存不够。

我分配的是2046M内存，还需要调大么？我一会儿试试看看，分配硬盘是60G

closed 2022-12-09 by aiamadeus