coolsnowwolf/lede #10568: [mt7613] [Driver mt7615d]: rai0 is not present when using MT Drivers

alessandrobrugnera, 2022-12-08:
Router model: TP-Link Archer C6 V3
Chipset: MT7621 (Wireless MT7603E + MT7613BE)
LEDE: Default except for WiFi drivers and luci mtk wifi addon. I used MTK Drivers (7603e and 7615d).

I compiled the firmware (22.03) and uploaded. It works, excepts for 5GHz WiFi. I can't bring the interface up, actually the interface (rai0) is not even present when I perform `ifconfig -a`. When I click Enable on the AP it just says "Please restart manually to create this interface.". When I then click Reload it refreshes the paga and in the log it says that the interface rai0 is not present.

What can I do to fix this and make the rai0 interface appear? The standard mt76 drivers find the interface but they are not at all stable. That's why I wanted to use the MTK drivers.

bricco1981, 2023-01-02:
yes i'm in the same situation, it's just driving me crazy

coolsnowwolf, 2026-09-05:
该报告使用 TP-Link Archer C6 V3 的厂商 MTK mt7615d 驱动与旧 MTK LuCI 插件；当前仓库主线使用 mac80211/mt76，厂商 `rai0` 接口不属于当前支持栈，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf