coolsnowwolf/lede #10569: 编译一个dropbear 的新版本报错

yuhongwei380, 2022-12-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
想要自编译一个dropbear 的新版本。
Q：是否搜索了issue (使用 "x" 选择)
* [√ ] 没有类似的issue

## 2. 详细叙述
想要自编译一个dropbear 的新版本，在替换了dropbear.tar.bz2后，同时也修改了dropbear 的Makefile中的内容为对应版本，hash也修改了。编译发现有报错，想问问怎么解决？
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：
mkdir -p /home/vesoft/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/vesoft/lede/tmp/.root-copy.flock -c 'cp -fpR /home/vesoft/lede/build_dir/target-x86_64_musl/linux-firmware-20221012/.pkgdir/bnx2x-firmware/. /home/vesoft/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/vesoft/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2x-firmware_installed
echo "bnx2x-firmware" >> /home/vesoft/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/vesoft/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#2.06#0.94#2.93
make[2]: Leaving directory '/home/vesoft/lede'
make[1]: *** [package/Makefile:110: /home/vesoft/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/vesoft/lede'
make[1]: Leaving directory '/home/vesoft/lede'
make: *** [/home/vesoft/lede/include/toplevel.mk:230: world] Error 2

yuhongwei380, 2022-12-09:
我目前直接暴力的把22.83的dropbear包 改成22.82的名字，然后在dropbear 的makefile 中修改了 hash为22.83的hash，目前编译还没有遇到问题。还在编译中

yuhongwei380, 2022-12-09:
因为22.82版本我们跨网段访问是有bug（准确说是有ipqos相关的，之前遇到过linux 的ssh的特定版本也有相关问题），之前给dropbear 的作者提了issue，人家出了个包，我就想测试下。

WYC-2020, 2022-12-09:
> 因为22.82版本我们跨网段访问是有bug（准确说是有ipqos相关的，之前遇到过linux 的ssh的特定版本也有相关问题），之前给dropbear 的作者提了issue，人家出了个包，我就想测试下。

原来如此，新版地址在哪里嘛 我来升级

yuhongwei380, 2022-12-09:
https://github.com/mkj/dropbear

yuhongwei380, 2022-12-09:
> > 因为22.82版本我们跨网段访问是有bug（准确说是有ipqos相关的，之前遇到过linux 的ssh的特定版本也有相关问题），之前给dropbear 的作者提了issue，人家出了个包，我就想测试下。
> 
> 原来如此，新版地址在哪里嘛 我来升级

https://github.com/mkj/dropbear

yuhongwei380, 2022-12-09:
果然暴力操作还是会报错，还是等老哥升级吧

WYC-2020, 2022-12-09:
[image]
[dropbear.tar.gz](https://github.com/coolsnowwolf/lede/files/10191776/dropbear.tar.gz)
[image]
自己拿去替换升级

yuhongwei380, 2022-12-09:
好的谢谢

mitsukileung, 2022-12-09:
大佬们，提个PR吧

yuhongwei380, 2022-12-09:
> [image] [dropbear.tar.gz](https://github.com/coolsnowwolf/lede/files/10191776/dropbear.tar.gz) [image] 自己拿去替换升级

[image]
我ti'h替换了dropbear的包以后，编译了一版，还是有报错。麻烦大佬看看?

mitsukileung, 2022-12-09:
替换后可以编译出来，我是 rm -rf ./tmp && rm -rf .config 后再编译的

[image]

yuhongwei380, 2022-12-09:
> 替换后可以编译出来，我是 rm -rf ./tmp && rm -rf .config 后再编译的
> 
> [image]

然后重新make menuconfig 跑make download 和make 吗；

mitsukileung, 2022-12-09:
> > 替换后可以编译出来，我是 rm -rf ./tmp && rm -rf .config 后再编译的
> > [image]
> 
> 然后重新make menuconfig 跑make download 和make 吗；

是呀

yuhongwei380, 2022-12-09:
OK感谢老哥的IPK，我刚刚重新编，还是会有问题，我先用老哥你这个ipk测

yuhongwei380, 2022-12-09:
测了下发现还是没解决我走ipec VPN  跨网段无法访问的问题，我还是老老实实用20.X那个版本吧

mitsukileung, 2022-12-09:
> 测了下发现还是没解决我走ipec VPN 跨网段无法访问的问题，我还是老老实实用20.X那个版本吧

IPEC VPN 连上后无法SSH？

WYC-2020, 2022-12-09:
> OK感谢老哥的IPK，我刚刚重新编，还是会有问题，我先用老哥你这个ipk测

有问题不是我给的那个有问题 是你其他的库，我这边编译x86 完全没问题

WYC-2020, 2022-12-09:
> 因为22.82版本我们跨网段访问是有bug（准确说是有ipqos相关的，之前遇到过linux 的ssh的特定版本也有相关问题），之前给dropbear 的作者提了issue，人家出了个包，我就想测试下。

他出的什么包那个提交在哪里 发给我 我看下 这个版本他加入了没

yuhongwei380, 2022-12-09:
> > 因为22.82版本我们跨网段访问是有bug（准确说是有ipqos相关的，之前遇到过linux 的ssh的特定版本也有相关问题），之前给dropbear 的作者提了issue，人家出了个包，我就想测试下。
> 
> 他出的什么包那个提交在哪里 发给我 我看下 这个版本他加入了没

https://github.com/mkj/dropbear/releases

yuhongwei380, 2022-12-09:
> > 测了下发现还是没解决我走ipec VPN 跨网段无法访问的问题，我还是老老实实用20.X那个版本吧
> 
> IPEC VPN 连上后无法SSH？

我们是公司内做了ipsec VPN，本机网段15.X  ， 对端OP 网段16.X，网段都互通的。  dropbear 的20.X的版本我们正常可以ssh上，22.82就无法ssh到16.X的op上。

yuhongwei380, 2022-12-09:
> > OK感谢老哥的IPK，我刚刚重新编，还是会有问题，我先用老哥你这个ipk测
> 
> 有问题不是我给的那个有问题 是你其他的库，我这边编译x86 完全没问题

不过我直接切回22.82 默认的包的话，确实编译的时候其他库也没报错，尬住了。

WYC-2020, 2022-12-09:
说是这个 pr修复了，那需要修改init 增加-z 自己试下https://github.com/mkj/dropbear/commit/2864c3d15499ac133b4fffbc27787671c11939d5

yuhongwei380, 2022-12-09:
> 说是这个 pr修复了，那需要修改init 增加-z 自己试下[mkj/dropbear@2864c3d](https://github.com/mkj/dropbear/commit/2864c3d15499ac133b4fffbc27787671c11939d5)

我提的issue，然后作者确实也是这么说的，所以应该在/etc/init.d/dropbear文件的第几行呢？我看了下好像没啥头绪

yuhongwei380, 2022-12-09:
> 

我能斗胆问你一句，在哪一行哪个位置吗；不太好判断这个ipqos 的标志位加在哪里啊，没看到关键词呢？

yuhongwei380, 2022-12-09:
> > 
> 
> 我能斗胆问你一句，在哪一行哪个位置吗；不太好判断这个ipqos 的标志位加在哪里啊，没看到关键词呢？
这里？procd_append_param command -b "${BannerFile}" 把-b改成在-z？

yuhongwei380, 2022-12-09:
> > > 
> > 
> > 
> > 我能斗胆问你一句，在哪一行哪个位置吗；不太好判断这个ipqos 的标志位加在哪里啊，没看到关键词呢？

yuhongwei380, 2022-12-13:
procd_set_param command "$PROG" -F -P "$pid_file" -z

WYC-2020, 2022-12-16:
> procd_set_param command "$PROG" -F -P "$pid_file" -z

那我修改一下luci 界面 吧这个改成可选

yuhongwei380, 2022-12-16:
嗯，dropbear也可以提个pr 更新到新版本22.83

WYC-2020, 2022-12-16:
[image]
[image]
麻烦。不提，要提你去提吧 @yuhongwei380

yuhongwei380, 2023-08-21:
> [image] [image] 麻烦。不提，要提你去提吧 @yuhongwei380

感觉还是得提个pr，现在新版本6.1的kernel已经不支持上面老哥编译出来的包了。感觉跨网段用的人很少。

closed 2026-09-05 by coolsnowwolf