coolsnowwolf/lede #10578: 电犀牛 R66s 日志 rss get rxnfc

miaoermua, 2022-12-10:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

日志提示 rss get rxnfc 是网卡问题吗，还是驱动问题亦或是插件冲突；求解答解决问题。

### (2) 路由器型号和固件版本

型号 | FastRhino R66S
架构 | ARMv8 Processor x 4 (1800MHz, 41.2°C)
固件版本 | OpenWrt R22.12.1 / LuCI Master (git-22.339.26541-ba1f838)

使用最新提交的代码 https://github.com/coolsnowwolf/lede/commit/7979de9ed969ce7178e662a72233f1db30f758be 编译

### 系统日志

```
Sat Aug  5 17:00:43 2017 user.info kernel: [    7.329687] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Sat Aug  5 17:00:43 2017 user.info kernel: [    7.333491] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Sat Aug  5 17:00:43 2017 kern.info kernel: [    7.548020] loop0: detected capacity change from 0 to 1302528
Sat Aug  5 17:00:43 2017 kern.info kernel: [    7.564741] loop0: detected capacity change from 1302528 to 990208
Sat Aug  5 17:00:43 2017 user.info kernel: [    7.566837] block: overlay filesystem in /dev/loop0 has not been formatted yet
Sat Aug  5 17:00:43 2017 kern.notice kernel: [    8.053328] F2FS-fs (loop0): Found nat_bits in checkpoint
Sat Aug  5 17:00:43 2017 kern.notice kernel: [    8.062963] F2FS-fs (loop0): Mounted with checkpoint version = 63bce0d3
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.063744] block: attempting to load /tmp/f2fs_cfg/upper/etc/config/fstab
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.064550] block: unable to load configuration (fstab: Entry not found)
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.065230] block: attempting to load /tmp/f2fs_cfg/etc/config/fstab
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.065871] block: unable to load configuration (fstab: Entry not found)
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.066516] block: attempting to load /etc/config/fstab
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.076750] block: unable to load configuration (fstab: Entry not found)
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.077402] block: no usable configuration
Sat Aug  5 17:00:43 2017 kern.info kernel: [    8.141617] loop0: detected capacity change from 0 to 1302528
Sat Aug  5 17:00:43 2017 kern.info kernel: [    8.156755] loop0: detected capacity change from 1302528 to 990208
Sat Aug  5 17:00:43 2017 kern.notice kernel: [    8.162518] F2FS-fs (loop0): Found nat_bits in checkpoint
Sat Aug  5 17:00:43 2017 kern.notice kernel: [    8.172130] F2FS-fs (loop0): Mounted with checkpoint version = 63bce0d5
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.213665] block: attempting to load /tmp/f2fs_cfg/upper/etc/config/fstab
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.214425] block: unable to load configuration (fstab: Entry not found)
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.215118] block: attempting to load /tmp/f2fs_cfg/etc/config/fstab
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.215836] block: unable to load configuration (fstab: Entry not found)
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.216525] block: attempting to load /etc/config/fstab
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.217153] block: unable to load configuration (fstab: Entry not found)
Sat Aug  5 17:00:43 2017 user.err kernel: [    8.217836] block: no usable configuration
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.219233] mount_root: overlay filesystem has not been fully initialized yet
Sat Aug  5 17:00:43 2017 user.info kernel: [    8.220397] mount_root: switching to f2fs overlay
Sat Aug  5 17:00:43 2017 kern.info kernel: [    8.585862] EXT4-fs (mmcblk0p1): mounted filesystem without journal. Opts: (null). Quota mode: none.
Sat Aug  5 17:00:43 2017 user.warn kernel: [    8.700435] urandom-seed: Seed file not found (/etc/urandom.seed)
Sat Aug  5 17:00:43 2017 user.info kernel: [    9.173880] procd: - early -
Sat Aug  5 17:00:43 2017 user.info kernel: [    9.174228] procd: - watchdog -
Sat Aug  5 17:00:43 2017 user.info kernel: [    9.951896] procd: - watchdog -
Sat Aug  5 17:00:43 2017 user.info kernel: [   10.057150] procd: - ubus -
Sat Aug  5 17:00:43 2017 user.info kernel: [   10.170264] procd: - init -
Sat Aug  5 17:00:43 2017 user.info kernel: [   10.650022] urngd: v1.0.2 started.
Sat Aug  5 17:00:43 2017 user.info kernel: [   11.354476] kmodloader: loading kernel modules from /etc/modules.d/*
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.425459] NET: Registered PF_ALG protocol family
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.435490] async_tx: api initialized (async)
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.443490] device-mapper: ioctl: 4.45.0-ioctl (2021-03-22) initialised: dm-devel@redhat.com
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.449674] ntfs3: Max link count 4000
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.450022] ntfs3: Enabled Linux POSIX ACLs support
Sat Aug  5 17:00:43 2017 kern.notice kernel: [   11.450449] ntfs3: Warning: Activated 64 bits per cluster. Windows does not support this
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.451153] ntfs3: Read-only LZX/Xpress compression included
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.455673] NET: Registered PF_KEY protocol family
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.457594] Initializing XFRM netlink socket
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.459305] tun: Universal TUN/TAP device driver, 1.6
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.466072] ipip: IPv4 and MPLS over IPv4 tunneling driver
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.472245] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
Sat Aug  5 17:00:43 2017 kern.info kernel: [   11.504645] cryptodev: driver 1.12 loaded.
[... 68 lines cut ...]
Sat Aug  5 17:00:47 2017 kern.info kernel: [   45.219248] eth1: 0xffff8000093d0000, 3a:e6:35:f4:44:97, IRQ 47
Sat Aug  5 17:00:47 2017 daemon.notice netifd: Interface 'wan' is enabled
Sat Aug  5 17:00:47 2017 daemon.notice netifd: Interface 'wan6' is enabled
Sat Aug  5 17:00:47 2017 daemon.notice netifd: Network device 'lo' link is up
Sat Aug  5 17:00:47 2017 daemon.notice netifd: Interface 'loopback' has link connectivity
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.124233 [info] AdGuard Home, version v0.107.18
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.124325 [info] This is the first time AdGuard Home is launched
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.124343 [info] Checking if AdGuard Home has necessary permissions
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.124748 [info] AdGuard failed to bind to port 53: listen tcp 127.0.0.1:53: bind: address already in use
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]:
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: Please note, that this is crucial for a DNS server to be able to use that port.
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.124784 [info] AdGuard Home can bind to port 53
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.124794 [info] AdGuard Home updates are disabled
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.129722 [error] creating dhcpv4 srv: dhcpv4: invalid IP is not an IPv4 address
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.129949 [info] Initializing auth module: /tmp/adguardhome/data/sessions.db
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.130171 [info] auth: initialized.  users:0  sessions:0
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.130216 [info] web: initializing
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.130252 [info] This is the first launch of AdGuard Home, redirecting everything to /install.html
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.130283 [info] AdGuard Home is available at the following addresses:
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.131479 [info] Go to http://127.0.0.1:3000
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.131526 [info] Go to http://[::1]:3000
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.131538 [info] Go to http://192.168.1.4:3000
Sat Aug  5 17:00:48 2017 daemon.err AdGuardHome[2596]: 2017/08/05 09:00:48.131550 [info] Go to http://[fd5d:422b:131c::1]:3000
Sat Aug  5 17:00:48 2017 daemon.err odhcpd[2665]: Failed to send to ff02::1%lan@br-lan (Address not available)
```

miaoermua, 2022-12-10:
```
Sat Aug  5 17:00:48 2017 authpriv.warn dropbear[3227]: Failed listening on '22': Error listening: Address not available
Sat Aug  5 17:00:48 2017 authpriv.info dropbear[3227]: Not backgrounding
Sat Aug  5 17:00:49 2017 daemon.notice procd: /etc/rc.d/S60dnsforwarder: luci for dnsforwarder
Sat Aug  5 17:00:49 2017 daemon.notice procd: /etc/rc.d/S60dnsforwarder: dnsforwarder is not enabled
Sat Aug  5 17:00:49 2017 daemon.notice procd: /etc/rc.d/S90ipsec: Command failed: ubus call service delete { "name": "ipsec" } (Not found)
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Successfully dropped root privileges.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: avahi-daemon 0.8 starting up.
Sat Aug  5 17:00:49 2017 daemon.warn avahi-daemon[3469]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: No service file found in /etc/avahi/services.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Joining mDNS multicast group on interface br-lan.IPv6 with address fd5d:422b:131c::1.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: New relevant interface br-lan.IPv6 for mDNS.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.1.4.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: New relevant interface br-lan.IPv4 for mDNS.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: New relevant interface lo.IPv6 for mDNS.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: New relevant interface lo.IPv4 for mDNS.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Network interface enumeration completed.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Registering new address record for fd5d:422b:131c::1 on br-lan.*.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Registering new address record for 192.168.1.4 on br-lan.IPv4.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Registering new address record for ::1 on lo.*.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Registering new address record for 127.0.0.1 on lo.IPv4.
Sat Aug  5 17:00:49 2017 daemon.info avahi-daemon[3469]: Registering HINFO record with values 'AARCH64'/'LINUX'.
Sat Aug  5 17:00:49 2017 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Sat Aug  5 17:00:49 2017 daemon.notice procd: /etc/rc.d/S90kms: uci: Entry not found
Sat Aug  5 17:00:50 2017 daemon.notice netifd: Network device 'eth1' link is up
Sat Aug  5 17:00:50 2017 daemon.notice netifd: Interface 'wan' has link connectivity
Sat Aug  5 17:00:50 2017 daemon.notice netifd: Interface 'wan' is setting up now
Sat Aug  5 17:00:50 2017 kern.info kernel: [   47.568447] r8125: eth1: link up
Sat Aug  5 17:00:50 2017 kern.info kernel: [   47.568817] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
Sat Aug  5 17:00:50 2017 daemon.notice netifd: Interface 'wan6' has link connectivity
Sat Aug  5 17:00:50 2017 daemon.notice netifd: Interface 'wan6' is setting up now
Sat Aug  5 17:00:50 2017 daemon.notice netifd: wan (3665): udhcpc: started, v1.35.0
Sat Aug  5 17:00:50 2017 daemon.err odhcp6c[3663]: Failed to send RS (Address not available)
Sat Aug  5 17:00:50 2017 daemon.notice netifd: wan (3665): udhcpc: broadcasting discover
Sat Aug  5 17:00:50 2017 daemon.notice netifd: wan (3665): udhcpc: broadcasting select for 192.168.65.92, server 192.168.65.4
Sat Aug  5 17:00:50 2017 daemon.notice netifd: wan (3665): udhcpc: lease of 192.168.65.92 obtained from 192.168.65.4, lease time 43200
Sat Aug  5 17:00:50 2017 daemon.info avahi-daemon[3469]: Joining mDNS multicast group on interface eth1.IPv4 with address 192.168.65.92.
Sat Aug  5 17:00:50 2017 daemon.info avahi-daemon[3469]: New relevant interface eth1.IPv4 for mDNS.
Sat Aug  5 17:00:50 2017 daemon.info avahi-daemon[3469]: Registering new address record for 192.168.65.92 on eth1.IPv4.
Sat Aug  5 17:00:50 2017 daemon.notice netifd: Interface 'wan' is now up
Sat Aug  5 17:00:50 2017 daemon.err odhcp6c[3663]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sat Aug  5 17:00:51 2017 daemon.info avahi-daemon[3469]: Server startup complete. Host name is CatWrt.local. Local service cookie is 46191152.
Sat Aug  5 17:00:51 2017 daemon.notice netifd: Network device 'eth0' link is up
Sat Aug  5 17:00:51 2017 daemon.notice netifd: bridge 'br-lan' link is up
Sat Aug  5 17:00:51 2017 daemon.notice netifd: Interface 'lan' has link connectivity
Sat Aug  5 17:00:51 2017 kern.info kernel: [   48.836445] r8125: eth0: link up
Sat Aug  5 17:00:51 2017 kern.info kernel: [   48.836838] br-lan: port 1(eth0) entered blocking state
Sat Aug  5 17:00:51 2017 kern.info kernel: [   48.837311] br-lan: port 1(eth0) entered forwarding state
Sat Aug  5 17:00:51 2017 kern.info kernel: [   48.838092] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Sat Aug  5 17:00:51 2017 daemon.err odhcpd[2665]: Failed to send to fe80::303a:c8c0:456e:a538%lan@br-lan (Address not available)
[... cut ...]
Sat Dec 10 12:41:59 2022 kern.info kernel: [   76.058269] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:41:59 2022 kern.info kernel: [   76.059420] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:41:59 2022 kern.info kernel: [   76.060062] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:41:59 2022 kern.info kernel: [   76.060838] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:41:59 2022 kern.info kernel: [   76.061350] r8125 0002:01:00.0 eth1: rss get rxnfc
Sat Dec 10 12:41:59 2022 kern.info kernel: [   76.061829] r8125 0002:01:00.0 eth1: rss get rxnfc
```

miaoermua, 2022-12-10:
内核日志

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.82 (by@by) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5525-7979de9ed) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Fri Dec 9 10:18:58 2022
[    0.000000] Machine model: FastRhino R66S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x000000003e400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33432 r8192 d32104 u73728
[    0.000000] pcpu-alloc: s33432 r8192 d32104 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1001532K/1046528K available (8768K kernel code, 2380K rwdata, 2476K rodata, 1664K init, 726K bss, 36804K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: MBI range [94:124]
[    0.000000] GICv3: MBI range [229:259]
[    0.000000] GICv3: MBI range [289:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001360] Console: colour dummy device 80x25
[    0.001848] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002901] pid_max: default: 32768 minimum: 301
[    0.003578] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004318] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007386] rcu: Hierarchical SRCU implementation.
[    0.008134] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.009619] smp: Bringing up secondary CPUs ...
[    0.010778] Detected VIPT I-cache on CPU1
[... 204 lines cut ...]
[    0.845144] vdda_0v9: supplied by vcc3v3_sys
[    0.849944] vdda0v9_pmu: supplied by vcc3v3_sys
[    0.854279] vccio_acodec: Bringing 600000uV into 3300000-3300000uV
[    0.857537] vccio_acodec: supplied by vcc3v3_sys
[    0.862228] vccio_sd: supplied by vcc3v3_sys
[    0.867042] vcc3v3_pmu: supplied by vcc3v3_sys
[    0.872214] vcca_1v8: supplied by vcc3v3_sys
[    0.877060] vcca1v8_pmu: supplied by vcc3v3_sys
[    0.881380] vcca1v8_image: Bringing 600000uV into 950000-950000uV
[    0.885742] vcca1v8_image: supplied by vcc3v3_sys
[    0.890464] vcc_3v3: supplied by vcc3v3_sys
[    0.893183] vcc3v3_sd: supplied by vcc3v3_sys
[    0.906174] rk808-rtc rk808-rtc: registered as rtc0
[    0.909272] rk808-rtc rk808-rtc: setting system clock to 2017-08-05T09:00:05 UTC (1501923605)
[    0.917046] sdhci: Secure Digital Host Controller Interface driver
[    0.917608] sdhci: Copyright(c) Pierre Ossman
[    0.917995] Synopsys Designware Multimedia Card Interface Driver
[... cut ...] form and OF driver helper
[    0.919720] dwmmc_rockchip fe2b0000.mmc: IDMAC supports 32-bit address mode.
[    0.920415] dwmmc_rockchip fe2b0000.mmc: Using internal DMA controller.
[    0.920471] ledtrig-cpu: registered to indicate activity on CPUs
[    0.921013] dwmmc_rockchip fe2b0000.mmc: Version ID is 270a
[    0.921703] arm-scmi firmware:scmi: Failed. SCMI protocol 17 not active.
[    0.922060] dwmmc_rockchip fe2b0000.mmc: DW MMC controller at irq 21,32 bit host data width,256 deep fifo
```

AutoCONFIG, 2022-12-15:
这是网卡驱动的有关代码：
int rtl8125_get_rxnfc(struct net_device *dev, struct ethtool_rxnfc *cmd,
                      u32 *rule_locs)
{
        struct rtl8125_private *tp = netdev_priv(dev);
        int ret = -EOPNOTSUPP;

        netif_info(tp, drv, tp->dev, "rss get rxnfc\n");

        if (!(dev->features & NETIF_F_RXHASH))
                return ret;

        switch (cmd->cmd) {
        case ETHTOOL_GRXRINGS:
                cmd->data = rtl8125_tot_rx_rings(tp);
                ret = 0;
                break;
        case ETHTOOL_GRXFH:
                ret = rtl8125_get_rss_hash_opts(tp, cmd);
                break;
        default:
                break;
        }

        return ret;
}

具体这串代码是用于什么的，我没研究，您可以看看这个函数在什么地方用的，应该就知道了。
不过至少可以看出不是网卡的问题，应该只是网卡接受到rxnfc然后用于调试的一串打印信息，是正常的，不影响正常使用。

miaoermua, 2022-12-20:
目前看起来虽然会报 rss get rxnfc，但应该没有掉网卡，如果我测试几天没问题就 close

wjz304, 2022-12-24:
从源码看，这并不是一个error，只是一个 info log，注掉这个log就好了。

miaoermua, 2022-12-29:
> 从源码看，这并不是一个error，只是一个 info log，注掉这个log就好了。

感谢，目前来看没问题，非常感谢 [AutoCONFIG](https://github.com/AutoCONFIG) 解答，该 issues 将关闭

closed 2022-12-29 by miaoermua