coolsnowwolf/lede #10587: 云编译报错

binge8, 2022-12-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云变异报错

### (2) 路由器型号和固件版本

A：x86，armv8

### (3) 详细日志

A：看截图
[image]
[image]

closed 2022-12-11 by aiamadeus

binge8, 2022-12-11:
这是怎么了什么话也不说就直接关闭了，就是不懂才问问啊，刚拉的最新源码最新编译环境还是同样的错误

WYC-2020, 2022-12-11:
能不被关才有鬼，日志都不详细，你是让人算命看是什么问题么

binge8, 2022-12-11:
[日志.txt](https://github.com/coolsnowwolf/lede/files/10202463/default.txt)

> 能不被关才有鬼，日志都不详细，你是让人算命看是什么问题么

只有这里报错啊大哥    我把全部日志都附上你看看

binge8, 2022-12-11:
2022-12-11T09:43:07.4972464Z WARNING: Makefile 'package/feeds/packages/adblock/Makefile' has a dependency on 'opkg', which does not exist
2022-12-11T09:43:09.3224729Z make[2]: Entering directory '/home/runner/work/op/op/openwrt/scripts/config'
2022-12-11T09:43:09.3225448Z make[2]: 'conf' is up to date.
2022-12-11T09:43:09.3226035Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt/scripts/config'
2022-12-11T09:43:11.9052185Z make[1]: Entering directory '/home/runner/work/op/op/openwrt'
2022-12-11T09:43:15.1530093Z make[2]: Entering directory '/home/runner/work/op/op/openwrt'
2022-12-11T09:43:18.2443885Z make[3]: Entering directory '/home/runner/work/op/op/openwrt/target/linux'
2022-12-11T09:43:18.2689820Z make[4]: Entering directory '/home/runner/work/op/op/openwrt/target/linux/x86'
2022-12-11T09:43:18.3702899Z rm -f /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/localversion
2022-12-11T09:43:18.3729255Z /home/runner/work/op/op/openwrt/scripts/kconfig.pl  + + /home/runner/work/op/op/openwrt/target/linux/generic/config-5.15 /home/runner/work/op/op/openwrt/target/linux/x86/config-5.15 /home/runner/work/op/op/openwrt/target/linux/x86/64/config-5.15 > /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.target
2022-12-11T09:43:18.4417674Z awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/op/op/openwrt/.config >> /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.target
2022-12-11T09:43:18.4497232Z echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.target
2022-12-11T09:43:18.4517292Z echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.target
2022-12-11T09:43:18.4536769Z echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.target
2022-12-11T09:43:18.4557270Z /home/runner/work/op/op/openwrt/scripts/package-metadata.pl kconfig /home/runner/work/op/op/openwrt/tmp/.packageinfo /home/runner/work/op/op/openwrt/.config 5.15 > /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.override
2022-12-11T09:43:18.5838088Z /home/runner/work/op/op/openwrt/scripts/kconfig.pl 'm+' '+' /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.target /dev/null /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.override > /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set
2022-12-11T09:43:18.6645012Z mv /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.old
2022-12-11T09:43:18.6680329Z grep -v INITRAMFS /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.old > /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set
2022-12-11T09:43:18.6736812Z echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set
2022-12-11T09:43:18.6757591Z echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set
2022-12-11T09:43:18.6812140Z rm -rf /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
2022-12-11T09:43:18.6838752Z cmp -s /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.prev || { cp /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config; cp /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.set /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/.config.prev; }
2022-12-11T09:43:18.6878973Z export MAKEFLAGS= ; [ -d /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/user_headers ] || make -C /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82 KCFLAGS="-fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/op/op/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Dec 11 07:28:35 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/op/op/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.82 CC="x86_64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.82/user_headers headers_install
[... cut ...]
2022-12-11T09:44:27.1046937Z make[2]: *** No rule to make target 'package/system/opkg/host/compile', needed by 'package/compile'.  Stop.
2022-12-11T09:44:27.1053532Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
2022-12-11T09:44:27.1077533Z make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2022-12-11T09:44:27.1094923Z make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
2022-12-11T09:44:27.1132874Z make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2

binge8, 2022-12-11:
自从12月8日源码更新后就一直报这个错误到今天

lovehackintosh, 2022-12-12:
`No rule to make target 'package/system/opkg/host/compile', needed by 'package/compile'. Stop.`
这是你一步一步编译的吗，`make -j$(nproc)` 再试试

binge8, 2022-12-12:
> `No rule to make target 'package/system/opkg/host/compile', needed by 'package/compile'. Stop.` 这是你一步一步编译的吗，`make -j$(nproc)` 再试试

[image]