coolsnowwolf/lede #10588: 12月11号新装的Debian11 编译出现如下错误，两次删除build_dir和staging_dir目录都无法过。

liuhejun2020, 2022-12-11:
N1底包

| on quite a few occasions.                                   |
|                                                             |
+-------------------------------------------------------------+

make[8]: Leaving directory '/home/lhj/lede/build_dir/host/gmp-6.2.1'
make[7]: Leaving directory '/home/lhj/lede/build_dir/host/gmp-6.2.1'
make[6]: Leaving directory '/home/lhj/lede/build_dir/host/gmp-6.2.1'
make[5]: Leaving directory '/home/lhj/lede/build_dir/host/gmp-6.2.1'
make[4]: Leaving directory '/home/lhj/lede/build_dir/host/gmp-6.2.1'
mkdir -p /home/lhj/lede/staging_dir/host/stamp
touch /home/lhj/lede/build_dir/host/gmp-6.2.1/.built
touch /home/lhj/lede/staging_dir/host/stamp/.gmp_installed
make[3]: Leaving directory '/home/lhj/lede/tools/gmp'
time: tools/gmp/compile#42.52#8.87#51.56
make[3]: Entering directory '/home/lhj/lede/tools/mpfr'
. /home/lhj/lede/include/shell.sh; xzcat /home/lhj/lede/dl/mpfr-4.1.0.tar.xz | tar -C /home/lhj/lede/build_dir/host/mpfr-4.1.0/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/lhj/lede/build_dir/host/mpfr-4.1.0

Applying ./patches/001-only_src.patch using plaintext:
patching file Makefile.am
patching file Makefile.in

Applying ./patches/002-Fix-mpfr_custom_get_kind-macro-bug.patch using plaintext:
patching file src/mpfr.h
Hunk #1 FAILED at 1027.
1 out of 1 hunk FAILED -- saving rejects to file src/mpfr.h.rej
patching file tests/tstckintc.c
Hunk #1 succeeded at 178 (offset -117 lines).
Hunk #2 FAILED at 382.
Hunk #3 FAILED at 407.
2 out of 3 hunks FAILED -- saving rejects to file tests/tstckintc.c.rej
Patch failed!  Please fix ./patches/002-Fix-mpfr_custom_get_kind-macro-bug.patch!
make[3]: *** [Makefile:38: /home/lhj/lede/build_dir/host/mpfr-4.1.0/.preparedeeab2b534858ab1c6cfa5061243de2b7_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/lhj/lede/tools/mpfr'
time: tools/mpfr/compile#0.14#0.03#0.15
    ERROR: tools/mpfr failed to build.
make[2]: *** [tools/Makefile:174: tools/mpfr/compile] Error 1
make[2]: Leaving directory '/home/lhj/lede'
make[1]: *** [tools/Makefile:170: /home/lhj/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnyynynyyyyyyyyyyyyyyyyynynnyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/lhj/lede'
make: *** [/home/lhj/lede/include/toplevel.mk:230：world] 错误 2
lhj@debian11:~/lede$

lovehackintosh, 2022-12-11:
`make tools/mpfr/{clean,download,compile} V=s`
002-Fix-mpfr_custom_get_kind-macro-bug.patch针对4.1.1版本,在4.1.0应用会报错

liuhejun2020, 2022-12-12:
怎么解决呢？是不是把编译环境升级成Debian11.5能解决呢？

closed 2022-12-12 by aiamadeus