coolsnowwolf/lede #10591: 更新12月10日的dnsmasq补丁会导致ssr中的chinadns-ng异常

wybb, 2022-12-11:
更新补丁后，启用chinadns-ng将导致系统的dns功能不能正常解析，禁用chinadns或回滚到之前的dnsmasq版本都是正常的

ZGNCZMY, 2022-12-11:
我是遇到了这个
Sun Dec 11 22:08:52 2022 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-adbyby.d: No such file or directory
Sun Dec 11 22:08:52 2022 daemon.crit dnsmasq[1]: FAILED to start up

minlang112, 2022-12-11:
请移步https://github.com/fw876/helloworld提交

starxu1971, 2022-12-12:
这个不是SSR的问题是dnsmasq更新后产生的问题且不是一个软件只要涉及DHCP或者防火墙操作就会出现问题。

wybb, 2022-12-12:
> 请移步 https://github.com/fw876/helloworld 提交

是dnsmasq的问题，不是ssr的，ssr这期间代码并没有更新

wybb, 2022-12-13:
root@OpenWrt:~# nslookup www.baidu.com
Server:         127.0.0.1
Address:        127.0.0.1:53

**_** server can't find www.baidu.com: REFUSED_**

Non-authoritative answer:
www.baidu.com   canonical name = www.a.shifen.com
Name:   www.a.shifen.com
Address: 180.101.49.13
Name:   www.a.shifen.com
Address: 180.101.49.14

jxph1123, 2022-12-13:
不更新这个dnsmasq不就行了

closed 2022-12-14 by coolsnowwolf