coolsnowwolf/lede #10593: 编译5.10内核固件报错

bi4tms, 2022-12-11:
make clean后拉取最新源码编译5.10内核x86固件报错，具体日志如下：
  CC      drivers/hv/hv_balloon.o
  AR      drivers/hv/built-in.a
  AR      drivers/perf/built-in.a
  CC      drivers/ras/ras.o
  CC      drivers/ras/debugfs.o
  AR      drivers/ras/built-in.a
  AR      drivers/hwtracing/intel_th/built-in.a
  CC [M]  drivers/pps/pps.o
  CC [M]  drivers/pps/kapi.o
  CC [M]  drivers/pps/sysfs.o
  LD [M]  drivers/pps/pps_core.o
  AR      drivers/built-in.a
  AR      sound/built-in.a
  CC      net/core/sock.o
  CC      net/core/request_sock.o
  CC      net/core/skbuff.o
  CC      net/core/datagram.o
  CC      net/core/stream.o
  CC      net/core/scm.o
  CC      net/core/gen_stats.o
  CC      net/core/gen_estimator.o
  CC      net/core/net_namespace.o
  CC      net/core/secure_seq.o
  CC      net/core/flow_dissector.o
  CC      net/core/sysctl_net_core.o
  CC      net/core/dev.o
/home/x86.5.10/lede/tmp/ccsTVEjV.s: Assembler messages:
/home/x86.5.10/lede/tmp/ccsTVEjV.s:1733: Error: symbol `__kstrtab_dev_set_threaded' is already defined
/home/x86.5.10/lede/tmp/ccsTVEjV.s:1735: Error: symbol `__kstrtabns_dev_set_threaded' is already defined
/home/x86.5.10/lede/tmp/ccsTVEjV.s:1741: Error: symbol `__ksymtab_dev_set_threaded' is already defined
make[7]: *** [scripts/Makefile.build:286: net/core/dev.o] Error 1
make[6]: *** [scripts/Makefile.build:503: net/core] Error 2
make[5]: *** [Makefile:1837: net] Error 2
make[5]: Leaving directory '/home/x86.5.10/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.158'
make[4]: *** [Makefile:29: /home/x86.5.10/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.158/.modules] Error 2
make[4]: Leaving directory '/home/x86.5.10/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/x86.5.10/lede/target/linux'
time: target/linux/compile#3454.23#1320.09#5450.14
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/x86.5.10/lede'
make[1]: *** [target/Makefile:23: /home/x86.5.10/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/x86.5.10/lede'
make: *** [/home/x86.5.10/lede/include/toplevel.mk:231: world] Error 2

lovehackintosh, 2022-12-12:
`git revert 177dddfaf2a72328f5911da70643ebf9813d5c64`

lovehackintosh, 2022-12-12:
https://github.com/coolsnowwolf/lede/blob/master/target/linux/generic/hack-5.10/993-export-dev-set-dev-symbol.patch
https://github.com/coolsnowwolf/lede/blob/master/target/linux/generic/backport-5.10/605-v5.12-net-export-dev_set_threaded-symbol.patch
这两个patch相同

closed 2022-12-12 by aiamadeus