coolsnowwolf/lede #10595: 最新拉取的源码编译报错，大佬们帮忙看一下

Z139724, 2022-12-12:
1反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
最新拉取的源码编译报错

### (2) 路由器型号和固件版本
N1
### (3) 详细日志
zhang@zhang:~/lede$ make -j1 V=s
make[2]: Entering directory '/home/zhang/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/zhang/lede/scripts/config'
make[1]: Entering directory '/home/zhang/lede'
make[2]: Entering directory '/home/zhang/lede'
make[3]: Entering directory '/home/zhang/lede/target/linux'
make[4]: Entering directory '/home/zhang/lede/target/linux/armvirt'
rm -f /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/localversion
/home/zhang/lede/scripts/kconfig.pl  + + /home/zhang/lede/target/linux/generic/config-5.10 /home/zhang/lede/target/linux/armvirt/config-5.10 /home/zhang/lede/target/linux/armvirt/64/config-5.10 > /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/zhang/lede/.config >> /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.target
/home/zhang/lede/scripts/package-metadata.pl kconfig /home/zhang/lede/tmp/.packageinfo /home/zhang/lede/.config 5.10 > /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.override
/home/zhang/lede/scripts/kconfig.pl 'm+' '+' /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.target /dev/null /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.override > /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set
mv /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.old
grep -v INITRAMFS /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.old > /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set
rm -rf /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/modules
cmp -s /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.prev || { cp /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config; cp /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.prev; }
export MAKEFLAGS= ; [ -d /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/user_headers ] || make -C /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158 KCFLAGS="-fmacro-prefix-map=/home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/zhang/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Dec 11 16:22:04 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/zhang/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.158 CC="aarch64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/user_headers headers_install
grep '=[ym]' /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.config.set | LC_ALL=C sort | /home/zhang/lede/staging_dir/host/bin/mkhash md5 > /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.vermagic
touch /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.configured
rm -f /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/vmlinux /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/System.map
make -C /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158 KCFLAGS="-fmacro-prefix-map=/home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/zhang/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Dec 11 16:22:04 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/zhang/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.158 CC="aarch64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158'
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:486:warning: multi-line strings not supported
.config:4448:warning: symbol value 'm' invalid for NF_CT_PROTO_GRE
#
# configuration written to .config
#
make[5]: Leaving directory '/home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158'
make -C /home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158 KCFLAGS="-fmacro-prefix-map=/home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/zhang/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Dec 11 16:22:04 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/zhang/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.158 CC="aarch64-openwrt-linux-musl-gcc" Image modules
make[5]: Entering directory '/home/zhang/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158'
  SYNC    include/config/auto.conf.cmd
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:486:warning: multi-line strings not supported
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      init/do_mounts.o
  CC      init/init_task.o
  AR      init/built-in.a
  CC      arch/arm64/net/bpf_jit_comp.o
  AR      arch/arm64/net/built-in.a
  AR      arch/arm64/built-in.a
  CC      kernel/fork.o
  CC      kernel/exit.o
  CC      kernel/sysctl.o
  CC      kernel/pid.o
  CC      kernel/extable.o
  CC      kernel/nsproxy.o
[... 140 lines cut ...]
  CC      net/core/gen_estimator.o
  CC      net/core/net_namespace.o
  CC      net/core/secure_seq.o
  CC      net/core/flow_dissector.o
  CC      net/core/sysctl_net_core.o
  CC      net/core/dev.o
/home/zhang/lede/tmp/ccqbZuZh.s: Assembler messages:
/home/zhang/lede/tmp/ccqbZuZh.s:1719: Error: symbol `__kstrtab_dev_set_threaded' is already defined
/home/zhang/lede/tmp/ccqbZuZh.s:1721: Error: symbol `__kstrtabns_dev_set_threaded' is already defined
/home/zhang/lede/tmp/ccqbZuZh.s:1727: Error: symbol `__ksymtab_dev_set_threaded' is already defined
[... cut ...]
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/zhang/lede/target/linux'
time: target/linux/compile#92.11#9.47#101.61
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/zhang/lede'
make[1]: *** [target/Makefile:23: /home/zhang/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/zhang/lede'
make: *** [/home/zhang/lede/include/toplevel.mk:231：world] 错误 2

Karmylr, 2022-12-12:
n1没有openwrt吧

Z139724, 2022-12-12:
有的呀

Z139724, 2022-12-12:
编译完成了，谢谢大佬

closed 2022-12-12 by aiamadeus