coolsnowwolf/lede #10596: Multi Stream Daemon Lite编译中找不到

Nothend, 2022-12-12:
大雕 是不是 把 Multi Stream Daemon Lite又去掉了？周末看到大雕新开发的功能，今天想加进去，结果编译的时候找不到。
默认config未设置状态：
[image]

luci中也没有

[image]

需要添加什么package吗

Nothend, 2022-12-12:
[image]

大雕是把这个功能 下架了吗？

sdf8057, 2022-12-12:
没开源

Nothend, 2022-12-12:
> 没开源

呃。。可惜了。 谢谢楼上！

WYC-2020, 2022-12-13:
难道不是这个https://github.com/rozhuk-im/msd_lite.git

Nothend, 2022-12-13:
> 难道不是这个https://github.com/rozhuk-im/msd_lite.git

牛逼啊 大佬，这个没有luci界面吧，大雕应该是适配了

WYC-2020, 2022-12-13:
luci 随便参考一个写下就是了啊

coolsnowwolf, 2022-12-13:
[image]

WYC-2020, 2022-12-13:
@Nothend https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite 这里 luci 自己根据自己的需求修改即可，或者直接用

Nothend, 2022-12-13:
> @Nothend https://github.com/syb999/openwrt-19.07.1/tree/master/package/network/services/msd_lite 这里 luci 自己根据自己的需求修改即可，或者直接用

谢谢大佬，我来试试，感谢！

WYC-2020, 2022-12-14:
那个版本我看了 没有汉化，并且布局的话 感觉不是很合理，参考l大那个截图 我仿了一个
[image]
[image]

Nothend, 2022-12-15:
> 那个版本我看了 没有汉化，并且布局的话 感觉不是很合理，参考l大那个截图 我仿了一个 [image] [image]

牛啊，大佬，目前你这个运行良好吗？怎么把你这个加入编译中呢？我看大雕的有人反应权限问题，启动不了
[image]

closed 2022-12-25 by aiamadeus