coolsnowwolf/lede #10597: 拨号偶尔会断开

Karmylr, 2022-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：拨号偶尔会断开

### (2) 路由器型号和固件版本

A：ESXi 8.0.0, j4125, i210, wan口网卡直通,OpenWrt R22.12.1 / LuCI Master (git-22.339.26541-ba1f838),6.0.12

### (3) 详细日志

A：
Mon Dec 12 14:18:11 2022 daemon.notice netifd: Network device 'eth1' link is down
Mon Dec 12 14:18:11 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Dec 12 14:18:11 2022 kern.info kernel: [ 1551.719925] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Down
Mon Dec 12 14:18:11 2022 daemon.info pppd[4891]: Terminating on signal 15
Mon Dec 12 14:18:11 2022 daemon.info pppd[4891]: Connect time 25.1 minutes.
Mon Dec 12 14:18:11 2022 daemon.info pppd[4891]: Sent 13979154 bytes, received 361707695 bytes.
Mon Dec 12 14:18:11 2022 daemon.notice netifd: Network device 'pppoe-wan' link is down
Mon Dec 12 14:18:11 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Mon Dec 12 14:18:12 2022 daemon.notice netifd: Network device 'eth1' link is up
Mon Dec 12 14:18:12 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Mon Dec 12 14:18:12 2022 kern.info kernel: [ 1553.330046] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Mon Dec 12 14:18:14 2022 daemon.notice pppd[4891]: Connection terminated.
Mon Dec 12 14:18:14 2022 daemon.info pppd[4891]: Connect time 25.1 minutes.
Mon Dec 12 14:18:14 2022 daemon.info pppd[4891]: Sent 13979154 bytes, received 361707695 bytes.
Mon Dec 12 14:18:14 2022 daemon.info pppd[4891]: Exit.
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' is now down
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' is disabled
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' is enabled
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' is setting up now
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Network device 'eth1' link is down
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Dec 12 14:18:14 2022 kern.info kernel: [ 1554.800313] 8021q: adding VLAN 0 to HW filter on device eth1
Mon Dec 12 14:18:14 2022 daemon.notice netifd: wan (30698): uci: Entry not found
Mon Dec 12 14:18:14 2022 daemon.notice netifd: wan (30698): sh: out of range
Mon Dec 12 14:18:14 2022 daemon.notice netifd: wan (30698): Command failed: ubus call network.interface notify_proto { "action": 1, "command": [ "\/usr\/sbin\/pppd", "nodetach", "ipparam", "wan", "ifname", "pppoe-wan", "nodefaultroute", "usepeerdns", "maxfail", "1", "user", "059107615516", "password", "615516", "ip-up-script", "\/lib\/netifd\/ppp-up", "ip-down-script", "\/lib\/netifd\/ppp-down", "mtu", "1492", "mru", "1492", "plugin", "pppoe.so", "nic-eth1" ], "interface": "wan" } (Permission denied)
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' is now down
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' is disabled
Mon Dec 12 14:18:14 2022 daemon.notice netifd: Interface 'wan' is enabled
Mon Dec 12 14:18:14 2022 kern.info kernel: [ 1554.880288] 8021q: adding VLAN 0 to HW filter on device eth1
Mon Dec 12 14:18:16 2022 daemon.notice netifd: Network device 'eth1' link is up
Mon Dec 12 14:18:16 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Mon Dec 12 14:18:16 2022 daemon.notice netifd: Interface 'wan' is setting up now
Mon Dec 12 14:18:16 2022 kern.info kernel: [ 1556.559771] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Mon Dec 12 14:18:16 2022 daemon.notice netifd: wan (30823): uci: Entry not found
Mon Dec 12 14:18:16 2022 daemon.notice netifd: wan (30823): sh: out of range
Mon Dec 12 14:18:16 2022 daemon.info pppd[30830]: Plugin pppoe.so loaded.
Mon Dec 12 14:18:16 2022 daemon.info pppd[30830]: PPPoE plugin from pppd 2.4.9
Mon Dec 12 14:18:16 2022 daemon.notice pppd[30830]: pppd 2.4.9 started by root, uid 0
Mon Dec 12 14:18:21 2022 daemon.info pppd[30830]: PPP session is 24689
Mon Dec 12 14:18:21 2022 daemon.warn pppd[30830]: Connected to cc:cc:81:eb:7c:a7 via interface eth1
Mon Dec 12 14:18:21 2022 daemon.info pppd[30830]: Renamed interface ppp0 to pppoe-wan
Mon Dec 12 14:18:21 2022 daemon.info pppd[30830]: Using interface pppoe-wan
Mon Dec 12 14:18:21 2022 daemon.notice pppd[30830]: Connect: pppoe-wan <--> eth1
Mon Dec 12 14:18:21 2022 daemon.info pppd[30830]: syncppp not active
Mon Dec 12 14:18:21 2022 kern.info kernel: [ 1561.588892] pppoe-wan: renamed from ppp0
Mon Dec 12 14:18:21 2022 daemon.info pppd[30830]: Remote message: Authentication success,Welcome!
Mon Dec 12 14:18:21 2022 daemon.notice pppd[30830]: PAP authentication succeeded
Mon Dec 12 14:18:21 2022 daemon.notice pppd[30830]: peer from calling number CC:CC:81:EB:7C:A7 authorized
Mon Dec 12 14:18:21 2022 daemon.notice pppd[30830]: local  LL address fe80::4c6c:74b6:b428:3243
Mon Dec 12 14:18:21 2022 daemon.notice pppd[30830]: remote LL address fe80::cecc:81ff:feeb:7ca7
[... 42 lines cut ...]
Mon Dec 12 14:22:31 2022 daemon.notice netifd: wan (2977): sh: out of range
Mon Dec 12 14:22:31 2022 daemon.info pppd[2984]: Plugin pppoe.so loaded.
Mon Dec 12 14:22:31 2022 daemon.info pppd[2984]: PPPoE plugin from pppd 2.4.9
Mon Dec 12 14:22:31 2022 daemon.notice pppd[2984]: pppd 2.4.9 started by root, uid 0
Mon Dec 12 14:22:36 2022 daemon.info pppd[2984]: PPP session is 9084
Mon Dec 12 14:22:36 2022 daemon.warn pppd[2984]: Connected to cc:cc:81:eb:7c:a7 via interface eth1
Mon Dec 12 14:22:36 2022 daemon.info pppd[2984]: Renamed interface ppp0 to pppoe-wan
Mon Dec 12 14:22:36 2022 daemon.info pppd[2984]: Using interface pppoe-wan
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: Connect: pppoe-wan <--> eth1
Mon Dec 12 14:22:36 2022 daemon.info pppd[2984]: syncppp not active
Mon Dec 12 14:22:36 2022 kern.info kernel: [ 1816.883862] pppoe-wan: renamed from ppp0
Mon Dec 12 14:22:36 2022 daemon.info pppd[2984]: Remote message: Authentication success,Welcome!
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: PAP authentication succeeded
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: peer from calling number CC:CC:81:EB:7C:A7 authorized
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: local  LL address fe80::6cf1:319e:2e64:d86b
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: remote LL address fe80::cecc:81ff:feeb:7ca7
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: local  IP address 175.42.16.226
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: remote IP address 175.42.16.1
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: primary   DNS address 58.22.96.66
Mon Dec 12 14:22:36 2022 daemon.notice pppd[2984]: secondary DNS address 218.104.128.106
Mon Dec 12 14:22:36 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Mon Dec 12 14:22:36 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Mon Dec 12 14:22:36 2022 daemon.notice netifd: Interface 'wan' is now up
Mon Dec 12 14:22:36 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Mon Dec 12 15:29:49 2022 user.notice sysfixtime: saved 'Mon Dec 12 15:29:49 CST 2022' to /dev/rtc0

GreatMichaelLee, 2022-12-13:
感觉是不是你内存有点小？

Karmylr, 2022-12-13:
> 感觉是不是你内存有点小？

我给大点试试

Karmylr, 2023-05-13:
给了1GB内存还是会偶尔断开（WAN口上下行流量均为0）。日志：
```
Sat May 13 08:50:01 2023 daemon.warn pppd[2250]: Timeout waiting for PADO packets
Sat May 13 08:50:01 2023 daemon.err pppd[2250]: Unable to complete PPPoE Discovery
Sat May 13 08:50:01 2023 daemon.info pppd[2250]: Exit.
Sat May 13 08:50:01 2023 daemon.notice netifd: Interface 'wan' is now down
Sat May 13 08:50:01 2023 daemon.notice netifd: Interface 'wan' is disabled
Sat May 13 08:50:01 2023 daemon.notice netifd: Interface 'wan' is enabled
Sat May 13 08:50:01 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sat May 13 08:50:01 2023 kern.info kernel: [   32.922160] vmxnet3 0000:02:05.0 eth1: intr type 3, mode 0, 5 vectors allocated
Sat May 13 08:50:01 2023 kern.info kernel: [   32.923675] vmxnet3 0000:02:05.0 eth1: NIC Link is Up 10000 Mbps
Sat May 13 08:50:01 2023 kern.info kernel: [   32.924824] 8021q: adding VLAN 0 to HW filter on device eth1
Sat May 13 08:50:01 2023 daemon.notice netifd: wan (5079): uci: Entry not found
Sat May 13 08:50:01 2023 daemon.notice netifd: wan (5079): sh: out of range
Sat May 13 08:50:01 2023 daemon.info pppd[5086]: Plugin pppoe.so loaded.
Sat May 13 08:50:01 2023 daemon.info pppd[5086]: PPPoE plugin from pppd 2.4.9
Sat May 13 08:50:02 2023 daemon.notice pppd[5086]: pppd 2.4.9 started by root, uid 0
Sat May 13 08:50:09 2023 daemon.err pppd[5086]: select (waitForPADO): Interrupted system call
Sat May 13 08:50:09 2023 daemon.warn pppd[5086]: Timeout waiting for PADO packets
Sat May 13 08:50:09 2023 daemon.err pppd[5086]: Unable to complete PPPoE Discovery
Sat May 13 08:50:09 2023 daemon.info pppd[5086]: Exit.
```

GreatMichaelLee, 2023-05-13:
你这PADO包都超时了，拨号间隔太短，被BRAS临时ban了吧？

Karmylr, 2023-05-13:
用硬路由立刻就能拨上去了...

Karmylr, 2023-05-13:
还是会断开
```
Sat May 13 19:38:38 2023 daemon.info pppd[24809]: IPCP terminated by peer
Sat May 13 19:38:38 2023 daemon.info pppd[24809]: Connect time 11.5 minutes.
Sat May 13 19:38:38 2023 daemon.info pppd[24809]: Sent 16343415 bytes, received 158589509 bytes.
Sat May 13 19:38:38 2023 daemon.notice netifd: Interface 'wan' has lost the connection
Sat May 13 19:38:38 2023 daemon.notice netifd: Interface 'wan_6' is now down
Sat May 13 19:38:38 2023 daemon.notice netifd: Interface 'wan_6' is disabled
Sat May 13 19:38:38 2023 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Sat May 13 19:38:38 2023 kern.info kernel: [18177.425763] netifd[1936]: segfault at 7fbf7b6c1278 ip 000000000040d886 sp 00007fff43e88050 error 4 in netifd[404000+1b000] likely on CPU 3 (core 0, socket 6)
Sat May 13 19:38:38 2023 kern.info kernel: [18177.427613] Code: 01 75 31 48 8b 55 58 be 43 fa 41 00 bf 02 00 00 00 31 c0 e8 1f 8f ff ff 48 89 ef e8 e6 d9 ff ff c7 85 90 00 00 00 00 00 00 00 <48> 8b 7d 68 5d e9 d7 7e ff ff 5d c3 41 57 41 56 41 55 41 54 49 89
Sat May 13 19:38:43 2023 user.notice : Added device handler type: 8021ad
Sat May 13 19:38:43 2023 user.notice : Added device handler type: 8021q
Sat May 13 19:38:43 2023 user.notice : Added device handler type: macvlan
Sat May 13 19:38:43 2023 user.notice : Added device handler type: veth
Sat May 13 19:38:43 2023 user.notice : Added device handler type: bridge
Sat May 13 19:38:43 2023 user.notice : Added device handler type: Network device
Sat May 13 19:38:43 2023 user.notice : Added device handler type: tunnel
Sat May 13 19:38:44 2023 daemon.notice pppd[24809]: Modem hangup
Sat May 13 19:38:44 2023 daemon.notice pppd[24809]: Connection terminated.
Sat May 13 19:38:44 2023 daemon.info pppd[24809]: Connect time 11.6 minutes.
Sat May 13 19:38:44 2023 daemon.info pppd[24809]: Sent 16343419 bytes, received 158803167 bytes.
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'lan' is enabled
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'lan' is setting up now
Sat May 13 19:38:44 2023 kern.info kernel: [18182.552752] vmxnet3 0000:02:02.0 eth0: intr type 3, mode 0, 5 vectors allocated
Sat May 13 19:38:44 2023 kern.info kernel: [18182.554342] vmxnet3 0000:02:02.0 eth0: NIC Link is Up 10000 Mbps
Sat May 13 19:38:44 2023 kern.info kernel: [18182.555362] 8021q: adding VLAN 0 to HW filter on device eth0
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'lan' is now up
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'loopback' is enabled
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'loopback' is setting up now
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'loopback' is now up
Sat May 13 19:38:44 2023 kern.info kernel: [18182.563323] vmxnet3 0000:02:05.0 eth1: intr type 3, mode 0, 5 vectors allocated
Sat May 13 19:38:44 2023 kern.info kernel: [18182.565005] vmxnet3 0000:02:05.0 eth1: NIC Link is Up 10000 Mbps
Sat May 13 19:38:44 2023 kern.info kernel: [18182.566301] 8021q: adding VLAN 0 to HW filter on device eth1
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'wan' is enabled
Sat May 13 19:38:44 2023 daemon.notice netifd: Network device 'lo' link is up
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Sat May 13 19:38:44 2023 daemon.notice netifd: Network device 'eth0' link is up
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Sat May 13 19:38:44 2023 daemon.notice netifd: Network device 'eth1' link is up
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Sat May 13 19:38:44 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sat May 13 19:38:44 2023 user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Sat May 13 19:38:44 2023 daemon.notice netifd: wan (30089): uci: Entry not found
Sat May 13 19:38:44 2023 daemon.notice netifd: wan (30089): sh: out of range
Sat May 13 19:38:44 2023 daemon.info pppd[30117]: Plugin pppoe.so loaded.
Sat May 13 19:38:44 2023 daemon.info pppd[30117]: PPPoE plugin from pppd 2.4.9
Sat May 13 19:38:44 2023 daemon.notice pppd[30117]: pppd 2.4.9 started by root, uid 0
Sat May 13 19:38:45 2023 kern.info kernel: [18183.607030] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sat May 13 19:38:45 2023 daemon.err procd: Got unexpected signal 1
Sat May 13 19:38:49 2023 daemon.info pppd[24809]: Exit.
Sat May 13 19:38:49 2023 daemon.info pppd[30117]: PPP session is 6224
Sat May 13 19:38:49 2023 daemon.warn pppd[30117]: Connected to cc:cc:81:eb:7c:a7 via interface eth1
Sat May 13 19:38:49 2023 kern.info kernel: [18187.767109] pppoe-wan: renamed from ppp0
Sat May 13 19:38:49 2023 daemon.info pppd[30117]: Renamed interface ppp0 to pppoe-wan
Sat May 13 19:38:49 2023 daemon.info pppd[30117]: Using interface pppoe-wan
Sat May 13 19:38:49 2023 daemon.notice pppd[30117]: Connect: pppoe-wan <--> eth1
Sat May 13 19:38:51 2023 daemon.err netdata[2766]: Cannot refresh interface pppoe-wan carrier state by reading '/sys/class/net/pppoe-wan/carrier'. Stop updating it.
Sat May 13 19:38:51 2023 daemon.info netdata[2766]: RRDSET: chart name 'net_operstate.pppoe_wan' on host 'OpenWrt' already exists.
Sat May 13 19:38:51 2023 daemon.info netdata[2766]: RRDSET: chart name 'net_mtu.pppoe_wan' on host 'OpenWrt' already exists.
Sat May 13 19:38:52 2023 daemon.info pppd[30117]: syncppp not active
[... cut ...]
Sat May 13 19:38:52 2023 daemon.notice netifd: Interface 'wan_6' is now up
Sat May 13 19:38:52 2023 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sat May 13 19:38:53 2023 daemon.info netdata[2766]: RRDSET: chart name 'net.pppoe_wan' on host 'OpenWrt' already exists.
Sat May 13 19:38:53 2023 daemon.info netdata[2766]: RRDSET: chart name 'net_packets.pppoe_wan' on host 'OpenWrt' already exists.
Sat May 13 19:38:57 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
```

Karmylr, 2023-05-27:
离谱，换了一个网口就稳定了

closed 2023-05-27 by Karmylr