coolsnowwolf/lede #10598: 刚刚12月12号上午重装的Debian11,全部重新开始，两边出一样的错误，似乎报源码里的一个符号已经被定义两次

liuhejun2020, 2022-12-12:
N1盒子底包：
  
LD [M]  sound/core/snd-compress.o
  CC [M]  sound/drivers/virmidi.o
  LD [M]  sound/drivers/snd-virmidi.o
  AR      sound/built-in.a
  CC [M]  sound/sound_core.o
  LD [M]  sound/soundcore.o
  CC      net/core/sock.o
  CC      net/core/request_sock.o
  CC      net/core/skbuff.o
  CC      net/core/datagram.o
  CC      net/core/stream.o
  CC      net/core/scm.o
  CC      net/core/gen_stats.o
  CC      net/core/gen_estimator.o
  CC      net/core/net_namespace.o
  CC      net/core/secure_seq.o
  CC      net/core/flow_dissector.o
  CC      net/core/sysctl_net_core.o
  CC      net/core/dev.o
/home/lhj/lede/tmp/cc0YYlOR.s: Assembler messages:
/home/lhj/lede/tmp/cc0YYlOR.s:1749: Error: symbol `__kstrtab_dev_set_threaded' is already defined
/home/lhj/lede/tmp/cc0YYlOR.s:1751: Error: symbol `__kstrtabns_dev_set_threaded' is already defined
/home/lhj/lede/tmp/cc0YYlOR.s:1757: Error: symbol `__ksymtab_dev_set_threaded' is already defined
make[7]: *** [scripts/Makefile.build:286: net/core/dev.o] Error 1
make[6]: *** [scripts/Makefile.build:503: net/core] Error 2
make[5]: *** [Makefile:1837: net] Error 2
make[5]: Leaving directory '/home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158'
make[4]: *** [Makefile:21: /home/lhj/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/linux-5.10.158/.modules] Error 2
make[4]: Leaving directory '/home/lhj/lede/target/linux/armvirt'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/lhj/lede/target/linux'
time: target/linux/compile#688.82#64.16#754.05
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/lhj/lede'
make[1]: *** [target/Makefile:23: /home/lhj/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lhj/lede'
make: *** [/home/lhj/lede/include/toplevel.mk:231：world] 错误 2
lhj@debian11:~/lede$ 

这是我的feed.conf:

src-git packages https://github.com/coolsnowwolf/packages
src-git helloworld https://github.com/fw876/helloworld
src-git luci https://github.com/coolsnowwolf/luci
#src-git routing https://git.openwrt.org/feed/routing.git
src-git routing https://github.com/coolsnowwolf/routing
src-git telephony https://git.openwrt.org/feed/telephony.git
#src-git oui https://github.com/zhaojh329/oui.git
src-git nas https://github.com/linkease/nas-packages.git;master
src-git nas_luci https://github.com/linkease/nas-packages-luci.git;main
src-git istore https://github.com/linkease/istore
src-git freifunk https://github.com/freifunk/openwrt-packages.git
src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small
#src-git OpenClass https://github.com/vernesong/OpenClash.git
#src-git passwall https://github.com/xiaorouji/openwrt-passwall.git
#src-git passwall2 https://github.com/xiaorouji/openwrt-passwall2.git
#src-git helloworld1 https://github.com/sbwml/openwrt_helloworld
#src-git AdGuardHome https://github.com/AdguardTeam/AdGuardHome
#src-git bypass https://github.com/kiddin9/openwrt-bypass.git
src-git kiddin https://github.com/kiddin9/openwrt-packages
#src-git maxminddb https://github.com/jerrykuku/lua-maxminddb.git  #git lua-maxminddb 依赖
#src-git vssr clone https://github.com/jerrykuku/luci-app-vssr.git

liuhejun2020, 2022-12-12:
编译两遍同一个错误，第一次错了删除了lede目录重来，又是一样

lovehackintosh, 2022-12-12:
`git pull`拉取最新源码试试

liuhejun2020, 2022-12-12:
这就试试反馈，结果要一个多小时到那里呢

closed 2022-12-12 by aiamadeus