coolsnowwolf/lede #10599: BPI-r3 编译问题

lanboeh5188, 2022-12-12:
rm "/home/ub/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.stripmeta"
bash: 8: Bad file descriptor
dd if=/home/ub/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/ub/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=44M conv=sync
dd: failed to open '/home/ub/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz': No such file or directory
make[5]: *** [Makefile:31: /home/ub/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz] Error 1
make[5]: Leaving directory '/home/ub/lede/target/linux/mediatek/image'
make[4]: *** [Makefile:17: install] Error 2
make[4]: Leaving directory '/home/ub/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/ub/lede/target/linux'
time: target/linux/install#117.35#8.49#109.13
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/ub/lede'
make[1]: *** [target/Makefile:24: /home/ub/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ub/lede'
make: *** [/home/ub/lede/include/toplevel.mk:231: world] Error 2
BPI-r3 编译了好几遍 都是这样的错误 这是最后要生成镜像的时候报错吗？如何解决呢

dd-bpir3, 2023-03-17:
解决了吗？我这个报错不是能生成sd镜像文件

coolsnowwolf, 2026-09-05:
当前 mediatek image Makefile 已不再使用 issue 日志中的 `bs=44M`/`stripmeta` 截断步骤，现行流程为统一的 `append-kernel`、`append-rootfs` 和 metadata 生成；该 2022 年 BPI-R3 旧错误不适用于当前源码。

closed 2026-09-05 by coolsnowwolf