coolsnowwolf/lede #10600: kernel 6.1 build fail: zstd_common.ko is missing

GreatMichaelLee, 2022-12-13:
提示缺zstd_common.ko这个内核模块。

nux-firmware-20221012/.pkgdir/i915-firmware/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.i915-firmware_installed
echo "i915-firmware" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/builder/lede_x86/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#0.99#1.37#3.25
make[3]: Entering directory '/home/builder/lede_x86/package/kernel/linux'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-lib-zstd is missing dependencies for the following libraries:
zstd_common.ko
make[3]: *** [modules/lib.mk:145: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-lib-zstd_6.1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#3.41#0.60#3.98
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

xnxy2012, 2022-12-13:
相同的错误，等待大佬解决，缺少zstd_common.ko依赖，mark以下

Michaol, 2022-12-13:
我是kmod-drm-i915出错，缺video.ko。如果去掉kmod-drm-i915驱动，又变成Intel igb出错……不知道有没有相同错误？

SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-drm-i915 is missing dependencies for the following libraries:
video.ko
make[3]: *** [modules/video.mk:1139: /workdir/openwrt/bin/targets/x86/64/packages/kmod-drm-i915_6.1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.63#0.23#2.85
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-12-13:
等会有时间我编译看下

Michaol, 2022-12-13:
> 等会有时间我编译看下

谢谢大佬，我除了SSRP没有添加其他package了。

GreatMichaelLee, 2022-12-13:
> 等会有时间我编译看下

老板，每次都等你出手啊....^^

softtip, 2022-12-13:
同样的问题，只要用到kmod-lib-zstd依赖的，都编译不过去

softtip, 2022-12-13:
比如 btrfs,比如n2n

coolsnowwolf, 2022-12-13:
> 我是kmod-drm-i915出错，缺video.ko。如果去掉kmod-drm-i915驱动，又变成Intel igb出错……不知道有没有相同错误？
>
> SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR [...]

因为你没有 rm -rf tmp ，用的还是之前的缓存

WYC-2020, 2022-12-13:
@GreatMichaelLee @Michaol 
[image]
这样修改 亲测有效

softtip, 2022-12-13:
> 

感谢，这次可以了！

softtip, 2022-12-13:
> @GreatMichaelLee @Michaol [image] 这样修改 亲测有效

感谢，修改了这两个文件，现在可以了
但是新问题又来了，kmod-igb-intel编译不过去，提示是源代码有error问题

coolsnowwolf, 2022-12-13:
kmod-igb-intel 不再需要

softtip, 2022-12-13:
> kmod-igb-intel 不再需要

谢谢大雕

Michaol, 2022-12-13:
> 因为你没有 rm -rf tmp ，用的还是之前的缓存

不是啊，大雕。云编译和本地编译同错。本地编译我懒，直接写成bash，每次都清空tmp的。

# /bin/bash
rm -rf tmp && rm -rf dl/go-mod-cache
sudo sh -c "apt update && apt upgrade -y && apt clean && apt autoremove"
# rm -rf feeds/luci/themes/luci-theme-argon && git clone -b 18.06 https://github.com/jerrykuku/luci-theme-argon.git feeds/luci/themes/luci-theme-argon
# rm -rf luci-app-ttnode && git clone https://github.com/jerrykuku/luci-app-ttnode.git
# rm -rf package/luci-argon-config && git clone https://github.com/jerrykuku/luci-app-argon-config.git package/luci-argon-config
git pull
# cd package/openclash && git pull && cd ../..
./scripts/feeds update -a && ./scripts/feeds install -a
make defconfig
make -j8 download
find dl -size -1024c -exec ls -l {} \;
find dl -size -1024c -exec rm -f {} \;
make -j$(($(nproc) + 1)) || make -j1 V=s

xiangfeidexiaohuo, 2022-12-13:
zstd 等待修复 哈哈

coolsnowwolf, 2022-12-13:
fixed

sss132919, 2022-12-13:
同样的zstd问题，我还以为只有我是这个问题，是不是zstd还没适配6.1的

closed 2022-12-13 by coolsnowwolf