coolsnowwolf/lede #10602: smbd 进程老是打没用的日志

GreatMichaelLee, 2022-12-13:
以前出过这个问题，后来莫名消失了，最近又出来了，不知道是什么情况，怎么关掉？

Mon Dec 12 11:45:32 2022 daemon.err smbd[10181]: [2022/12/12 11:45:32.603795,  0] ../../source3/printing/print_standard.c:71(std_pcap_cache_reload)
Mon Dec 12 11:45:32 2022 daemon.err smbd[10181]:   Unable to open printcap file /etc/printcap for read!
Mon Dec 12 11:58:02 2022 daemon.err smbd[10181]: [2022/12/12 11:58:02.703791,  0] ../../source3/printing/print_standard.c:71(std_pcap_cache_reload)
Mon Dec 12 11:58:02 2022 daemon.err smbd[10181]:   Unable to open printcap file /etc/printcap for read!

ecrasy, 2022-12-13:
set following lines to your samba configuration file “/etc/samba/smb.conf”:
  
load printers = no  
printcap name = /dev/null  
  
You might have to restart your samba server

smdx, 2022-12-13:
我也有smbd的错误日志

pve lxc运行op
Tue Dec 13 12:09:26 2022 daemon.err smbd[2420]: [2022/12/13 12:09:26.156052,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Tue Dec 13 12:09:26 2022 daemon.err smbd[2420]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Tue Dec 13 12:09:26 2022 daemon.err smbd[2420]: [2022/12/13 12:09:26.156294,  0] ../../source3/smbd/server.c:1082(smbd_open_one_socket)
Tue Dec 13 12:09:26 2022 daemon.err smbd[2420]:   smbd_open_one_socket: open_socket_in: Address not available
Tue Dec 13 12:09:26 2022 daemon.err smbd[2420]: [2022/12/13 12:09:26.156375,  0] ../../source3/smbd/server.c:1082(smbd_open_one_socket)
Tue Dec 13 12:09:26 2022 daemon.err smbd[2420]:   smbd_open_one_socket: open_socket_in: Address not available

GreatMichaelLee, 2022-12-14:
> set following lines to your samba configuration file “/etc/samba/smb.conf”:
> 
> load printers = no printcap name = /dev/null
> 
> You might have to restart your samba server

解决了，多谢！

closed 2022-12-14 by GreatMichaelLee