coolsnowwolf/lede #10608: kernel 6.1 build failed: xtables-addsons - cc1: some warnings being treated as errors

GreatMichaelLee, 2022-12-14:
make -f ./scripts/Makefile.build obj=/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/rtsp \
need-builtin= \
need-modorder=1 \

  x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/.xt_CHAOS.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=1024 -fstack-protector -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -fno-var-tracking -femit-struct-debug-baseonly -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves  -DMODULE  -DKBUILD_BASENAME='"xt_CHAOS"' -DKBUILD_MODNAME='"xt_CHAOS"' -D__KBUILD_MODNAME=kmod_xt_CHAOS -c -o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/xt_CHAOS.o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/xt_CHAOS.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --ibt --retpoline --stackval --static-call --uaccess  --link  --module /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/xt_CHAOS.o
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/xt_CHAOS.c: In function 'xt_chaos_total':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/xt_CHAOS.c:70:46: error: implicit declaration of function 'prandom_u32'; did you mean 'prandom_u32_max'? [-Werror=implicit-function-declaration]
   70 |         if (!ret || hotdrop || (unsigned int)prandom_u32() > delude_percentage)
      |                                              ^~~~~~~~~~~
      |                                              prandom_u32_max
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:250: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/xt_CHAOS.o] Error 1
make[7]: *** [Makefile:1992: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions] Error 2
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1'
make[6]: *** [Makefile:461: modules] Error 2
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
make[5]: *** [Makefile:495: all-recursive] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[4]: *** [Makefile:380: all] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[3]: *** [Makefile:177: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#1.43#1.25#3.46
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

GreatMichaelLee, 2022-12-14:
@WYC-2020 求大佬再看下哈 ^^

GreatMichaelLee, 2022-12-14:
> 升到 3.22 或应用上游补丁就好了
> 
> ```
> https://lore.kernel.org/all/20221023032239.808311-1-git@johnthomson.fastmail.com.au/T/#m51fec4e997089475c8c21f2620753b419d89e31a
> ```

不懂怎么弄啊

Beginner-Go, 2022-12-14:
Fixed.

closed 2022-12-14 by coolsnowwolf

xwings, 2023-01-03:
目前master 编译 package/kernel/mac80211 是有问题的 

最直接的方法是把 https://github.com/openwrt/openwrt/tree/master/package/kernel/mac80211 替换过来 问题就解决了