coolsnowwolf/lede #10617: DHCP无法分配IP地址，须手动指定

fxxccp, 2022-12-15:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：用12月15日的源码编译的x86固件全新刷机后DHCP无法分配IP地址，须手动指定。

### (2) 路由器型号和固件版本

A：x86/5.15.82

### (3) 详细日志

A：无

GreatMichaelLee, 2022-12-15:
一样，最新这个dnsmasq版本感觉有问题，dns也查询不了。

ecrasy, 2022-12-15:
建议L大把adbyby去掉
这个app和dnsmasq v2.87不兼容

1054009064, 2022-12-15:
log
```
Thu Dec 15 10:46:12 2022 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Thu Dec 15 10:46:22 2022 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Thu Dec 15 10:46:25 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Thu Dec 15 10:46:25 2022 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Thu Dec 15 10:46:28 2022 daemon.info dnsmasq[1]: started, version 2.87 cachesize 150
Thu Dec 15 10:46:28 2022 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset no-auth no-cryptohash no-DNSSEC loop-detect inotify dumpfile
Thu Dec 15 10:46:28 2022 daemon.info dnsmasq[1]: reading /etc/resolv.conf
Thu Dec 15 10:46:28 2022 daemon.warn dnsmasq[1]: ignoring nameserver 127.0.0.1 - local interface
Thu Dec 15 10:46:28 2022 daemon.warn dnsmasq[1]: ignoring nameserver ::1 - local interface
Thu Dec 15 10:46:28 2022 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Thu Dec 15 10:46:32 2022 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Thu Dec 15 10:46:42 2022 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Thu Dec 15 10:46:52 2022 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Thu Dec 15 10:47:00 2022 daemon.err dnsmasq[1]: failed to send packet: Network unreachable
Thu Dec 15 10:47:01 2022 daemon.err dnsmasq[1]: failed to send packet: Network unreachable
Thu Dec 15 10:47:01 2022 daemon.err dnsmasq[1]: failed to send packet: Network unreachable
Thu Dec 15 10:47:02 2022 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Thu Dec 15 10:47:02 2022 daemon.err dnsmasq[1]: failed to send packet: Network unreachable
Thu Dec 15 10:47:02 2022 daemon.err dnsmasq[1]: failed to send packet: Network unreachable
Thu Dec 15 10:47:02 2022 daemon.err dnsmasq[1]: failed to send packet: Network unreachable
```

coolsnowwolf, 2022-12-15:
fixed

closed 2022-12-15 by coolsnowwolf