coolsnowwolf/lede #10618: samba4网络共享会自动有两个共享目录sda1 sdb1，删除两个目录重开机后又出现，有什么办法删除后不自动恢复？

joedai007, 2022-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：samba4网络共享会自动有两个共享目录sda1 sdb1，删除两个目录重开机后又出现，有什么办法删除后不自动恢复？

B：另外有时删除两个目录后，编辑模板内容变空，不知道怎么回事？

### (2) 路由器型号和固件版本

A：x86 OpenWrt  R22.12.1

### (3) 详细日志

A：

[image]

ecrasy, 2022-12-15:
不要编译autosamba
否则这玩意会自动共享你挂载的block分区

joedai007, 2022-12-15:
> 不要编译autosamba
> 否则这玩意会自动共享你挂载的block分区
谢谢！是不是就算编译samba4,也取消autosamba这一项就可以了？

starxu1971, 2022-12-15:
> > 不要编译 autosamba
> > 否则这玩意会自动共享你挂载的 block 分区
> 
> 谢谢！是不是就算编译 samba4, 也取消 autosamba 这一项就可以了？
没错取消autosanba, 然后选samba4就行。

WYC-2020, 2022-12-15:
去掉那是下下策，最好的改法是
[image]
这样

joedai007, 2022-12-15:
> 去掉那是下下策，最好的改法是 [image] 这样

samba4没有自动共享这项啊

WYC-2020, 2022-12-15:
> > 去掉那是下下策，最好的改法是 [image] 这样
> 
> samba4没有自动共享这项啊

自己修改 几句话的事

joedai007, 2022-12-15:
> 软件包里面直接卸载autosamba

好的，我试试

WYC-2020, 2022-12-15:
> > 去掉那是下下策，最好的改法是 [image] 这样
> 
> samba4没有自动共享这项啊

[image]
这样增加即可

joedai007, 2022-12-15:
> > > 去掉那是下下策，最好的改法是 [image] 这样
> > 
> > 
> > samba4没有自动共享这项啊
> 
> [image] 这样增加即可

谢谢！

closed 2022-12-17 by aiamadeus

joedai007, 2023-04-01:
> > > > 去掉那是下下策，最好的改法是 [image] 这样
> > > 
> > > 
> > > samba4没有自动共享这项啊
> > 
> > 
> > [image] 这样增加即可
> 
> 谢谢！
[image]
大佬，添加的这项是英文的，怎么变成中文，谢谢！

joedai007, 2023-04-01:
> > > > 去掉那是下下策，最好的改法是 [image] 这样
> > > 
> > > 
> > > samba4没有自动共享这项啊
> > 
> > 
> > [image] 这样增加即可
> 
> 谢谢！

[image]
大佬，添加的这项是英文的，怎么改成中文，谢谢！