coolsnowwolf/lede #10621: X86 编译报错: cjdns build failed error1； target-x86_64_musl/stamp/.package_compile] Error 2

LarryLi-cn, 2022-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
X86编译报错
### (1) 具体问题

A：
cc1: note: someone does not honour COPTS correctly, passed 16 times
./util/test/AddrTools_test.c: In function 'AddrTools_test_main':
./util/test/AddrTools_test.c:22:18: error: 'AddrTools_parseIp' reading 40 bytes from a region of size 6 [-Werror=stringop-overread]
   22 |     Assert_true(!AddrTools_parseIp(out, (uint8_t*)"[::1]"));
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./util/test/AddrTools_test.c:22:18: note: referencing argument 2 of type 'const uint8_t *' {aka 'const unsigned char *'}
In file included from ./util/test/AddrTools_test.c:16:
./util/AddrTools.h:47:5: note: in a call to function 'AddrTools_parseIp'
   47 | int AddrTools_parseIp(uint8_t out[16], const uint8_t hexAddr[40]);
      |     ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors

    at error (/workdir/openwrt/build_dir/target-x86_64_musl/cjdns-cjdns-v21/node_build/builder.js:53:15)
    at /workdir/openwrt/build_dir/target-x86_64_musl/cjdns-cjdns-v21/node_build/builder.js:125:22
    at /workdir/openwrt/build_dir/target-x86_64_musl/cjdns-cjdns-v21/node_build/builder.js:95:13
    at ChildProcess.<anonymous> (/workdir/openwrt/build_dir/target-x86_64_musl/cjdns-cjdns-v21/tools/lib/Semaphore.js:7:30)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
make[3]: *** [Makefile:149: /workdir/openwrt/build_dir/target-x86_64_musl/cjdns-cjdns-v21/.built] Error 1
make[2]: *** [package/Makefile:116: package/feeds/routing/cjdns/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/cjdns'
time: package/feeds/routing/cjdns/compile#9.89#2.23#6.52
    ERROR: package/feeds/routing/cjdns failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

### (2) 路由器型号和固件版本

A：J4125

### (3) 详细日志

A：
[11_Compile the firmware.txt](https://github.com/coolsnowwolf/lede/files/10236826/11_Compile.the.firmware.txt)

WYC-2020, 2022-12-16:
这个我刚编译了，有错误但是和你这个不一样 官方是v21.1了这个版本也会有一个错误，加入这个标识即可，这个你得像上游反馈
[image]

LarryLi-cn, 2022-12-17:
> 这个我刚编译了，有错误但是和你这个不一样 官方是v21.1了这个版本也会有一个错误，加入这个标识即可，这个你得像上游反馈 [image]

多谢指导~  小白请问这个文件在哪里呀~

yyiu, 2023-01-12:
export FORCE_UNSAFE_CONFIGURE=1

coolsnowwolf, 2026-09-05:
该 issue 的 cjdns `-Werror=stringop-overread` 构建失败已被当前 cjdns Makefile 的 `-Wno-error=stringop-overread` 覆盖；同时本轮已针对同一包提交 `75994b736e2f` 并推送，关闭为 completed。

closed 2026-09-05 by coolsnowwolf