coolsnowwolf/lede #10625: 启用FullCone Nat高性能模式之后，出现nf_conntrack: expectation table full

4860575, 2022-12-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：启用FullCone Nat高性能模式之后，出现nf_conntrack: expectation table full
使用兼容模式则没有“nf_conntrack: expectation table full”

### (2) 路由器型号和固件版本

A：wrt3200，最新源码编译

### (3) 详细日志

A：
Fri Dec 16 10:09:21 2022 kern.warn kernel: [36572.424531] nf_conntrack: expectation table full
Fri Dec 16 10:09:21 2022 kern.warn kernel: [36572.429390] nf_conntrack: expectation table full
Fri Dec 16 10:09:21 2022 kern.warn kernel: [36572.434474] nf_conntrack: expectation table full
Fri Dec 16 10:09:22 2022 kern.warn kernel: [36573.364547] nf_conntrack: expectation table full
Fri Dec 16 10:09:22 2022 kern.warn kernel: [36573.369308] nf_conntrack: expectation table full
Fri Dec 16 10:09:22 2022 kern.warn kernel: [36573.374101] nf_conntrack: expectation table full
Fri Dec 16 10:09:22 2022 kern.warn kernel: [36573.378867] nf_conntrack: expectation table full
Fri Dec 16 10:09:22 2022 kern.warn kernel: [36573.383661] nf_conntrack: expectation table full
Fri Dec 16 10:09:22 2022 kern.warn kernel: [36573.388383] nf_conntrack: expectation table full
Fri Dec 16 10:09:26 2022 kern.warn kernel: [36577.575635] net_ratelimit: 47 callbacks suppressed
Fri Dec 16 10:09:26 2022 kern.warn kernel: [36577.575637] nf_conntrack: expectation table full
Fri Dec 16 10:09:26 2022 kern.warn kernel: [36577.585093] nf_conntrack: expectation table full
Fri Dec 16 10:09:37 2022 kern.warn kernel: [36588.465738] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.179183] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.183929] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.188668] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.193398] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.198111] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.202894] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.208202] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.212906] nf_conntrack: expectation table full
Fri Dec 16 10:09:38 2022 kern.warn kernel: [36589.217609] nf_conntrack: expectation table full
Fri Dec 16 10:09:43 2022 kern.warn kernel: [36594.036811] net_ratelimit: 464 callbacks suppressed
Fri Dec 16 10:09:43 2022 kern.warn kernel: [36594.036813] nf_conntrack: expectation table full

coolsnowwolf, 2022-12-16:
半开连接数池不够，需要调大

4860575, 2022-12-16:
> 半开连接数池不够，需要调大

我查看“/proc/sys/net/netfilter/nf_conntrack_expect_max”是128，需要调整这个吗？

Aes64X, 2022-12-16:
> > 半开连接数池不够，需要调大
> 
> 我查看“/proc/sys/net/netfilter/nf_conntrack_expect_max”是128，需要调整这个吗？ 我将其值设为了1024“echo 1024 > /proc/sys/net/netfilter/nf_conntrack_expect_max”，貌似问题依旧。

这数值。。。之前雕改了16384了，你这怎么还是128 - -

4860575, 2022-12-16:
> > > 半开连接数池不够，需要调大
> > 
> > 
> > 我查看“/proc/sys/net/netfilter/nf_conntrack_expect_max”是128，需要调整这个吗？ 我将其值设为了1024“echo 1024 > /proc/sys/net/netfilter/nf_conntrack_expect_max”，貌似问题依旧。
> 
> 这数值。。。之前雕改了16384了，你这怎么还是128 - - [58692d5](https://github.com/coolsnowwolf/lede/commit/58692d5c98169249eae7b8cb27b45ce6ecac1d92)
问题找到了，我自己修改了“sysctl-nf-conntrack.conf”，现在修改回来了问题解决了

4860575, 2022-12-16:
已经解决，问题出在我修了“sysctl-nf-conntrack.conf”

closed 2022-12-16 by 4860575