coolsnowwolf/lede #10629: dnsmasq 勾选“忽略解析文件”后，路由器本地dns仍然会默认为运营商dns

devioa, 2022-12-17:
dnsmasq `noresolv=1`，路由器本地dns仍然会默认为运营商dns
应该是这个commit`afb7eaa4bb0d68151d96f1777313ad28f1b32d90`改动造成的
这会导致访问外网的一些地址走运营商dns解析，然后返回污染的ip
正好我昨天pull最新源码编译，我路由器本地更新ssrp的gfwlist和chinaip就是这样
用GitHub RAW直接失败（`nslookup raw.githubusercontent.com`看ip地址是`0.0.0.0`），用fastlyCDN下载速度也异常慢
但是我LAN侧电脑和手机手动访问它们的地址速度却是正常的

需要`rm -rf /tmp/resolv.conf*`后，再`nslookup qq.com`看默认dns才是本地`127.0.0.1:53`
现在我靠在`/etc/rc.local`加入这行命令解决的

WYC-2020, 2022-12-17:
应该是dnsmasq的服务脚本被改动导致的，最近几次的dnsmasq更新变化太大了，特别服务脚本，和op官方的脚本简直区别太大，按理说这种应该都大同小异的，你可以试试回退应该就能解决

devioa, 2022-12-17:
> 应该是dnsmasq的服务脚本被改动导致的，最近几次的dnsmasq更新变化太大了，特别服务脚本，和op官方的脚本简直区别太大，按理说这种应该都大同小异的，你可以试试回退应该就能解决

我登后续各位大佬修复吧，我现在用rc.local开机删除暂时没啥问题

devioa, 2022-12-17:
> 应该是dnsmasq的服务脚本被改动导致的，最近几次的dnsmasq更新变化太大了，特别服务脚本，和op官方的脚本简直区别太大，按理说这种应该都大同小异的，你可以试试回退应该就能解决

我等后续各位大佬修复吧，我现在用rc.local暂时没啥问题

devioa, 2022-12-19:
最新commit实测已经解决，这个issue可以关闭了

closed 2022-12-19 by devioa