coolsnowwolf/lede #10630: 【服务器端口】检查都正常 但是谷歌连通性检查失败，也打不开谷歌

EatGoods, 2022-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
【服务器端口】检查都正常 但是谷歌连通性检查失败，也打不开谷歌
### (1) 具体问题

[image]

### (2) 路由器型号和固件版本

A：BROUNION R86S/R86S - Intel(R) Celeron(R) N5105 @ 2.00GHz : 4 Core 4 Thread
版本号：OpenWrt R22.11.11 GDQ Autumn[2022] Compiled by eSir / LuCI Master (git-22.330.64765-00290b6)
内核版本：5.15.78

### (3) 详细日志

2022-12-17 11:54:01: -----------end------------
2022-12-17 11:53:59: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:53:59: ----------start------------
2022-12-17 11:53:42: -----------end------------
2022-12-17 11:53:40: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:53:40: ----------start------------
2022-12-17 11:44:03: -----------end------------
2022-12-17 11:44:02: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:44:02: ----------start------------
2022-12-17 11:43:45: -----------end------------
2022-12-17 11:43:44: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:43:44: ----------start------------
2022-12-17 11:43:24: Try restart current server.
2022-12-17 11:43:24: -----------end------------
2022-12-17 11:43:23: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:43:23: ----------start------------
2022-12-17 11:43:19: Current server error, try to switch another server.
2022-12-17 11:31:54: Try restart current server.
2022-12-17 11:31:54: -----------end------------
2022-12-17 11:31:53: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:31:53: ----------start------------
2022-12-17 11:31:50: Current server error, try to switch another server.
2022-12-17 11:31:31: switch to default Alien proxy!
2022-12-17 11:31:31: -----------end------------
2022-12-17 11:31:30: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:31:30: ----------start------------
2022-12-17 11:31:27: Main server is avilable.
2022-12-17 11:31:25: Current server is not default Main server, try to switch back.
2022-12-17 11:20:18: Switch to hong proxy!
2022-12-17 11:20:18: -----------end------------
2022-12-17 11:20:17: Main node:naive 107.0.5304.87 , 4 Threads Started!
2022-12-17 11:20:16: ----------start------------
2022-12-17 11:20:13: Another server is avilable, now switching server.
2022-12-17 11:20:13: Current server error, try to switch another server.
2022-12-17 11:19:55: switch to default Alien proxy!
2022-12-17 11:19:55: -----------end------------
2022-12-17 11:19:54: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:19:54: ----------start------------
2022-12-17 11:19:50: Main server is avilable.
2022-12-17 11:19:48: Current server is not default Main server, try to switch back.
2022-12-17 11:08:41: Switch to vmess+grpc+tls proxy!
2022-12-17 11:08:41: -----------end------------
2022-12-17 11:08:40: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 11:08:40: ----------start------------
2022-12-17 11:08:37: Another server is avilable, now switching server.
2022-12-17 11:08:37: Current server error, try to switch another server.
2022-12-17 10:57:12: -----------end------------
2022-12-17 10:57:10: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:57:10: ----------start------------
2022-12-17 10:56:43: Switch to vmess+grpc+tls proxy!
2022-12-17 10:56:43: -----------end------------
2022-12-17 10:56:42: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:56:42: ----------start------------
2022-12-17 10:56:38: Another server is avilable, now switching server.
2022-12-17 10:56:38: Current server error, try to switch another server.
2022-12-17 10:45:13: -----------end------------
2022-12-17 10:45:12: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:45:12: ----------start------------
2022-12-17 10:42:34: -----------end------------
2022-12-17 10:42:33: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:42:33: ----------start------------
2022-12-17 10:42:29: -----------end------------
2022-12-17 10:42:28: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:42:28: ----------start------------
2022-12-17 10:41:48: -----------end------------
2022-12-17 10:41:46: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:41:46: ----------start------------
2022-12-17 10:41:18: -----------end------------
2022-12-17 10:41:16: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:41:16: ----------start------------
2022-12-17 10:41:12: -----------end------------
2022-12-17 10:41:11: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:41:11: ----------start------------
2022-12-17 10:41:01: -----------end------------
2022-12-17 10:41:00: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:41:00: ----------start------------
2022-12-17 10:39:49: -----------end------------
2022-12-17 10:39:48: Main node:Xray 1.6.4 (Xray, Penetrates Everything.) OpenWrt (go1.19.3 linux/amd64) Started!
2022-12-17 10:39:47: ----------start------------
2022-12-17 10:39:29: -----------end------------
2022-12-17 10:39:29: 禁止连接的域名加载完毕。
2022-12-17 10:39:29: 未启动主节点，禁止连接的域名正在加载。
2022-12-17 10:39:29: ----------start------------
2022-12-17 18:35:07: -----------end------------
2022-12-17 18:35:07: 禁止连接的域名加载完毕。
2022-12-17 18:35:07: 未启动主节点，禁止连接的域名正在加载。
2022-12-17 18:35:07: ----------start------------
2022-12-17 18:35:07: boot！

759071793, 2022-12-17:
不要使用6.1内核

guochan2019, 2022-12-17:
昨天晚上编译后出现同样问题，科学插件版本未更新排除，估计跟新版dnsmasq有关，无能力解决，暂时刷回老版

closed 2022-12-30 by coolsnowwolf