coolsnowwolf/lede #10631: 6.1内核版本挂载ntfs硬盘，中文目录无法显示

HawaiiGit, 2022-12-17:
root@OpenWrt:/mnt/sdb1/videos# ll
ls: ./__: No such file or directory
ls: ./_____: No such file or directory
ls: ./__: No such file or directory

英文文件夹可以正常显示，中文文件夹显示不出来，提示No such file or directory，请大佬解答

starxu1971, 2022-12-17:
这个和NFS没关系，是windows的事具体百度有解决方法。

mzl0408, 2022-12-17:
> root@OpenWrt:/mnt/sdb1/videos# ll ls: ./**: No such file or directory ls: ./**_**: No such file or directory ls: ./**: No such file or directory
> 
> 英文文件夹可以正常显示，中文文件夹显示不出来，提示No such file or directory，请大佬解答

我也遇到这个问题了，应该不仅是6.1版本，别的5.15也不行

mzl0408, 2022-12-17:
> 这个和NFS没关系，是windows的事具体百度有解决方法。

手机上的ES文件浏览器也不行

HawaiiGit, 2022-12-17:
> > root@OpenWrt:/mnt/sdb1/videos# ll ls: ./**: No such file or directory ls: ./**_**: No such file or directory ls: ./**: No such file or directory
> > 英文文件夹可以正常显示，中文文件夹显示不出来，提示No such file or directory，请大佬解答
> 
> 我也遇到这个问题了，应该不仅是6.1版本，别的5.15也不行

我之前用5.19内核版本，中文目录可以正常显示

mzl0408, 2022-12-17:
> > > root@OpenWrt:/mnt/sdb1/videos# ll ls: ./**: No such file or directory ls: ./**_**: No such file or directory ls: ./**: No such file or directory
> > > 英文文件夹可以正常显示，中文文件夹显示不出来，提示No such file or directory，请大佬解答
> > 
> > 
> > 我也遇到这个问题了，应该不仅是6.1版本，别的5.15也不行
> 
> 我之前用5.19内核版本，中文目录可以正常显示

我是昨天把编译的所有文件夹全部删除了，然后重新编译一遍，结果就都不行了。

starxu1971, 2022-12-18:
windows肯定可以的你们自己百度，我用的6.0内核，因为要截图说明直接贴个链接[https://blog.csdn.net/baidu_41909866/article/details/116305972](url)

cuiyf5516, 2022-12-18:
[image]
看着是不加-t参数mount，iocharset不会设置成utf-8，所以才导致的中文无法显示

HawaiiGit, 2022-12-18:
> 
感谢分享，硬盘已经格了，没机会试了

closed 2022-12-18 by HawaiiGit

mzl0408, 2022-12-18:
> [image] 看着是不加-t参数mount，iocharset不会设置成utf-8，所以才导致的中文无法显示

那意思是mount后面加个-t就可以了吗？

cuiyf5516, 2022-12-18:
> > [image] 看着是不加-t参数mount，iocharset不会设置成utf-8，所以才导致的中文无法显示
> 
> 那意思是mount后面加个-t就可以了吗？

加-t手动指定为ntfs就行，但自动挂载为什么会有问题需要大佬们帮忙排查一下