coolsnowwolf/lede #10635: 6.1内核自动挂载ntfs磁盘时，iocharset参数不对，导致无法访问中文目录

cuiyf5516, 2022-12-18:
[image]
看着是不加-t参数mount，iocharset不会设置成utf-8，所以才导致的中文无法显示

_Originally posted by @cuiyf5516 in https://github.com/coolsnowwolf/lede/issues/10631#issuecomment-1356729744_

tttt2001tttt, 2022-12-19:
同样遇到无法识别，已经换回旧固件，只能等修复吧

mzl0408, 2022-12-20:
同样的问题，不只是6.1的问题，我用的5.15编译也是，应该是最近源码问题

WYC-2020, 2022-12-21:
[image]
我这边测试 没问题哈哈，不过我用的samba4

mzl0408, 2022-12-21:
> [image] 我这边测试 没问题哈哈，不过我用的samba4
是不是源码更新了，我自从前几天彻底删除源码重新编译后就一直不行

cuiyf5516, 2022-12-22:
> [image] 我这边测试 没问题哈哈，不过我用的samba4

挂载到某个iocharset后添加的文件肯定没有编码问题呀，像我这种被挂载到不同的iocharset后就因为编码问题不能正常显示文件名
[image]

cuiyf5516, 2022-12-22:
[image]
fstools的最新修改，不再需要实现mount.ntfs脚本的方式来自动挂载，这样脚本里手动指定的iocharset也不会生效。
[image]
如果需要就把fstools的最新修改回滚一下再编译，或者只能手动指定类型为ntfs重新mount了。

PS：或者考虑提供修改CONFIG_NLS_DEFAULT参数的选项？

coolsnowwolf, 2022-12-22:
fixed 

https://github.com/coolsnowwolf/lede/commit/317d92dd3e0ebb07b5343ffc8ecd4db8fae36b2d

WYC-2020, 2022-12-23:
> > [image] 我这边测试 没问题哈哈，不过我用的samba4
> 
> 挂载到某个iocharset后添加的文件肯定没有编码问题呀，像我这种被挂载到不同的iocharset后就因为编码问题不能正常显示文件名 [image]

什么场景会用到一个磁盘挂载到不同编码上？现在编码那么多解决这个下一个说不定一样也会这样

WYC-2020, 2022-12-23:
l大的修改也就是fat ntfs固定用utf8也还不是丢失了你所谓的多种编码

cuiyf5516, 2022-12-23:
> l大的修改也就是fat ntfs固定用utf8也还不是丢失了你所谓的多种编码

觉得理解错我的问题了，这个问题是因为原来自动挂载因为有mount.ntfs中指定了utf-8所以一直没有问题，但是因为fstools最新版的修改不会再通过mount.ntfs挂载了，所以挂载时iocharset参数用的值是CONFIG_NLS_DEFAULT，所以会出现编码问题。

l大的修改就是为了达到跟以前一样的效果，自动挂载ntfs磁盘时默认的iocharset就是固定成utf-8。

cuiyf5516, 2022-12-23:
> fixed
> 
> [317d92d](https://github.com/coolsnowwolf/lede/commit/317d92dd3e0ebb07b5343ffc8ecd4db8fae36b2d)

verified

closed 2022-12-23 by cuiyf5516