coolsnowwolf/lede #10638: ShadowSocksR Plus+ 分流异常, 选绕过国内IP时也是全局代理

onlyJinx, 2022-12-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

SS Plus+分流异常

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：先说明一下, 服务端我的配置是 **禁止访问所有国内IP和网站,** 就是说如果把国内的域名代理了, 那会导致直接无法访问(被我服务端block了)
分流如图, 选的是绕过国内IP, 然后连通性检查的时候会提示错误, 我上服务端看日志的时候, 发现baidu这个域名被代理了, 只有GFW列表模式可以正常工作, 重新编译了也是同样的问题, 现在没有处理的头绪

[image]

[image]

### (2) 路由器型号和固件版本

A：x86 软路由

### (3) 详细日志

A：
```
2022-12-18 20:37:38: -----------end------------
2022-12-18 20:37:02: Main node:Xray 1.6.6-2 (Xray, Penetrates Everything.) OpenWrt (go1.19.4 linux/amd64) Started!
2022-12-18 20:37:02: ----------start------------
2022-12-18 20:36:10: -----------end------------
2022-12-18 20:35:34: Main node:Xray 1.6.6-2 (Xray, Penetrates Everything.) OpenWrt (go1.19.4 linux/amd64) Started!
2022-12-18 20:35:34: ----------start------------
2022-12-18 20:24:06: -----------end------------
2022-12-18 20:23:31: Main node:Xray 1.6.6-2 (Xray, Penetrates Everything.) OpenWrt (go1.19.4 linux/amd64) Started!
```

w311ang, 2022-12-19:
自己看防火墙规则吧

onlyJinx, 2022-12-19:
刚编译好的固件, 防火墙没做任何改动, 我试了passwall2也是同样的问题, 直接把所有连接都走了代理, Hello world 就可以正常工作, 下面是防火墙规则截图.

[image]

binge8, 2022-12-19:
显然不可能，我一直开绕过国内IP，就一个100G的小机场一个月都用不完，要是全部代理了100G早不够用了吧

onlyJinx, 2022-12-19:
我也觉得奇怪，难道是少编译了什么用来解析DNS的插件？我都是自己选的，不认识的全都被我取消了。。。只选了一些自己需要的

w311ang, 2022-12-20:
@onlyJinx 我没看到又ipset匹配china的数据包被return，你看看`ipset list china`有没有ip

onlyJinx, 2022-12-20:
@w311ang 应该不是这个问题，我刚才又编译了一遍，然后发现不是编译的问题，我用nslookup随便查了几个国内的域名，发现解析回来的ip全都是国外的，所以sorry plus才会按照国外ip分流出去的，现在把dns问题解决应该就可以，但是我不知道为什么会解析成国外ip，我路由器的DNS指向的是主路由

w311ang, 2022-12-20:
@onlyJinx dnsmasq的配置应该在类似/var/etc/dnsmasq.conf这个文件，你贴出来一下

onlyJinx, 2022-12-20:
这个路径下只有一个 dnsmasq.conf.cfg01411c
```
root@OpenWrt:~# ls /var/etc/
bypass.include          shadowsocksr.include    ssl                     uhttpd
dnsmasq.conf.cfg01411c  smartdns                ssrplus
root@OpenWrt:~#
```

里面的内容如下:
```
root@OpenWrt:~# cat /var/etc/dnsmasq.conf.cfg01411c
# auto-generated config file from /etc/config/dhcp
conf-file=/etc/dnsmasq.conf
dhcp-authoritative
domain-needed
localise-queries
read-ethers
enable-ubus=dnsmasq
expand-hosts
bind-dynamic
local-service
cache-size=8192
edns-packet-max=1232
domain=lan
local=/lan/
addn-hosts=/tmp/hosts
dhcp-leasefile=/tmp/dhcp.leases
resolv-file=/tmp/resolv.conf.d/resolv.conf.auto
stop-dns-rebind
rebind-localhost-ok
dhcp-broadcast=tag:needs-broadcast
conf-dir=/tmp/dnsmasq.d
user=dnsmasq
group=dnsmasq

dhcp-ignore-names=tag:dhcp_bogus_hostname
conf-file=/usr/share/dnsmasq/dhcpbogushostname.conf

srv-host=_vlmcs._tcp,OpenWrt,1688,0,100

bogus-priv
conf-file=/usr/share/dnsmasq/rfc6761.conf
no-dhcp-interface=eth0
enable-ra
quiet-ra

root@OpenWrt:~#
```

这是未开启ssr plus的时候nslookup反馈结果:
```
root@OpenWrt:~# nslookup www.baidu.com
Server:         127.0.0.1
Address:        127.0.0.1:53

Non-authoritative answer:
www.baidu.com   canonical name = www.a.shifen.com
Name:   www.a.shifen.com
Address: 14.215.177.38
Name:   www.a.shifen.com
Address: 14.215.177.39

Non-authoritative answer:
www.baidu.com   canonical name = www.a.shifen.com

root@OpenWrt:~#
```
这里的反馈是正常的, ip归属是国内,

下面的是开启了ssr plus的时候:

```
root@OpenWrt:~# nslookup www.baidu.com
Server:         127.0.0.1
Address:        127.0.0.1:53

** server can't find www.baidu.com: REFUSED

Non-authoritative answer:
www.baidu.com   canonical name = www.a.shifen.com
www.a.shifen.com        canonical name = www.wshifen.com
Name:   www.wshifen.com
Address: 104.193.88.77
Name:   www.wshifen.com
Address: 104.193.88.123

root@OpenWrt:~#
```

国内DNS我这里选的是腾讯的DNSPod, 我尝试更换成`WAN口下发的DNS`也是同样的结果

[image]

w311ang, 2022-12-20:
@onlyJinx 如果不选国内dns服务器会不会好

onlyJinx, 2022-12-20:
国内DNS留空不选的话也是一样的结果
[image]
[image]

```
root@OpenWrt:~# ipset list china
Name: china
Type: hash:net
Revision: 7
Header: family inet hashsize 262144 maxelem 1000000 bucketsize 12 initval 0x57666b87
Size in memory: 6576
References: 4
Number of entries: 0
Members:
root@OpenWrt:~#
```

onlyJinx, 2022-12-20:
@w311ang 我想我知道原因了, 我手动点击`【国内IP段】数据库`的时候是无法更新的, 这个更新地址应该属于国内, 然后走了代理之后被我的服务端block掉了, 我关闭代理之后点击更新, 然后成功更新了,  `ipset list china`也可以列出很多ip地址了, 现在能正常使用了, 应该是ssr plus的处理逻辑是第一次启动时会去自动更新这个ip段数据库, 然后刚好被我的服务端block一直无法更新成功, 才出现所有地址都当成国外ip处理的, 这个ip数据库一直得不到更新, 陷入死循环, 问题已解决, 多谢啦

onlyJinx, 2022-12-21:
很奇怪, 我做了改动, 一保存, `ipset list china`里的内容就被全部清空了, 然后, 国内网址有全部都被代理过去了.
现在再点击更新的时候就提示已经是最新的了, 请问这个要怎么把他更新出来呀
[image]

```
root@OpenWrt:~# ipset list china
Name: china
Type: hash:net
Revision: 7
Header: family inet hashsize 4096 maxelem 65536 bucketsize 12 initval 0xa6620f64
Size in memory: 528
References: 0
Number of entries: 0
Members:
root@OpenWrt:~#
```

w311ang, 2022-12-21:
@onlyJinx https://github.com/fw876/helloworld/search?q=chinaipset.sh&type=

onlyJinx, 2022-12-23:
@w311ang 好的， 就是说以后这个list空了的话， 就自己手动执行以下`/usr/share/shadowsocksr/chinaipset.sh /var/etc/ssrplus/china_ssr.txt`去更新对吧？
另外想请教一下，怎么看到ssr puls的分流日志呢？因为我服务端有时候会收到一些国内域名的请求， 我想看看是不是ssr plus的问题，要看看日志

w311ang, 2022-12-23:
@onlyJinx ssrplus依靠的是iptables和dnsmasq做的分流，具体的日志也还是只能看iptables和dnsmasq，不过你可以间接地通过nslookup查询dns和查看ipset的china和whitelist列表还有防火墙的计数看看走了代理还是直连

onlyJinx, 2022-12-23:
好的，我打开部分被墙域名的话经常会提示**连接已被重置**，就是那种我没打开代理的表现，要我手动刷新好几次才有概率打开，提示**连接已被重置**时，我看着服务端v2ray是没有相关的访问记录的，只有把页面刷新出来我在服务端日志才可以看到这个网址，我想问下这是不是和IPv6有关系，当返回IPv6时我无法访问，只有多次刷新返回IPv4的时候我才能正常代理出去然后访问页面？

如果我在DHCP里勾选了禁用解析IPv6的话，会导致整个dnsmasq无法启动，然后解析不了任何dns请求，下面是日志还有勾选了禁用IPv6后的配置文件内容。

[image]

```
Fri Dec 23 15:30:39 2022 daemon.crit dnsmasq[1]: bad option at line 12 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 15:30:39 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 15:30:44 2022 daemon.crit dnsmasq[1]: bad option at line 12 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 15:30:44 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 15:30:49 2022 daemon.crit dnsmasq[1]: bad option at line 12 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 15:30:49 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 15:30:54 2022 daemon.crit dnsmasq[1]: bad option at line 12 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 15:30:54 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 15:30:59 2022 daemon.crit dnsmasq[1]: bad option at line 12 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 15:30:59 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 15:31:04 2022 daemon.crit dnsmasq[1]: bad option at line 12 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 15:31:04 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 15:31:04 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Fri Dec 23 15:31:33 2022 daemon.err procd: Got unexpected signal 1
```

```
root@OpenWrt:~# cat /var/etc/dnsmasq.conf.cfg01411c
# auto-generated config file from /etc/config/dhcp
conf-file=/etc/dnsmasq.conf
dhcp-authoritative
filterwin2k
log-queries=extra
localise-queries
read-ethers
enable-ubus=dnsmasq
expand-hosts
bind-dynamic
local-service
filter-aaaa
edns-packet-max=1232
domain=lan
local=/lan/
addn-hosts=/tmp/hosts
dhcp-leasefile=/tmp/dhcp.leases
resolv-file=/tmp/resolv.conf.d/resolv.conf.auto
stop-dns-rebind
rebind-localhost-ok
dhcp-broadcast=tag:needs-broadcast
conf-dir=/tmp/dnsmasq.d
user=dnsmasq
group=dnsmasq

dhcp-ignore-names=tag:dhcp_bogus_hostname
conf-file=/usr/share/dnsmasq/dhcpbogushostname.conf

srv-host=_vlmcs._tcp,OpenWrt,1688,0,100

bogus-priv
conf-file=/usr/share/dnsmasq/rfc6761.conf
no-dhcp-interface=eth0

root@OpenWrt:~#
```

```
root@OpenWrt:~#  nslookup xxxbus.com
Server:         127.0.0.1
Address:        127.0.0.1:53

Non-authoritative answer:
Name:   xxxbus.com
Address: 104.25.111.111
Name:   xxxbus.com
Address: 172.67.80.44
Name:   xxxbus.com
Address: 104.25.112.111

Non-authoritative answer:
Name:   xxxbus.com
Address: 2606:4700:20::6819:6f6f
Name:   xxxbus.com
Address: 2606:4700:20::6819:706f
Name:   xxxbus.com
Address: 2606:4700:20::ac43:502c
```

w311ang, 2022-12-23:
@onlyJinx 这个过滤ipv6无效的问题建议另开issue，这个我也不知道怎么解决

onlyJinx, 2022-12-23:
嗯嗯，我想确认一下是否是IPv6这个原因导致的，因为有些网站确实需要刷新好几次才可以打开（打开之后，后续就都正常，等过段时间再开问题会复现，可能是DNS缓存刷新了），这段期间确认代理没问题的。

w311ang, 2022-12-23:
ssrplus不支持ipv6代理，所有ipv6连接都是直连，的确是可以怀疑走了ipv6

onlyJinx, 2022-12-23:
好的， 感谢。
ipv6问题我将重新开个issue或者重新编译一个固件，不勾选ipv6相关。

closed 2022-12-23 by onlyJinx

fengshengxie, 2023-01-14:
> 显然不可能，我一直开绕过国内IP，就一个100G的小机场一个月都用不完，要是全部代理了100G早不够用了吧

windows 11 才会出现楼主说的问题。我也出现了。

onlyJinx, 2023-01-14:
> > 显然不可能，我一直开绕过国内IP，就一个100G的小机场一个月都用不完，要是全部代理了100G早不够用了吧
> 
> windows 11 才会出现楼主说的问题。我也出现了。

不是客户端的问题，是openwrt的问题，我发现每次重启或者更改ssrp的配置保存后(会自动重启ssrp应用这些更改)就有概率让ipset里得China list全部丢失，然后ssrp就会把所有的ip判断为“国外”ip然后全部分流出去，现在我也不知道怎么搞了，只能手动运行一下那个脚本让China list重新导入然后尽量不重启

enterusernamecontinue, 2023-02-22:
+1

手里三台设备，尝试了不同的 OpenWRT 版本，SSRP+ 也试了最近两个版本，【绕过中国大陆模式】遇到了相同的问题，暂时还没有什么解决办法。

eriflleh, 2023-04-03:
我跟你遇到了相同的问题，由于我的技术底下，所以只能采取了一个笨办法，我每次启动节点之后，要是没有ip列表的话我就在ssrp的高级设置-中国大陆 IP 段更新 URL这里 切换一些 然后去更新一下库 就显示IP了 然后正常分流了

enterusernamecontinue, 2023-04-03:
已解决。

SSRP>访问控制>代理域名>ispip.clang.cn ，
然后主服务器上的路由规则里，在屏蔽大陆的规则前添加一条 ispip.clang.cn 直连的规则。

onlyJinx, 2023-04-03:
> 

我这里解决不了，ip数据库已经成功更新，然后重启ssrp，还是会把国内的ip全部代理出去
[image]

eriflleh, 2023-04-04:
> 我这里解决不了，ip数据库已经成功更新，然后重启ssrp，还是会把国内的ip全部代理出去
> [image]

我是先启动，再更新ip数据库，更新成功以后，不重启先用着

VwEl, 2023-04-05:
> > 
> 
> 我这里解决不了，ip数据库已经成功更新，然后重启ssrp，还是会把国内的ip全部代理出去 [image]

这个用啥命令看的

onlyJinx, 2023-04-05:
> 

这个是服务端的日志，贴出来想说明更新ip数据库是正常的，国内域名baidu被代理到了服务端(分流异常)

lsbjkc, 2023-04-13:
> > 
> 
> 这个是服务端的日志，贴出来想说明更新ip数据库是正常的，国内域名baidu被代理到了服务端(分流异常)

楼主解决了吗？我也遇到类似的问题，用五千多条的那个IP库可以顺利分流，用八千多条的IP库就跟开了全局一样。同样是clang.cn，不知道两个IP库为什么会造成这种区别。

onlyJinx, 2023-04-13:
> 楼主解决了吗？我也遇到类似的问题，用五千多条的那个IP库可以顺利分流，用八千多条的IP库就跟开了全局一样。同样是clang.cn，不知道两个IP库为什么会造成这种区别。

无力解决，一次弄好之后软件重启就会又出问题，懒得搞了，直接debian11 + iptables + xray-core一把梭。。

lsbjkc, 2023-04-14:
> > 楼主解决了吗？我也遇到类似的问题，用五千多条的那个IP库可以顺利分流，用八千多条的IP库就跟开了全局一样。同样是clang.cn，不知道两个IP库为什么会造成这种区别。
> 
> 无力解决，一次弄好之后软件重启就会又出问题，懒得搞了，直接debian11 + iptables + xray-core一把梭。。

我把访问控制里【不走代理的域名】和【强制走代理的域名】都清空，IP分流就正常了，好像是域名分流和IP分流有冲突

HuTao1996, 2023-04-18:
老版本的ShadowSocksR Plus+没有这个问题，新版本有这个问题，所以我刷回老版本了！

05u, 2023-06-16:
> 

老哥,你用的那个版本的

innojerry, 2023-07-28:
我也是同样的问题, 只要软路由重新启动, 选择是SS plus 绕过国内IP, 启动后, 谷歌可以连通, 但是百度就是不可以, 只要关掉墙, 再重新开启, 就可以连通百度!  如果选择GFW 模式, 启动没有问题, 但是许多网站没法解析, 也就上不去!

innojerry, 2023-07-28:
> > 老版本的ShadowSocksR Plus+没有这个问题，新版本有这个问题，所以我刷回老版本了！
> 
> 老哥,你用的那个版本的

老版本的Core是可以升级为1.83版本的?还是老的Xray Core?

innojerry, 2023-07-28:
> 

清空就可以吗>  我回家试试

innojerry, 2023-07-28:
> > > 楼主解决了吗？我也遇到类似的问题，用五千多条的那个IP库可以顺利分流，用八千多条的IP库就跟开了全局一样。同样是clang.cn，不知道两个IP库为什么会造成这种区别。
> > 
> > 
> > 无力解决，一次弄好之后软件重启就会又出问题，懒得搞了，直接debian11 + iptables + xray-core一把梭。。
> 
> 我把访问控制里【不走代理的域名】和【强制走代理的域名】都清空，IP分流就正常了，好像是域名分流和IP分流有冲突

清空访问代理里面的域名就可以吗? 我试试, 看看实际是否可以解决问题

innojerry, 2023-07-28:
> @w311ang 我想我知道原因了, 我手动点击`【国内IP段】数据库`的时候是无法更新的, 这个更新地址应该属于国内, 然后走了代理之后被我的服务端block掉了, 我关闭代理之后点击更新, 然后成功更新了, `ipset list china`也可以列出很多ip地址了, 现在能正常使用了, 应该是ssr plus的处理逻辑是第一次启动时会去自动更新这个ip段数据库, 然后刚好被我的服务端block一直无法更新成功, 才出现所有地址都当成国外ip处理的, 这个ip数据库一直得不到更新, 陷入死循环, 问题已解决, 多谢啦

但是软路由重新启动后, 还是一样的老问题啊?  需要再次手动关闭SS Plus, 再启动, 就可以连通百度了!  总不可以不重启软路由吧?>

onlyJinx, 2023-07-28:
@innojerry 我前段时间重新拉取源码编译之后好像就没有这个问题来了,你可以试试.

innojerry, 2023-07-28:
> @innojerry 我前段时间重新拉取源码编译之后好像就没有这个问题来了,你可以试试.

这个我还不知道如何解决呢?  请教一下, 如何解决?

onlyJinx, 2023-07-28:
> 这个我还不知道如何解决呢? 请教一下, 如何解决?

我好像记得当时是用dirclean清理,然后重新编译就好了,不过用了这个清理之后可能会让你回到第一次编译的那种状态需要编译很久,具体我也不是很清楚,跟着[教程](https://p3terx.com/archives/openwrt-compilation-steps-and-commands.html)误打误撞就搞好了

```
make dirclean
git pull
./scripts/feeds update -a
./scripts/feeds install -a
make defconfig
make download -j8
make V=s -j$(nproc)
```

chris-ss, 2023-07-31:
解决了，多谢@onlyJinx 的探索。
我是一样的情况，用绕过中国IP模式，全部网站都走了代理。
命令 ipset list china 查看为空，
通过命令 /usr/share/shadowsocksr/chinaipset.sh /var/etc/ssrplus/china_ssr.txt 手动去更新时候报错 name 已存在，
然后自己 cat /usr/share/shadowsocksr/chinaipset.sh 里面看执行的代码
猜测是之前已经有china的ipset导致重名无法创建了
注销掉了 /usr/share/shadowsocksr/chinaipset.sh 中 create china hash:net那一行再手动执行 /usr/share/shadowsocksr/chinaipset.sh /var/etc/ssrplus/china_ssr.txt 目前已成功。
ipset list china 有了且分流正常。
猜测应该是每次写入到 iptables 都会创建china的集合 create china hash:net，但是某些操作时候只会flush清空该集合而不是destory销毁该集合，这样就导致下次创建同名集合时候失败报错...

Issac-Lee, 2023-08-06:
我已经放弃了，改用openclash了

dogelee2, 2023-09-07:
我找到了个土办法
就是把直连的国内ip，手动加进配置文件里面,配置文件在/etc/config/shadowscoksr里面，缺点就是加完之后启动有点慢
config access_control
	option lan_ac_mode '0'
	option router_proxy '1'
	list Interface 'lan'
后面把这https://github.com/dogelee2/pic/raw/master/shadowsocksr 里的都加进去，加完快6000多行了
[image]
[image]

1242811099, 2023-11-29:
你好，我遇到了相似的问题，或者我认为相似的问题，不知道如何解决，我选择绕过国内IP并且把国内IP更新到最新后，国内的延迟却比国外高

exing2, 2024-03-18:
还是没解决？

MoreStudyLatter, 2024-12-08:
> > > > 楼主解决了吗？我也遇到类似的问题，用五千多条的那个IP库可以顺利分流，用八千多条的IP库就跟开了全局一样。同样是clang.cn，不知道两个IP库为什么会造成这种区别。
> > > 
> > > 
> > > 无力解决，一次弄好之后软件重启就会又出问题，懒得搞了，直接debian11 + iptables + xray-core一把梭。。
> > 
> > 
> > 我把访问控制里【不走代理的域名】和【强制走代理的域名】都清空，IP分流就正常了，好像是域名分流和IP分流有冲突
> 
> +1 这样可以了

+1 清空后，确实可以了。

johnxjin, 2024-12-09:
> 我也是一样的问题呢

更换成passwall2 不折腾了

chpiter, 2025-01-01:
把openwrt的ipv6关闭试试~~  @johnxjin