coolsnowwolf/lede #10639: 更新最新源码后编译出错

mitsukileung, 2022-12-18:
make[4]: Entering directory '/home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/dnsmasq-full/dnsmasq-2.87'
Package libnftables was not found in the pkg-config search path.
Perhaps you should add the directory containing `libnftables.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libnftables', required by 'virtual:world', not found
Package libnftables was not found in the pkg-config search path.
Perhaps you should add the directory containing `libnftables.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libnftables', required by 'virtual:world', not found
make[5]: Entering directory '/home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/dnsmasq-full/dnsmasq-2.87/src'
aarch64-openwrt-linux-musl-gcc -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/dnsmasq-full/dnsmasq-2.87=dnsmasq-2.87 -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -I/home/koysuke/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include -I/home/koysuke/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify -I/home/koysuke/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include -DHAVE_UBUS -DHAVE_POLL_H    -DHAVE_DNSSEC   -DHAVE_NFTSET -DHAVE_CONNTRACK -DNO_ID     -DVERSION='"2.87"'      -I/home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include    -I/home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include     -c nftset.c	
nftset.c:22:10: fatal error: nftables/libnftables.h: No such file or directory
   22 | #include <nftables/libnftables.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [/home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/dnsmasq-full/dnsmasq-2.87/Makefile:168: nftset.o] Error 1
make[5]: Leaving directory '/home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/dnsmasq-full/dnsmasq-2.87/src'
make[4]: *** [Makefile:93: all] Error 2
make[4]: Leaving directory '/home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/dnsmasq-full/dnsmasq-2.87'
make[3]: *** [Makefile:207: /home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/dnsmasq-full/dnsmasq-2.87/.built] Error 2
make[3]: Leaving directory '/home/koysuke/lede/package/network/services/dnsmasq'
time: package/network/services/dnsmasq/full/compile#0.26#0.14#0.36
    ERROR: package/network/services/dnsmasq failed to build (build variant: full).
make[2]: *** [package/Makefile:116: package/network/services/dnsmasq/compile] Error 1
make[2]: Leaving directory '/home/koysuke/lede'
make[1]: *** [package/Makefile:110: /home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/koysuke/lede'
make: *** [/home/koysuke/lede/include/toplevel.mk:231：world] 错误 2

大佬们，这是哪里出错？

ecrasy, 2022-12-18:
#10628 需要自己替换模块或者等L大升级代码

mitsukileung, 2022-12-19:
> #10628 需要自己替换模块或者等L大升级代码

谢谢，我试试

mitsukileung, 2022-12-19:
> #10628 需要自己替换模块或者等L大升级代码

把nftables升到 1.0.5 错误还是一样

mitsukileung, 2022-12-19:
> 我刚又测试了一遍 编译没啥问题 你可能需要清空download缓存 再make download刷新一下packages 然后编译试试看

大佬，能不能把 nftables 的makefile 配置贴一份出来
我修改 PKG_VERSION 和 PKG_HASH 重拉库，再编译也是不行，是不是还有什么要修改？

zhaochina2025, 2025-05-02:
怎么解决的？