coolsnowwolf/lede #10645: ipq806x编译失败，backports-6.1-rc8部分编译报错

arimitx, 2022-12-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ipq806x编译失败，backports-6.1-rc8部分编译报错

### (2) 路由器型号和固件版本

A：Netgear R7800，lede commit version b45566e，kernel version 5.4.227

### (3) 详细日志

A：完整log见 https://pastebin.com/HfWHDrs8 ，以下为报错部分日志
```
make[4]: Entering directory '/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8'
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath/main.o
In file included from /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/net/mac80211.h:20,
                 from /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath/ath.h:24,
                 from /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath/main.c:22:
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:313:9: error: expected specifier-qualifier-list before 'struct_group'
  313 |         struct_group(addrs,
      |         ^~~~~~~~~~~~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_is_frag':
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:804:19: error: 'struct ieee80211_hdr' has no member named 'seq_ctrl'
  804 |                hdr->seq_ctrl & cpu_to_le16(IEEE80211_SCTL_FRAG);
      |                   ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_get_SA':
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:4099:27: error: 'struct ieee80211_hdr' has no member named 'addr4'
 4099 |                 return hdr->addr4;
      |                           ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:4101:27: error: 'struct ieee80211_hdr' has no member named 'addr3'
 4101 |                 return hdr->addr3;
      |                           ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:4102:19: error: 'struct ieee80211_hdr' has no member named 'addr2'
 4102 |         return hdr->addr2;
      |                   ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_get_DA':
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:4118:27: error: 'struct ieee80211_hdr' has no member named 'addr3'
 4118 |                 return hdr->addr3;
      |                           ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:4120:27: error: 'struct ieee80211_hdr' has no member named 'addr1'
 4120 |                 return hdr->addr1;
      |                           ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h: In function '_ieee80211_is_group_privacy_action':
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/include/linux/ieee80211.h:4197:41: error: 'struct ieee80211_hdr' has no member named 'addr1'
 4197 |             !is_multicast_ether_addr(hdr->addr1))
      |                                         ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath/main.c: In function 'ath_is_mybeacon':
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath/main.c:67:44: error: 'struct ieee80211_hdr' has no member named 'addr3'
   67 |                 ether_addr_equal_64bits(hdr->addr3, common->curbssid);
      |                                            ^~
/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath/main.c:68:1: error: control reaches end of non-void function [-Werror=return-type]
   68 | }
      | ^
cc1: some warnings being treated as errors
make[10]: *** [scripts/Makefile.build:262: /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath/main.o] Error 1
make[9]: *** [scripts/Makefile.build:497: /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless/ath] Error 2
make[8]: *** [scripts/Makefile.build:497: /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1739: /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8'
make[3]: *** [Makefile:398: /mnt/data/openwrt/lede-r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/backports-6.1-rc8/.built] Error 2
make[3]: Leaving directory '/mnt/data/openwrt/lede-r7800/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.21#0.29#1.53
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/mnt/data/openwrt/lede-r7800'
make[1]: *** [package/Makefile:110: /mnt/data/openwrt/lede-r7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/data/openwrt/lede-r7800'
make: *** [/mnt/data/openwrt/lede-r7800/include/toplevel.mk:231: world] Error 2
```

coolsnowwolf, 2022-12-19:
已修复

arimitx, 2022-12-20:
谢谢大雕

closed 2022-12-20 by arimitx