coolsnowwolf/lede #10646: 今天编译的固件dnsmasq运行错误无法分配ip

smdx, 2022-12-19:
各位大佬好，今天编译的固件出现dnsmasq运行错误可以上网但无法分配ip的现象。

运行环境：PVE 7.1-7 内核5.13.19-6 LXC运行openwrt
系统日志：
```
Mon Dec 19 17:34:53 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 17:34:53 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 17:34:56 2022 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Mon Dec 19 17:34:56 2022 daemon.info procd: - init complete -
Mon Dec 19 17:34:58 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 17:34:58 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 17:35:03 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 17:35:03 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 17:35:04 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 17:35:04 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 17:35:05 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 17:35:05 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 17:35:05 2022 cron.err crond[12397]: crond (busybox 1.35.0) started, log level 5
Mon Dec 19 17:35:09 2022 daemon.err nmbd[2234]: [2022/12/19 17:35:09.300322,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Mon Dec 19 17:35:09 2022 daemon.err nmbd[2234]:   *****
Mon Dec 19 17:35:09 2022 daemon.err nmbd[2234]:
Mon Dec 19 17:35:09 2022 daemon.err nmbd[2234]:   Samba name server OPENWRT-LEAN is now a local master browser for workgroup WORKGROUP on subnet 10.0.0.1
Mon Dec 19 17:35:09 2022 daemon.err nmbd[2234]:
Mon Dec 19 17:35:09 2022 daemon.err nmbd[2234]:   *****
Mon Dec 19 17:35:09 2022 daemon.info procd: Instance sysntpd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Mon Dec 19 17:35:10 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 17:35:10 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 17:35:11 2022 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Mon Dec 19 17:35:15 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 17:35:15 2022 daemon.crit dnsmasq[1]: FAILED to start up
```

ricksuzade-maker, 2022-12-19:
抹掉所有设置，重置

weilai-all, 2022-12-19:
把 /etc/config/dhcp文件 dnsmasq 配置替换成初始配置就行

smdx, 2022-12-19:
已经切回旧固件了，等没人玩游戏我再试试。谢谢各位

BenjaminX, 2022-12-19:
删掉
	option filter_aaaa '1'
这行

MIAOBUSI, 2022-12-19:
Mon Dec 19 21:30:11 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 21:30:11 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 21:30:14 2022 cron.err crond[12847]: crond (busybox 1.35.0) started, log level 8
Mon Dec 19 21:30:14 2022 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Stop!
Mon Dec 19 21:30:14 2022 cron.err crond[13583]: crond (busybox 1.35.0) started, log level 8
Mon Dec 19 21:30:14 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 21:30:14 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 21:30:19 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 21:30:19 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 21:30:24 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 21:30:24 2022 daemon.crit dnsmasq[1]: FAILED to start up
Mon Dec 19 21:30:25 2022 daemon.info vnstatd[15535]: Info: vnStat daemon 1.18 started. (pid:15535 uid:0 gid:0)
Mon Dec 19 21:30:25 2022 daemon.info vnstatd[15535]: Info: Monitoring: br-lan (2500 Mbit) eth1 (2500 Mbit) eth0 (1000 Mbit)
Mon Dec 19 21:30:29 2022 daemon.err nmbd[9304]: [2022/12/19 21:30:29.759733,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Mon Dec 19 21:30:29 2022 daemon.err nmbd[9304]:   *****
Mon Dec 19 21:30:29 2022 daemon.err nmbd[9304]:
Mon Dec 19 21:30:29 2022 daemon.err nmbd[9304]:   Samba name server MIAOHOME is now a local master browser for workgroup WORKGROUP on subnet 10.0.0.1
Mon Dec 19 21:30:29 2022 daemon.err nmbd[9304]:
Mon Dec 19 21:30:29 2022 daemon.err nmbd[9304]:   *****
Mon Dec 19 21:30:29 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Mon Dec 19 21:30:29 2022 daemon.crit dnsmasq[1]: FAILED to start up

MIAOBUSI, 2022-12-19:
我也是这样 不知道有没有解决办法

BenjaminX, 2022-12-19:
/etc/config/dhcp 求一个原始文件看看

weilai-all, 2022-12-19:
`config dnsmasq
	option domainneeded	1
	option boguspriv	1
	option filterwin2k	0  # enable for dial on demand
	option localise_queries	1
	option rebind_protection 1  # disable if upstream must serve RFC1918 addresses
	option rebind_localhost 1  # enable for RBL checking and similar services
	#list rebind_domain example.lan  # whitelist RFC1918 responses for domains
	option local	'/lan/'
	option domain	'lan'
	option expandhosts	1
	option nonegcache	0
	option cachesize	8192
	option authoritative	1
	option readethers	1
	option leasefile	'/tmp/dhcp.leases'
	option resolvfile	'/tmp/resolv.conf.d/resolv.conf.auto'
	#list server		'/mycompany.local/1.2.3.4'
	option nonwildcard	1 # bind to & keep track of interfaces
	#list interface		br-lan
	#list notinterface	lo
	#list bogusnxdomain     '64.94.110.11'
	option localservice	1  # disable to allow DNS requests from non-local subnets
	option ednspacket_max	1232
	option filter_aaaa	0
	option filter_a		0
	#list addnmount		/some/path # read-only mount path to expose it to dnsmasq`

loydcheung, 2022-12-19:
> 删掉
> 
> 	option filter_aaaa '1'
> 
> 这行

亲测有效，谢谢。

weilai-all, 2022-12-19:
> /etc/config/dhcp 求一个原始文件看看

dhcp.conf
[https://github.com/coolsnowwolf/lede/blob/b0e3285bf0e45eb47512625c54cf0e02c28e472e/package/network/services/dnsmasq/files/dhcp.conf](url)

smdx, 2022-12-19:
```
config dnsmasq
	option domainneeded	1
	option boguspriv	1
	option filterwin2k	0  # enable for dial on demand
	option localise_queries	1
	option rebind_protection 1  # disable if upstream must serve RFC1918 addresses
	option rebind_localhost 1  # enable for RBL checking and similar services
	#list rebind_domain example.lan  # whitelist RFC1918 responses for domains
	option local	'/lan/'
	option domain	'lan'
	option expandhosts	1
	option nonegcache	0
	option cachesize	8192
	option authoritative	1
	option readethers	1
	option leasefile	'/tmp/dhcp.leases'
	option resolvfile	'/tmp/resolv.conf.d/resolv.conf.auto'
	#list server		'/mycompany.local/1.2.3.4'
	option nonwildcard	1 # bind to & keep track of interfaces
	#list interface		br-lan
	#list notinterface	lo
	#list bogusnxdomain     '64.94.110.11'
	option localservice	1  # disable to allow DNS requests from non-local subnets
	option ednspacket_max	1232
	option filter_aaaa	0
	option filter_a		0
	#list addnmount		/some/path # read-only mount path to expose it to dnsmasq

config dhcp lan
	option interface	lan
	option start 	100
	option limit	150
	option leasetime	12h

config dhcp wan
	option interface	wan
	option ignore	1

```

BenjaminX, 2022-12-19:
> 

删了后不报错了，但是 DNS 查询还是失败状态。

MIAOBUSI, 2022-12-19:
> > 
> 
> 删了后不报错了，但是 DNS 查询还是失败状态。

我的dhcp里面没有option filter_aaaa '1'这行

MIAOBUSI, 2022-12-19:
打开openclash后目前电脑上能用，WIFI什么的都用不了

coolsnowwolf, 2022-12-19:
大小写写错了。。。已修复

closed 2022-12-19 by coolsnowwolf