coolsnowwolf/lede #10652: 最新源码很多第三方插件编译时缺少upx依赖

firkerword, 2022-12-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
有
## 2. 详细叙述
使用最新源码编译mosdns等插件出现缺少upx依赖，与
#10622 该条pr有关
### (1) 具体问题

A：编译出错，请教各位大佬有没有简单的解决办法😅

### (2) 路由器型号和固件版本

A：x86  5.10、5.15、6.1最新源码

### (3) 详细日志

A：Building targets
# github.com/IrineSistiana/mosdns/v4
loadinternal: cannot find runtime/cgo

/workdir/openwrt/staging_dir/host/bin/upx --lzma --best /workdir/openwrt/build_dir/target-x86_64_musl/mosdns-f130472/.go_work/build/bin/mosdns
bash: /workdir/openwrt/staging_dir/host/bin/upx: No such file or directory
make[3]: *** [Makefile:93: /workdir/openwrt/build_dir/target-x86_64_musl/mosdns-f130472/.built] Error 127
make[3]: Leaving directory '/workdir/openwrt/package/openwrt-mos/mosdns'
time: package/openwrt-mos/mosdns/compile#1.65#0.64#2.63
    ERROR: package/openwrt-mos/mosdns failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:116: package/openwrt-mos/mosdns/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

lovehackintosh, 2022-12-20:
`echo CONFIG_PACKAGE_upx=m >> .config && make defconfig`
我会提个pr修复的

firkerword, 2022-12-20:
> `echo CONFIG_PACKAGE_upx=m >> .config && make defconfig` 我会提个pr修复的

好的 感谢大佬

WYC-2020, 2022-12-20:
我理解这样也是不能修复的 ，tools 下面移除后就不会在host目录下面产生upx，有的程序压缩 硬编码了用host目录下面的，即使package/lean/upx 下面有host 最终编译也是在hostpkg/bin 这个目录下，不信你们可以试下

lovehackintosh, 2022-12-20:
@WYC-2020 https://github.com/coolsnowwolf/lede/blob/6710357cce6a08fa677c561e922df66b862f2248/package/lean/upx/Makefile#L49
upx被复制到了 staging_dir/host/bin/upx

WYC-2020, 2022-12-20:
> @WYC-2020
> 
> https://github.com/coolsnowwolf/lede/blob/6710357cce6a08fa677c561e922df66b862f2248/package/lean/upx/Makefile#L49
> 
> 
> upx被复制到了 staging_dir/host/bin/upx

那提交的时候就应该把upx作为默认库编译才对，但是默认编译会不会有多安装了一个upx，改成m那需要修改默认的config配置了

lovehackintosh, 2022-12-20:
我认为可以在需要upx压缩的包里加一条`DEPENDS:=upx/host`

WYC-2020, 2022-12-20:
可惜很多库都没这么写，哈哈，需的一个一个的去改，我怀疑当时在tools里面加入这个，也是偷懒的做法哈哈

coolsnowwolf, 2022-12-20:
当时确实是因为历史原因，有这套源代码的时候很多 go 里面还没有 upx 选项的

closed 2022-12-20 by coolsnowwolf

zhaofg, 2023-03-26:
新下载的源码，还是出现了upx编译不通过的问题。尝试了以下两种方式都是不一样不行！
1、echo CONFIG_PACKAGE_upx=m >> .config && make defconfig
2、make make package/lean/upx/compile

WARNING: Makefile 'package/feeds/packages/midisport-firmware/Makefile' has a dependency on 'fxload', which does not exist
 make[1] package/lean/upx/compile
 make[2] -C package/libs/toolchain compile
 make[2] -C package/lean/ucl compile
 make[2] -C package/libs/zlib compile
 make[2] -C package/lean/upx compile

lovehackintosh, 2023-03-26:
@zhaofg 发V=s -j1日志

zhaofg, 2023-03-26:
> @zhaofg 发V=s -j1日志

make[3]: Entering directory '/home/zcr/lede/feeds/packages/lang/golang/golang'
mkdir -p /home/zcr/lede/dl
SHELL= flock /home/zcr/lede/tmp/.go1.4-bootstrap-20171003.tar.gz.flock -c '     /home/zcr/lede/scripts/download.pl "/home/zcr/lede/dl" "go1.4-bootstrap-20171003.tar.gz" "f4ff5b5eb3a3cae1c993723f3eab519c5bae18866b5e5f96fe1102f0cb5c3e52" "" "https://dl.google.com/go/" "https://mirrors.ustc.edu.cn/golang/" "https://mirrors.nju.edu.cn/golang/"    '
mkdir -p /home/zcr/lede/dl
SHELL= flock /home/zcr/lede/tmp/.go1.19.7.src.tar.gz.flock -c '         /home/zcr/lede/scripts/download.pl "/home/zcr/lede/dl" "go1.19.7.src.tar.gz" "775bdf285ceaba940da8a2fe20122500efd7a0b65dbcee85247854a8d7402633" "" "https://dl.google.com/go/" "https://mirrors.ustc.edu.cn/golang/" "https://mirrors.nju.edu.cn/golang/"    '
make[3]: Leaving directory '/home/zcr/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.53#1.12#1.19
make[3]: Entering directory '/home/zcr/lede/package/lean/ucl'
mkdir -p /home/zcr/lede/dl
SHELL= flock /home/zcr/lede/tmp/.ucl-1.03.tar.gz.flock -c '     /home/zcr/lede/scripts/download.pl "/home/zcr/lede/dl" "ucl-1.03.tar.gz" "b865299ffd45d73412293369c9754b07637680e5c826915f097577cd27350348" "" "http://www.oberhumer.com/opensource/ucl/download"    '
make[3]: Leaving directory '/home/zcr/lede/package/lean/ucl'
time: package/lean/ucl/host-compile#0.28#0.48#0.59
make[3]: Entering directory '/home/zcr/lede/package/lean/upx'
mkdir -p /home/zcr/lede/dl
SHELL= flock /home/zcr/lede/tmp/.upx-2022-01-15.tar.gz.flock -c '        /home/zcr/lede/scripts/dl_github_archive.py --dl-dir="/home/zcr/lede/dl" --url="https://github.com/upx/upx.git" --version="a46b63817a9c6ad5af7cf519332e859f11558592" --subdir="upx-2022-01-15-a46b6381" --source="upx-2022-01-15.tar.gz" --hash="x" || (  echo "Checking out files from the git repository..."; mkdir -p /home/zcr/lede/tmp/dl && cd /home/zcr/lede/tmp/dl && rm -rf upx-2022-01-15-a46b6381 && [ \! -d upx-2022-01-15-a46b6381 ] && git clone  https://github.com/upx/upx.git upx-2022-01-15-a46b6381 && (cd upx-2022-01-15-a46b6381 && git checkout a46b63817a9c6ad5af7cf519332e859f11558592 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd upx-2022-01-15-a46b6381 && git log -1 --format='\''@%ct'\''` && rm -rf upx-2022-01-15-a46b6381/.git &&    tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c upx-2022-01-15-a46b6381 |      gzip -nc > /home/zcr/lede/tmp/dl/upx-2022-01-15.tar.gz && mv /home/zcr/lede/tmp/dl/upx-2022-01-15.tar.gz /home/zcr/lede/dl/ && rm -rf upx-2022-01-15-a46b6381; );     '
Traceback (most recent call last):
  File "/home/zcr/lede/scripts/dl_github_archive.py", line 19, in <module>
    import ssl
  File "/home/zcr/lede/staging_dir/hostpkg/lib/python3.10/ssl.py", line 99, in <module>
    import _ssl             # if we can't import it, let the error propagate
ModuleNotFoundError: No module named '_ssl'
Checking out files from the git repository...
Cloning into 'upx-2022-01-15-a46b6381'...
remote: Enumerating objects: 36934, done.
remote: Counting objects: 100% (326/326), done.
remote: Compressing objects: 100% (188/188), done.
remote: Total 36934 (delta 152), reused 256 (delta 134), pack-reused 36608
Receiving objects: 100% (36934/36934), 62.45 MiB | 1.69 MiB/s, done.
Resolving deltas: 100% (29678/29678), done.
fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592
make[3]: *** [Makefile:67: /home/zcr/lede/dl/upx-2022-01-15.tar.gz] Error 128
make[3]: Leaving directory '/home/zcr/lede/package/lean/upx'
time: package/lean/upx/host-compile#5.56#2.84#42.83
    ERROR: package/lean/upx [host] failed to build.
make[2]: *** [package/Makefile:114: package/lean/upx/host/compile] Error 1
make[2]: Leaving directory '/home/zcr/lede'
make[1]: *** [package/Makefile:110: /home/zcr/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zcr/lede'
make: *** [/home/zcr/lede/include/toplevel.mk:230: world] Error 2

lovehackintosh, 2023-03-26:
upx git仓库的commit又变了,有时间了我提个pr

zhaofg, 2023-03-26:
> upx git仓库的commit又变了,有时间了我提个pr

但是我刚刚在githut上，使用自动编译又能成功？弄不清楚情况了？

QQ40451, 2023-04-19:
UPX 失败

make[3]: Entering directory '/home/k3ok/lede/package/lean/upx'
mkdir -p /home/k3ok/lede/dl
SHELL= flock /home/k3ok/lede/tmp/.upx-2022-01-15.tar.gz.flock -c '  	 /home/k3ok/lede/scripts/dl_github_archive.py --dl-dir="/home/k3ok/lede/dl" --url="https://github.com/upx/upx.git" --version="a46b63817a9c6ad5af7cf519332e859f11558592" --subdir="upx-2022-01-15-a46b6381" --source="upx-2022-01-15.tar.gz" --hash="x" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/k3ok/lede/tmp/dl && cd /home/k3ok/lede/tmp/dl && rm -rf upx-2022-01-15-a46b6381 && [ \! -d upx-2022-01-15-a46b6381 ] && git clone  https://github.com/upx/upx.git upx-2022-01-15-a46b6381 && (cd upx-2022-01-15-a46b6381 && git checkout a46b63817a9c6ad5af7cf519332e859f11558592 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd upx-2022-01-15-a46b6381 && git log -1 --format='\''@%ct'\''` && rm -rf upx-2022-01-15-a46b6381/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c upx-2022-01-15-a46b6381 | 	gzip -nc > /home/k3ok/lede/tmp/dl/upx-2022-01-15.tar.gz && mv /home/k3ok/lede/tmp/dl/upx-2022-01-15.tar.gz /home/k3ok/lede/dl/ && rm -rf upx-2022-01-15-a46b6381; );     '
upx-2022-01-15.tar.gz: Download from https://github.com/upx/upx.git failed
upx-2022-01-15.tar.gz: Requires sha256sum for verification
Checking out files from the git repository...
Cloning into 'upx-2022-01-15-a46b6381'...
remote: Enumerating objects: 37197, done.
remote: Counting objects: 100% (589/589), done.
remote: Compressing objects: 100% (256/256), done.
remote: Total 37197 (delta 368), reused 494 (delta 329), pack-reused 36608
Receiving objects: 100% (37197/37197), 62.86 MiB | 537.00 KiB/s, done.
Resolving deltas: 100% (29896/29896), done.
fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592
Makefile:66: recipe for target '/home/k3ok/lede/dl/upx-2022-01-15.tar.gz' failed
make[3]: *** [/home/k3ok/lede/dl/upx-2022-01-15.tar.gz] Error 128
make[3]: Leaving directory '/home/k3ok/lede/package/lean/upx'
time: package/lean/upx/host-compile#11.12#8.30#125.69
    ERROR: package/lean/upx [host] failed to build.
package/Makefile:114: recipe for target 'package/lean/upx/host/compile' failed
make[2]: *** [package/lean/upx/host/compile] Error 1
make[2]: Leaving directory '/home/k3ok/lede'
package/Makefile:110: recipe for target '/home/k3ok/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/k3ok/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/k3ok/lede'
/home/k3ok/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
k3ok@k3ok:~/lede$

freedyYang, 2023-05-16:
remote: Counting objects: 100% (833/833), done.
remote: Compressing objects: 100% (309/309), done.
remote: Total 37441 (delta 575), reused 706 (delta 520), pack-reused 36608
Receiving objects: 100% (37441/37441), 63.07 MiB | 2.58 MiB/s, done.
Resolving deltas: 100% (30101/30101), done.
fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592
make[3]: *** [Makefile:67: /home/freedy/lede/dl/upx-2022-01-15.tar.gz] Error 128
make[3]: Leaving directory '/home/freedy/lede/package/lean/upx'
time: package/lean/upx/host-compile#11.11#4.22#31.88
    ERROR: package/lean/upx [host] failed to build.
make[2]: *** [package/Makefile:114: package/lean/upx/host/compile] Error 1
make[2]: Leaving directory '/home/freedy/lede'
make[1]: *** [package/Makefile:110: /home/freedy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2

wumode, 2024-01-09:
> remote: Counting objects: 100% (833/833), done. remote: Compressing objects: 100% (309/309), done. remote: Total 37441 (delta 575), reused 706 (delta 520), pack-reused 36608 Receiving objects: 100% [...]

[https://github.com/coolsnowwolf/lede/pull/10479/commits/187bb0b68642a46a8241bc11800e14b8cc0a7e38](url)按照这个commit去改package/lean/upx/Makefile里的hash