coolsnowwolf/lede #10662: 为什么闭源无线重启WIFI都要手动打开的啊

yuzhe1129, 2022-12-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

昨天编译了小米2100的固件，刷了发现，每次重启 WIFI都是默认关闭的，都是要手动打开的，能不能让重启WIFI自动打开啊，还是说闭源无线驱动都是这样的？
### (2) 路由器型号和固件版本

小米2100，

### (3) 详细日志

A：

lovehackintosh, 2022-12-21:
尝试在系统->启动项->本地启动脚本的exit 0前面添加以下脚本
``` shell
# Wireless starter and Interface checker by Kenvix <i@kenvix.com>

# Start ra unconditionally
sleep 2s
ip link set rai0 up
ip link set ra0 up
echo "Wireless interface started"

# Lan Check
lanCheck=`uci get network.lan.ifname`
if [ $? -eq 0 ]; then
    echo $lanCheck | grep rai0 > /dev/null
    if [ $? -ne 0 ]; then
        uci set network.lan.ifname="$lanCheck rai0 ra0"
        uci commit
        echo "Updated wireless config of LAN Interface"
    fi
    echo "No need to update wireless config of LAN Interface"
else
    echo "wireless config of LAN Interface check failed. Interface may renamed." >&2
fi
```

yuzhe1129, 2022-12-21:
> 

> 尝试在系统->启动项->本地启动脚的exit 0前面添加以下脚
> 
> ```shell
> # Wireless starter and Interface checker by Kenvix <i@kenvix.com>
> 
> # Start ra unconditionally
> sleep 2s
> ip link set rai0 up
> ip link set ra0 up
> echo "Wireless interface started"
> 
> # Lan Check
> lanCheck=`uci get network.lan.ifname`
> if [ $? -eq 0 ]; then
>     echo $lanCheck | grep rai0 > /dev/null
>     if [ $? -ne 0 ]; then
>         uci set network.lan.ifname="$lanCheck rai0 ra0"
>         uci commit
>         echo "Updated wireless config of LAN Interface"
>     fi
>     echo "No need to update wireless config of LAN Interface"
> else
>     echo "wireless config of LAN Interface check failed. Interface may renamed." >&2
> fi
> ```

> 尝试在系统->启动项->本地启动脚本的exit 0前面添加以下脚本
>
> ```shell
> # Wireless starter and Interface checker by Kenvix <i@kenvix.com>
>
> # Start ra unconditionally
> sleep 2s
> ip link set rai0 up
> ip link set ra0 up
> echo [...]

# Lan Check  后面的也要添加进去？这么多的吗？

lovehackintosh, 2022-12-21:
是的,复制进去就行了

yuzhe1129, 2022-12-25:
> 是的,复制进去就行了

我靠，我复制在exit 0前，也点提交了，怎么重启后 启动项变空白了，咋回事，而且WIFI也没自动打开

lovehackintosh, 2022-12-25:
ssh进去然后`wget https://github.com/lovehackintosh/lede/raw/temp/rc.local -O /etc/rc.local`

yuzhe1129, 2022-12-25:
单单执行wget https://github.com/lovehackintosh/lede/raw/temp/rc.local -O /etc/rc.local这行代码就行了？上面那一大堆还要填在启动项吗？

yuzhe1129, 2022-12-25:
> ssh进去然后`wget https://github.com/lovehackintosh/lede/raw/temp/rc.local -O /etc/rc.local`

单单执行wget https://github.com/lovehackintosh/lede/raw/temp/rc.local -O /etc/rc.local这行代码就行了？上面那一大堆还要填在启动项吗？

lovehackintosh, 2022-12-25:
不用

yuzhe1129, 2022-12-25:
> 不用

多谢，果然可以了，请问下这个怎么弄的吗，哪天你删库呢

lovehackintosh, 2022-12-25:
你可以看一下这个链接 https://github.com/lovehackintosh/lede/raw/temp/rc.local 本质就是上面那一大串代码，正常情况下在luci页面上粘贴上去就可以了

yuzhe1129, 2022-12-25:
> 你可以看一下这个链接 https://github.com/lovehackintosh/lede/raw/temp/rc.local 本质就是上面那一大串代码，正常情况下在luci页面粘贴上去就可以了

复制到启动项吗？

lovehackintosh, 2022-12-25:
是的，exit 0前面，但你已经不用再复制一遍了

closed 2023-01-03 by aiamadeus

coolmaxter, 2023-01-19:
请问你们有遇到使用mt hwnat以后驱动还需要再 重启一次的问题么? 我这遇到一样的问题,首先是wifi默认是关闭的然后重启以后还是无法访问网络直到再mtwifi页面点击重启以后才行,感觉是和hwna有关,表现就是内网是okay的, 比如DHCP地址获取,但是无法上外网