coolsnowwolf/lede #10669: 编译报错

FenghenHome, 2022-12-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：网件R7800

### (3) 详细日志

A：Applying /workdir/openwrt/target/linux/generic/pending-5.4/499-mtd-spi-nor-add-xmc-xm25qh128c.patch using plaintext: 
patching file drivers/mtd/spi-nor/spi-nor.c
Hunk #1 FAILED at 2566.
1 out of 1 hunk FAILED -- saving rejects to file drivers/mtd/spi-nor/spi-nor.c.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/generic/pending-5.4/499-mtd-spi-nor-add-xmc-xm25qh128c.patch!
make[3]: *** [Makefile:122: /workdir/openwrt/build_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/linux-5.4.228/.prepared] Error 1
make[3]: Leaving directory '/workdir/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#10.29#4.27#12.49
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

lovehackintosh, 2022-12-23:
[image]
我apply这个patch没有问题，你是不是自己加了什么patch

FenghenHome, 2022-12-23:
没有加任何patch，也没有修改任何东西，使用的云编译

lovehackintosh, 2022-12-23:
把你仓库的target/linux/generic/pending-5.4/485-mtd-spi-nor-add-xmc-xm25qh128c.patch删掉试试
它与499-mtd-spi-nor-add-xmc-xm25qh128c.patch冲突

closed 2022-12-23 by FenghenHome