coolsnowwolf/lede #10673: 禁用IPv6解析时导致dnsmasq崩溃

onlyJinx, 2022-12-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

当我在高级设置里禁用IPv6解析的时候, 会导致dnsmasq崩溃, 插件仅用了SSR Plus+,没有用其他去广告之类的东西.

[image]

dns无法解析

```
root@OpenWrt:~# nslookup baidu.com
nslookup: write to '127.0.0.1': Connection refused
nslookup: write to '::1': Connection refused
;; connection timed out; no servers could be reached

root@OpenWrt:~#
```

勾选的时候配置文件内容如下:

```
root@OpenWrt:~# cat /var/etc/dnsmasq.conf.cfg01411c
# auto-generated config file from /etc/config/dhcp
conf-file=/etc/dnsmasq.conf
dhcp-authoritative
domain-needed
localise-queries
read-ethers
enable-ubus=dnsmasq
expand-hosts
bind-dynamic
local-service
filter-aaaa
cache-size=8192
edns-packet-max=1232
domain=lan
local=/lan/
addn-hosts=/tmp/hosts
dhcp-leasefile=/tmp/dhcp.leases
dhcp-script=/usr/lib/dnsmasq/dhcp-script.sh
script-arp
resolv-file=/tmp/resolv.conf.d/resolv.conf.auto
stop-dns-rebind
rebind-localhost-ok
dhcp-broadcast=tag:needs-broadcast
conf-dir=/tmp/dnsmasq.d
user=dnsmasq
group=dnsmasq

dhcp-ignore-names=tag:dhcp_bogus_hostname
conf-file=/usr/share/dnsmasq/dhcpbogushostname.conf

bogus-priv
conf-file=/usr/share/dnsmasq/rfc6761.conf
no-dhcp-interface=eth0

root@OpenWrt:~#
```

A：

### (2) 路由器型号和固件版本

x86软路由

A：

### (3) 详细日志

A：
```
Fri Dec 23 17:36:14 2022 authpriv.notice dropbear[9611]: Password auth succeeded for 'root' from 192.168.1.100:4458
Fri Dec 23 17:45:04 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 17:45:04 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 17:45:09 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 17:45:09 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 17:45:14 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 17:45:14 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 17:45:19 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 17:45:19 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 17:45:24 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 17:45:24 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 17:45:29 2022 daemon.crit dnsmasq[1]: bad option at line 11 of /var/etc/dnsmasq.conf.cfg01411c
Fri Dec 23 17:45:29 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Dec 23 17:45:29 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
```

chiyu521, 2022-12-23:
你仔细看看 启动项也是进不去的 死循环 貌似是dnsmasq和防火墙冲突

onlyJinx, 2022-12-23:
> 你仔细看看 启动项也是进不去的 死循环 貌似是dnsmasq和防火墙冲突

请问有什么解决方法吗？其实我也没有ipv6需求，在编译的时候没有勾选那个什么ipv6helper了，但是编译出来好像还是有ipv6

chiyu521, 2022-12-23:
我早就去掉IPV6了 但是 从12.8以后编译 就自带V6了  然后就有你上面的问题了 我怀疑是DNS和防火墙升级的问题 但是找不到日志和错误的地方 目前只能等修复了~

onlyJinx, 2022-12-23:
我不需要ipv6，编译时应该如何勾选？

chiyu521, 2022-12-23:
我也没勾选啊 但是他就带啊

markvlenvision, 2022-12-23:
一样，开启ssrp，就不行了。

lovehackintosh, 2022-12-23:
https://github.com/coolsnowwolf/lede/blob/317d92dd3e0ebb07b5343ffc8ecd4db8fae36b2d/config/Config-build.in#L157
把这一行的y改成n可以不编译ipv6,但也可能会导致编译报错

onlyJinx, 2022-12-23:
> https://github.com/coolsnowwolf/lede/blob/317d92dd3e0ebb07b5343ffc8ecd4db8fae36b2d/config/Config-build.in#L157
> 
> 
> 把这一行的y改成n可以不编译ipv6,但也可能会导致编译报错

会报错, 缺少依赖, 我打算取消勾选` kmod-nft-core`发现他被其他东西依赖, 拉扯一大堆, 不知道怎么勾了

```
echo "bnx2x-firmware" >> /home/ruo/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/ruo/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#2.62#0.45#3.04
make[3]: Entering directory '/home/ruo/lede/package/kernel/linux'
mkdir -p /home/ruo/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/ruo/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ruo/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/ruo/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ruo/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-nft-core is missing dependencies for the following libraries:
nf_reject_ipv6.ko
modules/netfilter.mk:1107: recipe for target '/home/ruo/lede/bin/targets/x86/64/packages/kmod-nft-core_6.1-1_x86_64.ipk' failed
make[3]: *** [/home/ruo/lede/bin/targets/x86/64/packages/kmod-nft-core_6.1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/ruo/lede/package/kernel/linux'
time: package/kernel/linux/compile#4.38#0.57#5.04
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[2]: *** [package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/ruo/lede'
package/Makefile:110: recipe for target '/home/ruo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ruo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ruo/lede'
/home/ruo/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
ruo@build-env:~/lede$
```

[image]

coolsnowwolf, 2022-12-23:
git pull

markvlenvision, 2022-12-24:
ddns添加之后也一样，今早刚编译的。

onlyJinx, 2022-12-24:
@coolsnowwolf pull之后重新编译已经可以禁用IPv6了, 谢谢
@markvlenvision 你试试 `git pull`后重新编译看看

closed 2022-12-24 by coolsnowwolf