coolsnowwolf/lede #10675: 在treewide: sync with upstream之后无法成功编译

w311ang, 2022-12-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ERROR: tools/e2fsprogs failed to build.

### (2) 路由器型号和固件版本

A：斐讯K2P https://github.com/coolsnowwolf/lede/commit/5acc56143fb74aa122bc8fdaca5a0a6e333237b6

### (3) 详细日志

A：https://github.com/w311angg/Actions-K2P/actions/runs/3769908162/jobs/6409417214

w311ang, 2022-12-25:
#10655

lovehackintosh, 2022-12-25:
d2460dfb82ae1ee2a41a20bfe479f216056c2dcc 没有动过e2fsprogs，大概率是缓存和.config的问题，删掉重新配置试试，我亲测可以编译成功

w311ang, 2022-12-25:
@lovehackintosh https://github.com/w311angg/Actions-K2P/actions/runs/3773756857/jobs/6415441332 没有cache也是无法编译成功，并且按照默认的[.config](https://github.com/w311angg/Actions-K2P/blob/nopkg/.config)

lovehackintosh, 2022-12-25:
[image]

https://github.com/lovehackintosh/Actions-LEDE/actions/runs/3774327276/jobs/6416360674
我在ubuntu-22.04上编译成功，用的是lean仓库的依赖

把`sudo -E apt-get -qq install $(curl -fsSL git.io/depends-ubuntu-2004)`改成下面我仓库的依赖试试
https://github.com/lovehackintosh/Actions-LEDE/blob/2b8dc606c448a1731a56ae202860d5d3c34ba6d5/.github/workflows/build.yml#L43

w311ang, 2022-12-26:
经过反复测试，发现是设置了`PURE`这个环境变量导致的，删除后就一切正常了

closed 2022-12-26 by w311ang