coolsnowwolf/lede #10680: 路由重启后ipset china被清空导致所有域名都走代理（SSRP+）

onlyJinx, 2022-12-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：每一次重启都会导致ipset里的china表被清空，然后所有的域名都会被判断为“非中国大陆IP”走了代理。

[image]

之前提交过issue最后用`/usr/share/shadowsocksr/chinaipset.sh /etc/ssrplus/china_ssr.txt`弄完之后好了，但是重启后就又会丢失。

我尝试重启很多遍，最后发现是当重启完立即执行`ipset list china`时是可以看到ip列表的，但是过了几十秒后再次执行时list就会变成空白（可能这段时间里SSRP+启动过程中对ipset做了改动但是未成功？）。

[image]

下图为百度检测连通性时走了服务端代理的日志（正常的话baidu不应该走代理才对）
[image]

另外也有一个问题就是，当开启SSRP+时，路由器本身nslookup可以正常解析域名，但是ping的时候却无法ping通，这个很奇怪。关了SSRP+就没有这个问题。

```
root@OpenWrt:~# nslookup www.baidu.com
Server:         127.0.0.1
Address:        127.0.0.1:53

Non-authoritative answer:
www.baidu.com   canonical name = www.a.shifen.com
Name:   www.a.shifen.com
Address: 14.215.177.38
Name:   www.a.shifen.com
Address: 14.215.177.39

** server can't find www.baidu.com: REFUSED

root@OpenWrt:~# ping www.baidu.com
ping: bad address 'www.baidu.com'
root@OpenWrt:~#
root@OpenWrt:~#
root@OpenWrt:~#
root@OpenWrt:~#
root@OpenWrt:~# ping 14.215.177.38
PING 14.215.177.38 (14.215.177.38): 56 data bytes
64 bytes from 14.215.177.38: seq=0 ttl=56 time=7.009 ms
64 bytes from 14.215.177.38: seq=1 ttl=56 time=6.378 ms
64 bytes from 14.215.177.38: seq=2 ttl=56 time=6.373 ms
64 bytes from 14.215.177.38: seq=3 ttl=56 time=7.041 ms
^C
--- 14.215.177.38 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 6.373/6.700/7.041 ms
```

### (2) 路由器型号和固件版本

A：x86软路由

### (3) 详细日志

A：

onlyJinx, 2022-12-28:
转用passwall目前正常。

closed 2022-12-28 by onlyJinx

coldrook, 2022-12-30:
国内DNS服务器设置保持默认设置:禁用
然后一切正常