coolsnowwolf/lede #10687: package/kernel/mt76编译出错

mahakalaya, 2022-12-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译到package/kernel/mt76/compile 出错

### (2) 路由器型号和固件版本

A：N1 lede 内核6.1.1-flippy-80+

### (3) 详细日志

A：Applying ./patches/001-Revert-sync-with-upstream.patch using plaintext: 
patching file debugfs.c
patching file dma.c
Hunk #1 succeeded at 910 (offset 20 lines).
patching file eeprom.c
patching file mac80211.c
patching file mt7603/beacon.c
patching file mt7603/dma.c
patching file mt7603/mac.c
patching file mt7603/main.c
patching file mt7615/dma.c
patching file mt7615/main.c
patching file mt7615/mcu.c
patching file mt7615/soc.c
patching file mt76_connac_mcu.c
patching file mt76x02.h
patching file mt76x02_beacon.c
patching file mt76x02_mac.c
patching file mt76x02_mmio.c
patching file mt76x02_util.c
patching file mt76x2/pci.c
patching file mt7915/Kconfig
patching file mt7915/debugfs.c
patching file mt7915/dma.c
patching file mt7915/init.c
patching file mt7915/mac.c
patching file mt7915/main.c
patching file mt7915/mcu.c
patching file mt7915/soc.c
patching file mt7921/dma.c
patching file mt7921/init.c
patching file mt7921/mac.c
patching file mt7921/main.c
Hunk #2 succeeded at 685 (offset -5 lines).
Hunk #3 succeeded at 755 (offset -5 lines).
Hunk #4 succeeded at 786 (offset -5 lines).
Hunk #5 succeeded at 1633 (offset -5 lines).
Hunk #6 succeeded at 1661 (offset -5 lines).
patching file mt7921/mcu.c
patching file mt7921/pci_mac.c
patching file tx.c
patching file usb.c

Applying ./patches/002-Revert-wifi-mt76-mt7921-introduce-chanctx-support.patch using plaintext: 
patching file mt7921/init.c
patching file mt7921/mac.c
patching file mt7921/main.c
Hunk #1 succeeded at 851 (offset -5 lines).
Hunk #2 succeeded at 884 (offset -5 lines).
Hunk #3 succeeded at 1644 (offset -5 lines).
Hunk #4 FAILED at 1681.
Hunk #5 succeeded at 1827 (offset -8 lines).
1 out of 5 hunks FAILED -- saving rejects to file mt7921/main.c.rej
patching file mt7921/mt7921.h
patching file mt7921/pci.c
patching file mt7921/sdio.c
patching file mt7921/usb.c
Patch failed!  Please fix ./patches/002-Revert-wifi-mt76-mt7921-introduce-chanctx-support.patch!
make[2]: *** [Makefile:529: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/mt76-2022-12-22-5b509e80/.prepared_08e68a273deb81912730aebb845e8ef2_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[2]: Leaving directory '/home/simon/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#0.42#0.64#0.92
    ERROR: package/kernel/mt76 failed to build.
make[1]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[1]: Leaving directory '/home/simon/lede'
make: *** [/home/simon/lede/include/toplevel.mk:231：package/kernel/mt76/compile] 错误 2

CJLUGaoXinyu, 2022-12-27:
同报错

lovehackintosh, 2022-12-27:
`make package/kernel/mt76/clean && wget https://github.com/coolsnowwolf/lede/pull/10689.patch && git am 10689.patch`
再试试

CJLUGaoXinyu, 2022-12-27:
> `make package/kernel/mt76/clean && wget https://github.com/coolsnowwolf/lede/pull/10689.patch && git am 10689.patch` 再试试

可以了，谢谢大佬

PR #10689 (2022-12-27): mt76: refresh patches
Fixes: #10687 

Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
closed 2022-12-27 by aiamadeus