coolsnowwolf/lede #10688: 极路由3  编译报错

selfcan, 2022-12-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：极路由  5861

### (3) 详细日志

A："mipsel-openwrt-linux-gcc"   -fvisibility-inlines-hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -std=gnu++20 -fPIC -pthread -O3 -finline-functions -Wno-inline -Wall -fvisibility=hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -DBOOST_ALL_NO_LIB=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK=1 -DNDEBUG  -I"."  -c -o "bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/utf8_codecvt_facet.o" "libs/program_options/src/utf8_codecvt_facet.cpp"

...failed gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/utf8_codecvt_facet.o...
gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/positional_options.o
mipsel-openwrt-linux-gcc: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?

    "mipsel-openwrt-linux-gcc"   -fvisibility-inlines-hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -std=gnu++20 -fPIC -pthread -O3 -finline-functions -Wno-inline -Wall -fvisibility=hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -DBOOST_ALL_NO_LIB=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK=1 -DNDEBUG  -I"."  -c -o "bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/positional_options.o" "libs/program_options/src/positional_options.cpp"

...failed gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/positional_options.o...
gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/convert.o
mipsel-openwrt-linux-gcc: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?

    "mipsel-openwrt-linux-gcc"   -fvisibility-inlines-hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -std=gnu++20 -fPIC -pthread -O3 -finline-functions -Wno-inline -Wall -fvisibility=hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -DBOOST_ALL_NO_LIB=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK=1 -DNDEBUG  -I"."  -c -o "bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/convert.o" "libs/program_options/src/convert.cpp"

...failed gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/convert.o...
gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/winmain.o
mipsel-openwrt-linux-gcc: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?

    "mipsel-openwrt-linux-gcc"   -fvisibility-inlines-hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -std=gnu++20 -fPIC -pthread -O3 -finline-functions -Wno-inline -Wall -fvisibility=hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -DBOOST_ALL_NO_LIB=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK=1 -DNDEBUG  -I"."  -c -o "bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/winmain.o" "libs/program_options/src/winmain.cpp"

...failed gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/winmain.o...
gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/split.o
mipsel-openwrt-linux-gcc: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?

    "mipsel-openwrt-linux-gcc"   -fvisibility-inlines-hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -std=gnu++20 -fPIC -pthread -O3 -finline-functions -Wno-inline -Wall -fvisibility=hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0=boost_1_81_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DBOOST_NO_FENV_H -fPIC -ffunction-sections -fdata-sections -flto  -DBOOST_ALL_NO_LIB=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK=1 -DNDEBUG  -I"."  -c -o "bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/split.o" "libs/program_options/src/split.cpp"

...failed gcc.compile.c++ bin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden/split.o...
...skipped <pbin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden>libboost_program_options.so.1.81.0 for lack of <pbin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden>cmdline.o...
...skipped <p/workdir/openwrt/build_dir/target-mipsel_24kc_musl/boost_1_81_0/ipkg-install/lib>libboost_program_options.so.1.81.0 for lack of <pbin.v2/libs/program_options/build/gcc-8.4.0/release/threading-multi/visibility-hidden>libboost_program_options.so.1.81.0...
[... cut ...]
time: package/feeds/packages/boost/compile#17.38#0.28#17.58
    ERROR: package/feeds/packages/boost failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/boost/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

selfcan, 2022-12-28:
> 你是不是也开了passwall插件，我只要一启用就报错

 对，开启了SSR

wuzhuang123, 2022-12-29:
报错原因：error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?
把feeds/packages/libs/boost/Makefile文件的-std=gnu++20的20改成2a

xpfoxs, 2022-12-30:
> 

云编译要怎么改？

ImIvey, 2022-12-30:
> > 
> 
> 云编译要怎么改？

我已经改好了，测试小米r4a可以正常编译，fork我的仓库就行了
https://github.com/ImIvey/lede
https://github.com/ImIvey/packages
改动
https://github.com/ImIvey/lede/commit/d91d71fca3bdc45143284284924b20a357c1aee1
https://github.com/ImIvey/packages/commit/00b2f0b02bf427cf7aebe5dcb64580eb67bdf2b1

xpfoxs, 2022-12-30:
我试试

closed 2023-01-01 by coolsnowwolf