coolsnowwolf/lede #10692: macos13.1下编译出错

bosense, 2022-12-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译出错

### (2) 路由器型号和固件版本

A：路由器华硕map-ac2200

### (3) 详细日志

A：make[3]: Entering directory '/Volumes/OpenWrt/lede/package/lean/upx'
UPX_UCLDIR=/Volumes/OpenWrt/lede/staging_dir/host make -C /Volumes/OpenWrt/lede/build_dir/hostpkg/upx-3.96/src LDFLAGS="-L/Volumes/OpenWrt/lede/staging_dir/host/lib -L/Volumes/OpenWrt/lede/staging_dir/hostpkg/lib -L/Volumes/OpenWrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/lib" CXX="g++"
make[4]: Entering directory '/Volumes/OpenWrt/lede/build_dir/hostpkg/upx-3.96/src'
g++ -I/Volumes/OpenWrt/lede/staging_dir/host/include -O2 -fno-delete-null-pointer-checks -fno-strict-aliasing -fwrapv -funsigned-char -Wall -W -Wcast-align -Wcast-qual -Wmissing-declarations -Wpointer-arith -Wshadow -Wvla -Wwrite-strings -Werror -o p_wcle.o -c p_wcle.cpp
p_wcle.cpp:739:27: error: variable 'n' set but not used [-Werror,-Wunused-but-set-variable]
    unsigned count,object,n,r;
                          ^
1 error generated.
make[4]: *** [Makefile:92: p_wcle.o] Error 1
make[4]: Leaving directory '/Volumes/OpenWrt/lede/build_dir/hostpkg/upx-3.96/src'
make[3]: *** [Makefile:62: /Volumes/OpenWrt/lede/build_dir/hostpkg/upx-3.96/.built] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/lede/package/lean/upx'
time: package/lean/upx/host-compile#0.48#0.31#0.83
    ERROR: package/lean/upx [host] failed to build.
make[2]: *** [package/Makefile:114: package/lean/upx/host/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/OpenWrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:231: world] Error 2

lovehackintosh, 2022-12-28:
方法一：更新到最新版本 `make package/lean/upx/clean && wget https://github.com/lovehackintosh/lede/commit/8a6ef953a9ecdc2e01c176af0812f71684c0bef2.patch && git am 8a6ef953a9ecdc2e01c176af0812f71684c0bef2.patch`
方法二：src/p_wcle.cpp里添加一行`UNUSED(n);`
方法三：TARGET_CFLAGS加一条`-Wno-error=unused-but-set-variable`

bosense, 2022-12-28:
编译又出了新错误，再验证吧

commit 331a9f65 (2022-09-27): scripts/download.pl: fix downloads with wget

Several users of wget for downloads (curl is not available in the
system) have reported broken download functionality:

 wget --tries=5 --timeout=20 --output-document=-  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.142.tar.xz
 http://: Invalid host name.

Thats all happening due to '' was passed as an argument, which got later
expanded to http://.

In the context of a list constructor '' is not nothing, it is an empty
string element.  So fix it by using () as it will yield "nothing" and
thus not introduce an empty string element.

Fixes: [... cut]
closed 2022-12-28 by bosense