coolsnowwolf/lede #10693: openwrt ipv4 和 ipv6 域名解析优先级调换咨询

copperfieldd, 2022-12-28:
大佬好，NAS ddns需要用到ipv6公网，目前开启路由ipv6域名解析后pc和手机上外网速度受严重影响了，一个域名同时具备ipv4和ipv6默认会先解析ipv6，请问可以在路由上调换ipv6和ipv4的解析优先级吗？（需要优先解析ipv4，ipv4不能用再解析ipv6）linux下可以修改/etc/gai.conf文件实现，opwrt里面未找到相应配置
label ::1/128 0
label ::/0 1
label 2002::/16 2
label ::/96 3
label ::ffff:0:0/96 4
precedence ::1/128 50
precedence ::/0 40
precedence 2002::/16 30
precedence ::/96 20
precedence ::ffff:0:0/96 10
[image]

hwliu11, 2022-12-28:
smartdns有双栈优选，dnsmasq不会设置

minlang112, 2022-12-29:
vim /etc/dnsmasq.conf
增加条目并保存后，重启dnsmasq

# 关闭ipv6解析，以下为举例
address=/bilibili.com/::
address=/netflix.com/::

jxph1123, 2022-12-29:
安装个smartdns吧，挺好用，双栈优选，也可以屏蔽IPV6

coolsnowwolf, 2026-09-05:
这是关于 glibc `/etc/gai.conf` 地址选择策略的配置咨询；OpenWrt 使用 musl 与其 resolver 行为，不能通过通用 LuCI/包补丁直接套用 gai.conf，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf