coolsnowwolf/lede #10694: 编译出错，每次更新都有这个问题

286700234, 2022-12-28:
make[3]: *** [Makefile:513: /home/lyus/lede/build_dir/target-mipsel_24kc_musl/boost_1_81_0/.built] Error 1
make[3]: Leaving directory '/home/lyus/lede/feeds/packages/libs/boost'
time: package/feeds/packages/boost/compile#8.58#0.62#10.27
    ERROR: package/feeds/packages/boost failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/boost/compile] Error 1
make[2]: Leaving directory '/home/lyus/lede'
make[1]: *** [package/Makefile:110: /home/lyus/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lyus/lede'
make: *** [/home/lyus/lede/include/toplevel.mk:231：world] 错误 2

lovehackintosh, 2022-12-28:
日志没发全

wuzhuang123, 2022-12-29:
boost更新了,修改feeds/packages/libs/boost/Makefile文件，找到-std=gnu++20，把20改成2a

yuzhe1129, 2022-12-29:
> boost更新了,修改feeds/packages/libs/boost/Makefile文件，找到-std=gnu++20，把20改成2a

我也是这个错误改了2a一样不行啊

286700234, 2022-12-29:
我改了之后可以了

yuzhe1129, 2022-12-29:
> 我改了之后可以了

feeds源码地址 678910 你有去掉#吗？

286700234, 2022-12-29:
没有

closed 2022-12-30 by coolsnowwolf