coolsnowwolf/lede #10695: mt闭源编译出错

bauw2008, 2022-12-28:
1.关于你要提交的问题
 没有类似的issue

 2. 详细叙述
(1) 把闲置的e8820s用官方5.15源码搭配mt闭源驱动，编译不通过ERROR: modpost: "wireless_send_event"。
(2) 路由器型号和固件版本
路由器:ztc-e8820s
固件版本:openwr master
KERNEL_PATCHVER:=5.15

 (3) 详细日志
make[3]: Entering directory '/home/bauw/openwrt23/package/op/mt/drivers/mt7603e'
rm -f /home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/.built
touch /home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/.built_check
make -C "/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.85" V=1 CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" M="/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap"   CONFIG_RLT_WIFI=y CONFIG_MT_WIFI_PATH="rlt_wifi" CONFIG_FIRST_IF_MT7603E=y   CONFIG_FIRST_IF_EEPROM_FLASH=y CONFIG_RT_FIRST_CARD_EEPROM="flash"  CONFIG_MULTI_INF_SUPPORT=y CONFIG_WSC_INCLUDED=y  CONFIG_DOT11N_DRAFT3=y  CONFIG_DOT11W_PMF_SUPPORT=y         CONFIG_NEW_RATE_ADAPT_SUPPORT=y     CONFIG_RTMP_FLASH_SUPPORT=y    CONFIG_ATE_SUPPORT=y CONFIG_RT2860V2_AP_V24_DATA_STRUCTURE=y      CONFIG_UAPSD=y            CONFIG_RTMP_PCI_SUPPORT=y   CONFIG_RLT_AP_SUPPORT=y CONFIG_RALINK_MT7603E=y CONFIG_MT_MAC=y   CONFIG_MBSS_SUPPORT=y   CONFIG_APCLI_SUPPORT=y CONFIG_MULTI_APCLI_SUPPORT=y  CONFIG_MAC_REPEATER_SUPPORT=y     CONFIG_LLTD_SUPPORT=y              modules
make[4]: Entering directory '/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.85'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap \
single-build= \
need-builtin=1 need-modorder=1
sh ./scripts/modules-check.sh /home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/modules.order
make -f ./scripts/Makefile.modpost
  sed 's/\.ko$/\.o/' /home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/modules.order | scripts/mod/modpost    -o /home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/Module.symvers -e -i Module.symvers   -T -
ERROR: modpost: "wireless_send_event" [/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/mt7603e.ko] undefined!
scripts/Makefile.modpost:133: recipe for target '/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/Module.symvers' failed
make[5]: *** [/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/Module.symvers] Error 1
make[5]: *** Deleting file '/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/mt7603_wifi_ap/Module.symvers'
Makefile:1818: recipe for target 'modules' failed
make[4]: *** [modules] Error 2
make[4]: Leaving directory '/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.85'
Makefile:62: recipe for target '/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/.built' failed
make[3]: *** [/home/bauw/openwrt23/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7603_wifi/.built] Error 2
make[3]: Leaving directory '/home/bauw/openwrt23/package/op/mt/drivers/mt7603e'
time: package/op/mt/drivers/mt7603e/compile#2.25#1.22#3.39
    ERROR: package/op/mt/drivers/mt7603e failed to build.
package/Makefile:114: recipe for target 'package/op/mt/drivers/mt7603e/compile' failed
make[2]: *** [package/op/mt/drivers/mt7603e/compile] Error 1
make[2]: Leaving directory '/home/bauw/openwrt23'
package/Makefile:110: recipe for target '/home/bauw/openwrt23/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/bauw/openwrt23/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bauw/openwrt23'
/home/bauw/openwrt23/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

closed 2022-12-30 by coolsnowwolf

bauw2008, 2023-05-25:
> 

明天试试，感谢回复

ewangsoft, 2023-10-28:
> > 
> 
> 明天试试，感谢回复

这个问题解决了吗，如何解决的，我也遇到了。

bauw2008, 2023-10-28:
> > > 
> > 
> > 
> > 明天试试，感谢回复
> 
> 这个问题解决了吗，如何解决的，我也遇到了。
不能解决，用回5.4内核版本吧

ewangsoft, 2023-10-28:
> > > > 
> > > 
> > > 
> > > 明天试试，感谢回复
> > 
> > 
> > 这个问题解决了吗，如何解决的，我也遇到了。
> > 不能解决，用回5.4内核版本吧

前一段，我用openwrt-main（5.15内核）还是可以的，后来我切换到openwrt-22.03（5.10内核）编译成功。最近想再切换到openwrt-main，发现出问题了。这里面gcc版本发生变化，内核也有新的升级，是不是这些变化造成不清楚。这个mt7615d估计没开发者愿意再修补了吧