coolsnowwolf/lede #10704: OrangePi R1 PLUS  LTS编译失败！

QYKing233, 2022-12-31:
## 1.关于你要提交的问题

* [x] 没有类似的issue

## 2. 详细叙述

A：OrangePi R1 PLUS  LTS编译失败

错误处日志：
2022-12-30T05:28:51.2300196Z make[8]: *** No rule to make target 'drivers/net/phy/motorcomm.o', needed by 'drivers/net/phy/built-in.a'.  Stop.
2022-12-30T05:28:51.2303264Z make[7]: *** [scripts/Makefile.build:503: drivers/net/phy] Error 2
2022-12-30T05:28:51.2303744Z make[6]: *** [scripts/Makefile.build:503: drivers/net] Error 2
2022-12-30T05:28:51.2304110Z make[5]: *** [Makefile:1837: drivers] Error 2
2022-12-30T05:28:51.2305273Z make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161'
2022-12-30T05:28:51.2305995Z make[4]: *** [Makefile:26: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.modules] Error 2
2022-12-30T05:28:51.2306839Z make[3]: *** [Makefile:11: compile] Error 2
2022-12-30T05:28:51.2314469Z make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
2022-12-30T05:28:51.2315229Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2022-12-30T05:28:51.2315606Z time: target/linux/compile#632.32#58.47#690.30
2022-12-30T05:28:51.2317412Z     ERROR: target/linux failed to build.
2022-12-30T05:28:51.2320303Z make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
2022-12-30T05:28:51.2325600Z make[2]: Leaving directory '/workdir/openwrt'
2022-12-30T05:28:51.2333192Z make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
2022-12-30T05:28:51.2413577Z make[1]: Leaving directory '/workdir/openwrt'
2022-12-30T05:28:51.2475217Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2022-12-30T05:28:51.2701391Z ##[error]Process completed with exit code 2.

详细日志在附件Compile the firmware.txt

### (2) 路由器型号和固件版本

设备：OrangePi R1 PLUS  LTS
固件版本：20221230主线版本

### (3) 详细日志

[Compile the firmware.txt](https://github.com/coolsnowwolf/lede/files/10326653/Compile.the.firmware.txt)

coolsnowwolf, 2023-01-01:
不再支持 5.10 内核

QYKing233, 2023-01-01:
> 不再支持 5.10 内核

好的老大，前段时间5.15内核的yt8531c不识别，我再新编译一个看看。

closed 2023-01-02 by QYKing233