coolsnowwolf/lede #10705: 一直都编译成功的.config遇到kmod-fast-classifier和kmod-shortcut-fe-cm冲突

gztingting, 2022-12-31:
云编译X84-64, 以前每次编译都成功的.config，这几天遇到kmod-fast-classifier和kmod-shortcut-fe-cm冲突，尝试过在.config里面设置# CONFIG_PACKAGE_kmod-shortcut-fe-cm is not set或者# CONFIG_PACKAGE_kmod-fast-classifier is not set，都编译失败，貌似选了Luci-app-TurboACC自动就把这两个插件带上，请教各位大神，还需要TurboACC的话，如何设置？

出错部分日志如下：
* check_conflicts_for: The following packages conflict with kmod-fast-classifier:
Configuring luci-app-wol.
Configuring luci-app-jd-dailybonus.
 * check_conflicts_for: 	kmod-shortcut-fe-cm * 
Configuring luci-app-samba4.
 * opkg_install_cmd: Cannot install package kmod-fast-classifier.
Configuring luci-i18n-wol-zh-cn.
 * check_conflicts_for: The following packages conflict with kmod-fast-classifier:
Configuring luci-app-upnp.
 * check_conflicts_for: 	kmod-shortcut-fe-cm * 
Configuring uhttpd-mod-ubus.
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
Configuring rpcd-mod-rrdns.
make[2]: *** [package/Makefile:70: package/install] Error 255
Configuring luci.
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
Configuring default-settings.
Configuring kmod-fs-nfs.
Configuring luci-i18n-dockerman-zh-cn.
Configuring ip6tables.
Configuring luci-i18n-samba4-zh-cn.
Configuring kmod-sound-i8x0.
Configuring luci-i18n-ttyd-zh-cn.
Configuring htop.
Configuring kmod-iavf.
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

starxu1971, 2022-12-31:
x86不要选kmod-shortcut-fe-cm没用的。那是高通芯片用的。

starxu1971, 2022-12-31:
是在luCI里不选shortcut-fe-cm

starxu1971, 2022-12-31:
在TurboACC选项里只选第一个，我就是这样编译完成的，而且现在不会两个都选选了第一个 kmod-shortcut-fe-cm 这个就不显示也不会编译

gztingting, 2022-12-31:
> 在TurboACC选项里只选第一个，我就是这样编译完成的，而且现在不会两个都选选了第一个 kmod-shortcut-fe-cm 这个就不显示也不会编译

我也是这样选的，结果还是编译失败

starxu1971, 2022-12-31:
那就不清楚请大佬们看看怎么办。或者重新拉取源码试试

Plutonium141, 2022-12-31:
使用rm -rf ./tmp && rm -rf .config删除掉原配置 然后再用make menuconfig重新配置一下

ZGNCZMY, 2023-01-01:
内核模块那边的网络支持把shortcut-fe-cm取消就可以

closed 2023-01-01 by gztingting