coolsnowwolf/lede #10711: drivers mtk tm7663

bricco1981, 2023-01-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
can you please add mtk proprietary drivers mt7663 
sorry it's not an issues, thanks

aiamadeus, 2023-01-03:
Unfortunately, we do not have the corresponding driver and router, you can only find it yourself.

bricco1981, 2023-01-03:
i found yhis repo, tree version of the drivers but that fail to comipe, and i cannot understand why.
https://github.com/DragonBluep/mtk-openwrt-sdk-v4.1.0.4/tree/master/wifi_drv

bricco1981, 2023-01-05:
@aiamadeus we need a way to make it work,under my76 that is the worst radio, the worst 5ghz driver i have ever seen.
it's a week that i'm trying to compile but that fail.
we need make it

closed 2023-02-18 by bricco1981

ravichaudhary133, 2024-01-04:
@bricco1981 were you able to compile the proprietary drivers?!

bricco1981, 2024-01-04:
No, my free time Is very limited now