coolsnowwolf/lede #10714: 操作Turbo AAC后频繁报错

bi4tms, 2023-01-02:
拉取最新源码编译的x86的6.1内核固件，在Turbo AAC中进行任意操作后系统日志频繁报错：
Tue Jan  3 06:43:11 2023 kern.alert kernel: [  360.114426] fast-classifier: shutting down
Tue Jan  3 06:43:11 2023 kern.alert kernel: [  360.296802] fast-classifier: starting up
Tue Jan  3 06:43:11 2023 kern.alert kernel: [  360.300820] fast-classifier: registered
Tue Jan  3 06:43:15 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:20 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:25 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:30 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:35 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:40 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:45 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:50 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:43:55 2023 daemon.err uhttpd[2517]: sh: out of range
Tue Jan  3 06:44:00 2023 daemon.err uhttpd[2517]: sh: out of range
另外，采用115网盘下载文件无法跑满千兆带宽，在Turbo AAC中进行任意操作后可以跑满千兆带宽，但重启系统后再次出现无法跑满千兆带宽的情况
@coolsnowwolf

Fyred, 2023-03-06:
Turbo AAC有bug，别进它的配置页面就不会报错

coolsnowwolf, 2026-09-05:
旧版 TurboACC 页面触发 fast-classifier 重载后持续 `sh: out of range`，当前 TurboACC 已改用新的 fastpath 配置与 init 脚本，原日志没有当前版本复现条件。

closed 2026-09-05 by coolsnowwolf

Fyred, 2026-09-06:
> 旧版 TurboACC 页面触发 fast-classifier 重载后持续 `sh: out of range`，当前 TurboACC 已改用新的 fastpath 配置与 init 脚本，原日志没有当前版本复现条件。

大佬牛逼，3年了终于修复此bug！