coolsnowwolf/lede #10716: 玩客云编译出错

Rutenioum, 2023-01-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

编译过程中报错,在网上没搜到类似的错误,请问如何解决?
### (2) 路由器型号和固件版本
玩客云
A：

### (3) 详细日志

A：
ld FLAGS
========
-nostartfiles -Map=/home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/binaries/sama5d3_xplained-nandflashboot-uboot-3.9.3.map --cref -static -T elf32-littlearm.lds --gc-sections -Ttext 0x300000

  AS        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/crt0_gnu.S
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/main.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/board/sama5d3_xplained/sama5d3_xplained.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/lib/string.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/lib/eabi_utils.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/lib/div.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/lib/consttime_memequal.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/debug.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/at91_slowclk.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/common.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/at91_pio.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/pmc/common.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/pmc/periph-clk-sam9x5.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/pmc/utmi-clk.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/pmc/pll-clk.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/at91_pit.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/at91_wdt.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/at91_usart.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/at91_rstc.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/ddramc.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/nandflash.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/pmecc.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/pm.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/at91_twi.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/act8865.c
  CC        /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/driver/macb.c
mkdir -p /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/binaries
  LD        sama5d3_xplained-nandflashboot-uboot-3.9.3.elf
arm-openwrt-linux-muslgnueabi-ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)
make[4]: *** [Makefile:328: /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/binaries/sama5d3_xplained-nandflashboot-uboot-3.9.3.bin] Error 1
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3'
make[3]: *** [Makefile:145: /home/ubuntu/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/package/boot/at91bootstrap'
time: package/boot/at91bootstrap/sama5d3_xplainednf_uboot/compile#2.85#0.31#3.14
    ERROR: package/boot/at91bootstrap failed to build (build variant: sama5d3_xplainednf_uboot).
make[2]: *** [package/Makefile:116: package/boot/at91bootstrap/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-arm_cortex-a5+vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2

aiamadeus, 2023-01-03:
你这都编到北极去了，不看看自己选对了没

Rutenioum, 2023-01-03:
@aiamadeus 
[image]

玩客云就是这么选没错的

aiamadeus, 2023-01-03:
... 真佩服你的自信，明明是 meson
少看网上那些只取 rootfs 的垃圾教程

ZqinKing, 2023-01-04:
目前这个编译出来不支持emmc，没啥大用。

aiamadeus, 2023-01-04:
> 目前这个编译出来不支持emmc，没啥大用。

那你行你来搞呗

ZqinKing, 2023-01-05:
你是不是姓喷名子，叫喷子？！说话阴阳怪气的，没爹妈教你怎么说话？

Rutenioum, 2023-01-05:
他话说的是真难听,但是也确实帮我解决了问题,我是跑再docker里面用的,所以问题算是解决了
他还把自己的发言全删了么? 佩服!

aiamadeus, 2023-01-05:
不要以伸手浪费别人时间为荣

closed 2023-01-05 by aiamadeus

ZqinKing, 2023-01-05:
你可真能装，人家没弄明白不能问了？什么叫以浪费别人时间为荣？你是警察啊还是法官，这就给定性了？傻逼玩意

yifang5775, 2023-04-20:
怎么解决的，我也一样的报错