coolsnowwolf/lede #10718: 每5天电信释放ip后，自动重新拨号，WAN获取不到IPv6 PD地址然后断网，要手动点重新拨号才能正常上网，有什么很好的解决办法吗

eram9, 2023-01-03:
每5天电信释放ip后，自动重新拨号，WAN获取不到IPv6 PD地址然后断网，要手动点重新拨号才能正常上网，有什么很好的解决办法吗

aiamadeus, 2023-01-03:
自己设置一下 PPPoE 拨号 LCP 响应那些

starryloki, 2023-01-03:
遇到了同样的问题，我这边自动重新拨号后没有自动添加 0.0.0.0/0 的默认路由导致无法上网，手动重新拨号才能正常上网

longxisutdio, 2023-01-04:
我WAN口都是禁用IPv6

starryloki, 2023-01-04:
> 我WAN口都是禁用IPv6

问一下wan口禁用ipv6会影响内网的ipv6吗？

geniusyjiang, 2023-01-06:
设置定期任务每5天（或者4天）运行“/sbin/ifup wan”，或者重启路由器都可以解决这个问题。

nkcfc, 2023-01-11:
这难道是一个bug，我最近也遇到这种问题了，移动宽带。

feifan1113, 2023-01-13:
联通也有这样情况 不过是隔3天

MageDelfador, 2023-01-14:
WAN设置里  高级设置  LCP 响应故障阈值  填个非0值  比如5  然后保存

longxisutdio, 2023-01-16:
> > 我WAN口都是禁用IPv6
> 
> 问一下wan口禁用ipv6会影响内网的ipv6吗？

我没太注意。。。。。。平时我都不用Ip6 一长串地址 看了头晕

longxisutdio, 2023-01-16:
> 设置定期任务每5天（或者4天）运行“/sbin/ifup wan”，或者重启路由器都可以解决这个问题。

我的感觉是网卡会掉驱动一样  我从后台日志看 网卡直接down了  然后直接又up 我也不知道是什么原因  然后换了普通的家用路由器 就不会自动断网  过几天假期再测试一下看看 感谢各位回复

sss132919, 2023-01-25:
遇到了同样的问题，请问现在怎么样了啊

eram9, 2023-01-26:
> 遇到同样的问题，请问现在怎么样了啊

自己编译官方源码的固件解决了

feifan1113, 2023-01-26:
> WAN设置里 高级设置 LCP 响应故障阈值 填个非0值 比如5 然后保存

这样设置不会断网了，但是设备过了租期就没法自动获取ipv6 设备要重连才能重新获取v6地址，不知道是什么原因

MageDelfador, 2023-01-26:
> > WAN设置里 高级设置 LCP 响应故障阈值 填个非0值 比如5 然后保存
> 
> 这样设置不会断网了，但是设备过了租期就没法自动获取ipv6 设备要重连才能重新获取v6地址，不知道是什么原因

我猜可能是ipv6相关的软件包装的不对？
luci-proto-ipv6
odhcp6c
odhcpd-ipv6only
ip6tables
我装的这4个，没有任何问题

xkelvinx666, 2023-01-28:
随手写的自动restart shell，貌似是linux 6内核有这个问题，以前5.15的没出现过

放到 /etc/hotplug.d/iface/99-ppp-wan
`
#!/bin/sh
[ "$INTERFACE" = wan ] || exit 0
[ "$ACTION" = ifdown ] || exit 0
echo "$(date) wan interface down, auto restarting" >> /var/log/restart-wan.log 2>&1
sleep 1s
ifup wan
`

顺便passwall开启ipv6可能也会有问题，可以自己在最后加/etc/init.d/passwall restart

rrkai, 2023-02-16:
用个WatchCat

yanxin152133, 2023-02-20:
同上

MageDelfador, 2023-02-20:
我猜。。是不是WAN拨号设置里，“获取IPV6地址”选择“手动”了？
这里最好选择“自动”，用它自动生成的wan_6。不要选择手动然后自己配置wan6，这样会有各种问题
断网的看这个

> WAN设置里 高级设置 LCP 响应故障阈值 填个非0值 比如5 然后保存

longxisutdio, 2023-02-21:
我的wan6是关闭掉的

MageDelfador, 2023-02-21:
还有就是把WAN设置里的“使用默认网关”钩上，其他接口（如果有）不要勾。
还没解决的可以把自己的设置贴出来，先看看设置有没有问题

longxisutdio, 2023-03-09:
> 计划任务中添加
> 0 4 * * * /sbin/ifup wan
> 每天凌晨4点重启wan口，重新拨号

他这个可不是每天只断一次 有时几分钟就断一次 有时会连续断  设备已经被我出掉了 又换回了锐捷的路由器。。。。。。。

lialosiu, 2023-06-12:
我用的原版openwrt22.03.5也有这个问题，搜索到这里来了
目前来看是因为pppoe被踢下线重拨时，wan_6还在，然后新的pppoe连上之后wan_6才断，导致的默认路由没了，而且新的wan_6也不会出来，直接无了

ATang007ZH, 2024-03-23:
> 我用的原版openwrt22.03.5也有这个问题，搜索到这里来了 目前来看是因为pppoe被踢下线重拨时，wan_6还在，然后新的pppoe连上之后wan_6才断，导致的默认路由没了，而且新的wan_6也不会出来，直接无了

我也碰到了，应该是软件逻辑有问题，wan_6是wan生成的虚拟接口，在wan口断线后应该立刻被清除。
现在的问题是wan_6还没来得及清除，wan已经又连上了，就出问题了。

Elec4d0, 2024-04-06:
解决了，这是一个配置问题。
参考了https://www.right.com.cn/forum/thread-8289799-1-1.html后，

在wan口上   获取ipv6地址 选择 已禁用（最好也把v6内置管理关了？ 这个作用不明确）
[image]

然后在wan6上，请求ipv6地址 设置为 disabled。 指定长度设置为自动

实测这样在手动刷新wan口模拟自动拨号后，第一时间就拿到了ipv6-pd
[image]

建议看到邮件的各位试试看

sss132919, 2024-04-06:
> 解决了，这是一个配置问题。 参考了https://www.right.com.cn/forum/thread-8289799-1-1.html 后，
> 
> 在wan口上 获取ipv6地址 选择 已禁用（最好也把v6内置管理关了？ 这个作用不明确） ![image](https://private-user-images.githubusercontent.com/56445378/320163848-5475524a-a391-432d-9875-8ae70957154b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzODgwNzEsIm5iZiI6MTcxMjM4Nzc3MSwicGF0aCI6Ii81NjQ0NTM3OC8zMjAxNjM4NDgtNTQ3NTUyNGEtYTM5MS00MzJkLTk4NzUtOGFlNzA5NTcxNTRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA2VDA3MTYxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzYjkwYzNmY2RlNTVkN2QzMmY2ZGE2Y2I1NjFhODgxZDAxODE5MzcyYzg0NGJhOTZiZDVlNWQ3ZmM5ZGU5ZDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pEJvtD_b7uWAFQtK2jutTcVwsWdzjQmkvm_Dllf1eZ0)
> 
> 然后在wan6上，请求ipv6地址 设置为 disabled。 指定长度设置为自动
> 
> 实测这样在手动刷新wan口模拟自动拨号后，第一时间就拿到了ipv6-pd ![image](https://private-user-images.githubusercontent.com/56445378/320164013-87cbe58d-4f8b-488e-928d-f6fed44d6ebd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzODgwNzEsIm5iZiI6MTcxMjM4Nzc3MSwicGF0aCI6Ii81NjQ0NTM3OC8zMjAxNjQwMTMtODdjYmU1OGQtNGY4Yi00ODhlLTkyOGQtZjZmZWQ0NGQ2ZWJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA2VDA3MTYxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViNWU3NjEwODIxZGVjNTE4YzQ1MTQ2ZTNjN2Y0MmNlNjEzOWNiYTMxNmEyMzYzMDUxNzk1ZDU2NmZiMmEyYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yYypxiqExmFW6O7EiQA5UavihHZHUAql2Rn8aJx6DGY)
> 
> 建议看到邮件的各位试试看

搞来搞去，发现没那么复杂，广东电信pppoe，直接把wan6删掉就行了。没有任何影响

Elec4d0, 2024-04-06:
> 搞来搞去，发现没那么复杂，广东电信pppoe，直接把wan6删掉就行了。没有任何影响

删掉wan6能拿到 v6-pd吗？ 我之前就是只有wan，只给v6没有pd

sss132919, 2024-04-08:
> 删掉wan6能拿到 v6-pd吗？ 我之前就是只有wan，只给v6没有pd

不是很懂，但是ipv6正常用，就没管了。仅限于本源码

Magic989, 2024-04-28:
我写了个简单的小脚本来解决这个问题，思路就是检查 WAN6 的 IPv6-PD 是否为空，如果为空，则重启网络，每隔多少时间执行由大家自行决定。这个一般是在二级路由或者多级路由上执行，因为主路由重拨后一般能获取 IPv6-PD，但多级路由在主路由重拨之后 IPv6-PD 和 DHCPv6 会消失，只留下个 SLAAC 地址

```ipv6_pd_check.sh
#!/bin/sh
ipv6_pd_status=$(ubus call network.interface.wan6 status | jsonfilter -e '@["ipv6-prefix"][0]["address"]')
if [ -z "$ipv6_pd_status" ]; then
    echo "IPv6-PD address not found on wan6 interface. Restarting network..."
    /etc/init.d/network restart
    echo "network restarted."
else
    echo "IPv6-PD address found on wan6 interface: $ipv6_pd_status"
fi
```

ATang007ZH, 2024-09-10:
给需要的人看看吧：
网上搜到了最简单的解决办法，我的是21.02 版imm 修改了文件
package/network/services/ppp/files/lib/netifd/ppp-down

在文件末增加一行
sleep 3

重新编译后问题再没有出现了。

coolsnowwolf, 2026-09-05:
现有复现讨论表明问题取决于 PPPoE 的 wan/wan6 配置和运营商重拨时序：多个用户通过禁用重复的 wan6 DHCPv6 客户端、让 wan 负责 PD，或在 ppp-down 后等待接口状态稳定解决；没有统一的 LEDE 源码错误或稳定复现日志。当前 issue 不足以支持修改 netifd/odhcp6c，关闭为配置/运营商特定问题。

closed 2026-09-05 by coolsnowwolf