coolsnowwolf/lede #10719: X64 云编译出错

NeedforGit, 2023-01-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：X64 云编译出错, 提示 The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files.

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：

Hunk #1 FAILED at 10.
1 out of 1 hunk FAILED -- saving rejects to file go.sum.rej
Patch failed!  Please fix ./patches/010-chore-deps-update-iploc-to-fix-build-with-mips.patch!
make[3]: Leaving directory '/workdir/openwrt/package/small/brook'
make[3]: *** [Makefile:51: /workdir/openwrt/build_dir/target-x86_64_musl/brook-20230101/.prepared_08355bf4ad0ed061ec4523d40e61a259_6664517399ebbbc92a37c5bb081b5c53] Error 1
time: package/small/brook/compile#0.20#0.06#0.26
    ERROR: package/small/brook failed to build.
**make[2]: *** [package/Makefile:116: package/small/brook/compile] Error 1**
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
**make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2**
Error: Process completed with exit code 2.

1054009064, 2023-01-03:
你要去自定义的软件源反馈 package/small

closed 2023-01-03 by aiamadeus

NeedforGit, 2023-01-04:
> 你要去自定义的软件源反馈 package/small

好的，我去反馈一下，感谢回复。