coolsnowwolf/lede #10721: 拉取最新源码编译报错，提示缺少依赖

bi4tms, 2023-01-03:
拉取最新源码编译x86的6.1内核的固件报错，具体如下：
SHELL= flock /home/x86.6.1/lede/tmp/.root-copy.flock -c 'cp -fpR /home/x86.6.1/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/x86.6.1/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/x86.6.1/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-drm-amdgpu is missing dependencies for the following libraries:
video.ko
make[3]: *** [modules/video.mk:346: /home/x86.6.1/lede/bin/targets/x86/64/packages/kmod-drm-amdgpu_6.1.2-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/x86.6.1/lede/package/kernel/linux'
time: package/kernel/linux/compile#3.76#2.60#6.50
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/x86.6.1/lede'
make[1]: *** [package/Makefile:110: /home/x86.6.1/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.6.1/lede'
make: *** [/home/x86.6.1/lede/include/toplevel.mk:231: world] Error 2

closed 2023-01-03 by bi4tms

mulder125, 2023-01-04:
此问题似乎是6.1.2出现的，但看到你提了issue又关闭了。后来怎么解决的？改编译5.15内核？

WYC-2020, 2023-01-04:
[image]
撤销这里的修改，我这边编译没问题

commit fa691af8 (2023-01-04): kmod-drm-*: fixes depends on x86 with linux 6.1

Fixes: #10721