coolsnowwolf/lede #10722: 最近半个月编译红米AX6000报错

ramondelee, 2023-01-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近半个月编译红米AX6000报错

### (2) 路由器型号和固件版本

A：红米AX6000，最近1月3日之前约半个月的源码

### (3) 详细日志

A：make[3]: Entering directory '/opt/lede/package/kernel/mt76'
make[3]: *** No rule to make target '/opt/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/backport/autoconf.h', needed by '/opt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2022-12-22-5b509e80/.configured_b1c6ae92f8e974e60b47b3e5ad3573cc'.  Stop.
make[3]: Leaving directory '/opt/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#0.07#0.05#0.12
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/opt/lede'
make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
make: *** [/opt/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2023-01-05:
日志

ramondelee, 2023-01-11:
make[3]: Entering directory '/opt/lede/package/network/utils/resolveip'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/opt/lede/package/network/utils/resolveip'
time: package/network/utils/resolveip/compile#0.08#0.03#0.17
make[3]: Entering directory '/opt/lede/feeds/packages/net/shadowsocks-libev'
mkdir -p /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /opt/lede/tmp/.root-copy.flock -c 'cp -fpR /opt/lede/build_dir/target-aarch64_cortex-a53_musl/shadowsocks-libev-3.3.5/.pkgdir/shadowsocks-libev-ss-local/. /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.shadowsocks-libev-ss-local_installed
echo "shadowsocks-libev-ss-local" >> /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/shadowsocks-libev.default.install
mkdir -p /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /opt/lede/tmp/.root-copy.flock -c 'cp -fpR /opt/lede/build_dir/target-aarch64_cortex-a53_musl/shadowsocks-libev-3.3.5/.pkgdir/shadowsocks-libev-ss-redir/. /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.shadowsocks-libev-ss-redir_installed
echo "shadowsocks-libev-ss-redir" >> /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/shadowsocks-libev.default.install
make[3]: Leaving directory '/opt/lede/feeds/packages/net/shadowsocks-libev'
time: package/feeds/packages/shadowsocks-libev/compile#0.09#0.03#0.12
make[3]: Entering directory '/opt/lede/package/kernel/gpio-button-hotplug'
mkdir -p /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /opt/lede/tmp/.root-copy.flock -c 'cp -fpR /opt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/gpio-button-hotplug/.pkgdir/kmod-gpio-button-hotplug/. /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.kmod-gpio-button-hotplug_installed
echo "kmod-gpio-button-hotplug" >> /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/gpio-button-hotplug.default.install
make[3]: Leaving directory '/opt/lede/package/kernel/gpio-button-hotplug'
time: package/kernel/gpio-button-hotplug/compile#0.07#0.04#0.18
make[3]: Entering directory '/opt/lede/package/kernel/leds-ws2812b'
mkdir -p /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /opt/lede/tmp/.root-copy.flock -c 'cp -fpR /opt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/.pkgdir/kmod-leds-ws2812b/. /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.kmod-leds-ws2812b_installed
echo "kmod-leds-ws2812b" >> /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/leds-ws2812b.default.install
make[3]: Leaving directory '/opt/lede/package/kernel/leds-ws2812b'
time: package/kernel/leds-ws2812b/compile#0.07#0.03#0.10
make[3]: Entering directory '/opt/lede/package/firmware/wireless-regdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/opt/lede/package/firmware/wireless-regdb'
time: package/firmware/wireless-regdb/compile#0.06#0.02#0.07
make[3]: Entering directory '/opt/lede/package/libs/libtool'
make[3]: Leaving directory '/opt/lede/package/libs/libtool'
time: package/libs/libtool/compile#0.06#0.03#0.15
make[3]: Entering directory '/opt/lede/package/libs/wolfssl'
mkdir -p /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
echo '5.5.4.e624513f' | cmp -s - /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libwolfssl.version || { echo '5.5.4.e624513f' > /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libwolfssl.version;  cp /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libwolfssl.version /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/libcyassl.version;  } 
SHELL= flock /opt/lede/tmp/.root-copy.flock -c 'cp -fpR /opt/lede/build_dir/target-aarch64_cortex-a53_musl/wolfssl-regular/wolfssl-5.5.4-stable/.pkgdir/libwolfssl/. /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.libwolfssl_installed
echo "libwolfssl" >> /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/wolfssl.regular.install
make[3]: Leaving directory '/opt/lede/package/libs/wolfssl'
time: package/libs/wolfssl/regular/compile#0.08#0.12#0.45
make[3]: Entering directory '/opt/lede/package/network/services/hostapd'
mkdir -p /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /opt/lede/tmp/.root-copy.flock -c 'cp -fpR /opt/lede/build_dir/target-aarch64_cortex-a53_musl/hostapd-wpad-basic-wolfssl/hostapd-2022-01-16-cff80b4f/.pkgdir/hostapd-common/. /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.hostapd-common_installed
echo "hostapd-common" >> /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/hostapd.wpad-basic-wolfssl.install
mkdir -p /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
[... cut ...] 3573cc'.  Stop.
make[3]: Leaving directory '/opt/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#0.08#0.05#0.12
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/opt/lede'
make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
make: *** [/opt/lede/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2026-09-05:
当前 mt76 Makefile 已将 `mac80211-backport/backport/autoconf.h` 纳入 `STAMP_CONFIGURED_DEPENDS`，会在 mac80211 生成该头文件后再配置 mt76；旧的缺失依赖问题已修复，关闭为 completed。

closed 2026-09-05 by coolsnowwolf